verilator/verilator #5525: Issue with foreach[i, j] on queue and dynamic array

YilouWang, 2024-10-09:
Hi,
I have encountered a problem with Verilator where the `foreach[i, j]` construct works correctly for unpacked array, but does not behave as expected for queue and dynamic array. 
It seems like: for queue and dyn_arr, which are size-variable data structures, `foreach[i, j, ...]` does not properly handle varying sizes across each dimension. Instead, it treats the outermost dimension's size as the bound for all subsequent dimensions, resulting in incomplete traversal.
Note that: if a nested foreach loop is used, it works as expected and correctly traverses all elements. The problem only occurs when using the `foreach[i, j]` form.

```
module t;
    int dyn_arr[][];
    int unpacked_arr [3:1][9:8];
    int queue[$][$];

    initial begin
        queue = '{'{1, 2, 3}, '{4, 5, 6, 0, 10}, '{6, 7, 8, 9}};
        $display("Queue:");
        foreach(queue[i]) begin
            foreach(queue[i][j]) begin
                $display("i = %d, j = %d." , i, j);
            end
        end
        $display("--------------------------------------------");
        foreach(queue[i, j]) begin
            $display("i = %d, j = %d." , i, j);
        end

        $display("Dyn Arr:");
        dyn_arr = '{'{1, 2, 3}, '{4, 5, 6, 0, 10}, '{6, 7, 8, 9}};
        foreach(dyn_arr[i]) begin
            foreach(dyn_arr[i][j]) begin
                $display("i = %d, j = %d." , i, j);
            end
        end
        $display("--------------------------------------------");
        foreach(dyn_arr[i, j]) begin
            $display("i = %d, j = %d." , i, j);
        end

        $display("Unpacked Arr:");
        foreach(unpacked_arr[i]) begin
            foreach(unpacked_arr[i][j]) begin
                $display("i = %d, j = %d." , i, j);
            end
        end
        $display("--------------------------------------------");
        foreach(unpacked_arr[i, j]) begin
            $display("i = %d, j = %d." , i, j);
        end
    end

endmodule

```
Output: expected vs. actual:
```
Queue:
i =           0, j =           0.
i =           0, j =           1.
i =           0, j =           2.
i =           1, j =           0.
i =           1, j =           1.
i =           1, j =           2.
i =           1, j =           3.
i =           1, j =           4.
i =           2, j =           0.
i =           2, j =           1.
i =           2, j =           2.
i =           2, j =           3.
--------------------------------------------
i =           0, j =           0.
i =           0, j =           1.
i =           0, j =           2.
i =           1, j =           0.
i =           1, j =           1.
i =           1, j =           2.
i =           2, j =           0.
i =           2, j =           1.
i =           2, j =           2.
Dyn Arr:
i =           0, j =           0.
i =           0, j =           1.
i =           0, j =           2.
i =           1, j =           0.
i =           1, j =           1.
i =           1, j =           2.
i =           1, j =           3.
i =           1, j =           4.
i =           2, j =           0.
i =           2, j =           1.
i =           2, j =           2.
i =           2, j =           3.
--------------------------------------------
i =           0, j =           0.
i =           0, j =           1.
i =           0, j =           2.
i =           1, j =           0.
i =           1, j =           1.
i =           1, j =           2.
i =           2, j =           0.
i =           2, j =           1.
i =           2, j =           2.
Unpacked Arr:
i =           3, j =           9.
i =           3, j =           8.
i =           2, j =           9.
i =           2, j =           8.
i =           1, j =           9.
i =           1, j =           8.
--------------------------------------------
i =           3, j =           9.
i =           3, j =           8.
i =           2, j =           9.
i =           2, j =           8.
i =           1, j =           9.
i =           1, j =           8.
- S i m u l a t i o n   R e p o r t: Verilator 5.029 devel
```

YilouWang, 2024-10-09:
I'd like to fix this but could anyone please give any hints or suggestions?

wsnyder, 2024-10-09:
Thanks for looking into a fix.  I'd first make the test self-checking without golden file, then check another simulator (though it does seem correct), then run Verilator with `--debug --debugi-V3Begin 9` and looking at the .tree files generated you'll see V3Begin.cpp is what transforms the `foreach` into `while` loops.  So, something needs fixing inVv3Begin.

YilouWang, 2024-10-10:
Hello, @wsnyder 
thank you for your suggestion. I followed your advice and did some debugging. First, I prepared a self-test and ran it with **vsim**, and it indeed worked as expected. Then, I went into V3Begin.cpp to debug further, made some changes, and now I've pushed the update.

closed 2024-10-10 by wsnyder