verilator/verilator #5526: Coverage missing lines that were definitely hit

zackglazewski, 2024-10-10:
I am encountering an issue where I generate a coverage report on a simple finite state machine file. 
I have a simple test bench with some simple tests. 

I'm running on Mac Sonoma 14.6, verilator version: Verilator 5.028 2024-08-21 rev UNKNOWN.REV
I'm not sure about the git master version, I installed it from brew, im assuming that is main.

[image]
[image]

[image]
[image]
[image]
[image]
[image]

I have checked the wave forms and the expected signals are going off. 
However, the entire fsm logic is getting 0 hits, and showing poor code coverage. 
Upon adding a single display statement in the fsm code block, all of a sudden the coverage report starts making sense and showing that the right lines are being hit. 

I'm reporting this as a bug, because it seems like it is, but please let me know if there is something that is being overlooked here.

wsnyder, 2024-10-10:
Can you please post a self-contained test case that shows no coverage in the report, then shows is coverage with the delay added?

zackglazewski, 2024-10-10:
@wsnyder Sure, sorry I kinda forgot I could just put the code in markdown 

Here is the testbench
```verilog
`timescale 1ns/1ps

module testbench;

    // Parameters
    parameter CLK_PERIOD = 10; // Clock period in ns

    // Signals
    logic clk;
    logic rst_n;
    logic IN;
    logic MATCH;

    // Instantiate the DUT
    verified_fsm dut (
        .IN(IN),
        .MATCH(MATCH),
        .CLK(clk),
        .RST(~rst_n)
    );

    // Clock generation
    always begin
        clk = 1'b1;
        #(CLK_PERIOD/2);
        clk = 1'b0;
        #(CLK_PERIOD/2);
    end

    // Test stimulus
    initial begin
        // Initialize inputs
        rst_n = 1'b0;
        IN = 1'b0;

        // Reset sequence
        #(CLK_PERIOD*5);
        rst_n = 1'b1;

        // Test case 4: Complete sequence
        #(CLK_PERIOD);
        IN = 1'b1;
        #(CLK_PERIOD);
        IN = 1'b1;
        assert(MATCH === 1'b1) else $error("Test 4 failed: Complete sequence not working");

        // End simulation
        #(CLK_PERIOD*10);

        $display("All tests passed");
        $finish;
    end

    // Monitor and check outputs
    always @(posedge clk) begin
        if (rst_n) begin
            // Add checks for DUT outputs here
        end
    end

    // Dump waveforms
    initial begin
        $dumpfile("testbench.vcd");
        $dumpvars(0, testbench);
    end

endmodule
```
Without the display:
```
0 : always@(ST_cr or IN) begin
25            0 :         temp_next = 1'b1;
26            0 :         case(ST_cr)
27            0 :                 s0:begin
28            0 :                         if (IN==0)
29            0 :                                 ST_nt = s0;
30              :                         else
31            0 :                                 ST_nt = s1;
32              :                 end
```

With the display:
```
24            3 : always@(ST_cr or IN) begin
25            3 :         temp_next = 1'b1;
26            3 :         $display("test");
27            3 :         case(ST_cr)
28            2 :                 s0:begin
29            1 :                         if (IN==0)
30            1 :                                 ST_nt = s0;
31              :                         else
32            1 :                                 ST_nt = s1;
33              :                 end
```
      
      
 Here is the dut:
 ```verilog
 `timescale 1ns/1ns

module verified_fsm(IN,MATCH,CLK,RST);
input IN,CLK,RST;
output reg MATCH;

typedef enum reg [2:0] {s0,s1,s2,s3,s4,s5} state_t;

reg temp_next, temp;

state_t ST_cr,ST_nt;

always@(posedge CLK or posedge RST) begin
	if(RST) begin
		ST_cr <= s0;
                temp <= 0;
        end
	else begin
		ST_cr <= ST_nt;
                temp <= temp_next;
        end
end

always@(ST_cr or IN) begin
        temp_next = 1'b1;
        // PUTTING A DISPLAY HERE CAUSES A CHANGE
        // $display("hi");
	case(ST_cr)
		s0:begin
			if (IN==0)
				ST_nt = s0;
			else
				ST_nt = s1;
		end
		
		s1:begin
                        if (IN==0)
                                ST_nt = s2;
                        else
                                ST_nt = s1;
                end

                s2:begin
                        if (IN==0)
                                ST_nt = s3;
                        else
                                ST_nt = s1;
                end

                s3:begin
                        if (IN==0)
                                ST_nt = s0;
                        else
                                ST_nt = s4;
                end

                s4:begin
                        if (IN==0)
                                ST_nt = s2;
                        else
                                ST_nt = s5;
                end

                s5:begin
                        if (IN==0)
                                ST_nt = s2;
                        else
                                ST_nt = s1;
                end

	endcase
end

always@(posedge CLK or posedge RST) begin
        if(RST)
                MATCH  <= 0;
        else if (ST_cr == s4 && IN == 1)
                MATCH  <= 1;
	else 
		MATCH  <= 0;
end

endmodule
 ```

wsnyder, 2024-10-10:
This is #5473/#5474, see the workaround or patch there please.

closed 2024-10-10 by wsnyder

zackglazewski, 2024-10-10:
> This is #5473/#5474, see the workaround or patch there please.

Thank you! I'll try this out.