verilator/verilator #5538: Internal Error: ../V3AstNodeDType.h:1177: Typedef not linked

solomatnikov, 2024-10-15:
```
%Error: Internal Error: l2_smoke_test_gen.sv:6:12: ../V3AstNodeDType.h:1177: Typedef not linked
    6 |    typedef uvm_object_registry#(l2_smoke_test_gen,"l2_smoke_test_gen") type_id; 
      |            ^~~~~~~~~~~~~~~~~~~
```

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

Subset of UVM using https://github.com/solomatnikov/uvm-verilator/tree/current-patches-deprecated-api-no-timing

forked from https://github.com/antmicro/uvm-verilator/tree/current-patches-deprecated-api

What 'verilator' command line do we use to run your example?

`verilator  -Wno-LATCH -DUVM_NO_DPI -Wno-lint -Wno-style -Wno-CONSTRAINTIGN -Wno-ZERODLY -Wno-SYMRSVDWORD --cc --exe -sv --assert --trace-fst --trace-params --trace-structs --report-unoptflat --no-timing --timescale 1ps/1ps --x-assign unique ...`

What 'verilator --version' are you using?  Did you try it with the git master version?

`Verilator 5.029 devel rev v5.028-201-ga3d0cc652`

What OS and distribution are you using?

`Linux 5.14.0-427.33.1.el9_4.x86_64`

May we assist you in trying to fix this in Verilator yourself?

closed 2024-10-15 by wsnyder

solomatnikov, 2024-10-15:
This has nothing to do with UVM

solomatnikov, 2024-10-16:
It seems the attitude was more reasonable in the past (structs, etc.). 

If 15-old UVM is still not supported, then it might not be supported for 5, 10 or more years in the future (just extrapolating from the past). 

Does it mean any bug that also affects UVM would not be fixed and would be immediately closed no matter what? Seems strange

solomatnikov, 2024-10-16:
Just for the record: this can be a very small test case:

```
virtual class uvm_void;
endclass
virtual class uvm_object extends uvm_void;
  extern function new (string name="");
  virtual function uvm_object create (string name=""); return null; endfunction
  local string m_leaf_name;
  local int m_inst_id;
  static protected int m_inst_count;
endclass
function uvm_object::new (string name="");
  m_inst_id = m_inst_count++;
  m_leaf_name = name;
endfunction
virtual class uvm_object_wrapper;
  virtual function uvm_object create_object (string name="");
    return null;
  endfunction
endclass
class uvm_object_registry #(type T=uvm_object, string Tname="<unknown>")
                                        extends uvm_object_wrapper;
  typedef uvm_object_registry #(T,Tname) this_type;
  virtual function uvm_object create_object(string name="");
    T obj;
    if (name=="") obj = new();
    else obj = new(name);
    return obj;
  endfunction
  static function T create (string name="",  
                            string contxt="");
     return null;
  endfunction
endclass
class uvm_abstract_object_registry #(type T=uvm_object, string Tname="<unknown>")
                                        extends uvm_object_wrapper;
  typedef uvm_abstract_object_registry #(T,Tname) this_type;
  virtual function uvm_object create_object(string name="");
    return null;
  endfunction
  static function T create (string name="",  
                            string contxt="");
     return null;
  endfunction
endclass
virtual class l_base_test_gen extends uvm_object;
   typedef uvm_abstract_object_registry#(l_base_test_gen,"l_base_test_gen") type_id; 
   function new(string name = "l_base_test_gen");
      super.new(name);
   endfunction
endclass  
class l_smoke_test_gen extends l_base_test_gen;
   typedef uvm_object_registry#(l_smoke_test_gen,"l_smoke_test_gen") type_id; 
   function new(string name = "l_smoke_test_gen");
      super.new(name);
   endfunction
endclass  
module l_bfm_sender
  (
   input  logic     clk,
   input  logic     reset_l
   );
   l_base_test_gen test_gen = null;  
   task automatic execute_test(l_base_test_gen _test_gen);
      test_gen = _test_gen;
   endtask  
endmodule  
module test_core_wrapper
  (
   input  logic     clk,
   input  logic     reset_l
   );
   l_bfm_sender mem_l_bfm ( .* );
   integer cyc_cnt = 0;
   l_base_test_gen mem_test_gen;
   initial begin
      mem_test_gen  = l_smoke_test_gen::type_id::create("l_smoke_test_gen");
      mem_l_bfm.execute_test(mem_test_gen);
   end
   always @ (posedge clk) begin
      if (!reset_l) begin
	 cyc_cnt <= 0;
      end
      else begin
	 cyc_cnt <= cyc_cnt + 1;
      end
   end  
endmodule  
module top
(
 input clk,
 input reset_l
 );
   test_core_wrapper tile( .* );
endmodule  
```

with the same error:

```
%Error: Internal Error: test.sv:51:12: ../V3AstNodeDType.h:1177: Typedef not linked
   51 |    typedef uvm_object_registry#(l_smoke_test_gen,"l_smoke_test_gen") type_id; 
      |            ^~~~~~~~~~~~~~~~~~~
```

wsnyder, 2024-10-16:
(Mis-)handling of UVM type_id is a known bug.  type_id is mentioned in #1538 which I referenced earlier.

solomatnikov, 2024-10-16:
> (Mis-)handling of UVM type_id is a known bug. type_id is mentioned in https://github.com/verilator/verilator/issues/1538 which I referenced earlier.

No, it is specific to this particular case, i.e. removing just one level of module hierarchy:

```
virtual class uvm_void;
endclass
virtual class uvm_object extends uvm_void;
  extern function new (string name="");
  virtual function uvm_object create (string name=""); return null; endfunction
  local string m_leaf_name;
  local int m_inst_id;
  static protected int m_inst_count;
endclass
function uvm_object::new (string name="");
  m_inst_id = m_inst_count++;
  m_leaf_name = name;
endfunction
virtual class uvm_object_wrapper;
  virtual function uvm_object create_object (string name="");
    return null;
  endfunction
endclass
class uvm_object_registry #(type T=uvm_object, string Tname="<unknown>")
                                        extends uvm_object_wrapper;
  typedef uvm_object_registry #(T,Tname) this_type;
  virtual function uvm_object create_object(string name="");
    T obj;
    if (name=="") obj = new();
    else obj = new(name);
    return obj;
  endfunction
  static function T create (string name="",  
                            string contxt="");
     return null;
  endfunction
endclass
class uvm_abstract_object_registry #(type T=uvm_object, string Tname="<unknown>")
                                        extends uvm_object_wrapper;
  typedef uvm_abstract_object_registry #(T,Tname) this_type;
  virtual function uvm_object create_object(string name="");
    return null;
  endfunction
  static function T create (string name="",  
                            string contxt="");
     return null;
  endfunction
endclass
virtual class l_base_test_gen extends uvm_object;
   typedef uvm_abstract_object_registry#(l_base_test_gen,"l_base_test_gen") type_id; 
   function new(string name = "l_base_test_gen");
      super.new(name);
   endfunction
endclass  
class l_smoke_test_gen extends l_base_test_gen;
   typedef uvm_object_registry#(l_smoke_test_gen,"l_smoke_test_gen") type_id; 
   function new(string name = "l_smoke_test_gen");
      super.new(name);
   endfunction
endclass  
module l_bfm_sender
  (
   input  logic     clk,
   input  logic     reset_l
   );
   l_base_test_gen test_gen = null;  
   task automatic execute_test(l_base_test_gen _test_gen);
      test_gen = _test_gen;
   endtask  
endmodule  
module top // test_core_wrapper
  (
   input  logic     clk,
   input  logic     reset_l
   );
   l_bfm_sender mem_l_bfm ( .* );
   integer cyc_cnt = 0;
   l_base_test_gen mem_test_gen;
   initial begin
      mem_test_gen  = l_smoke_test_gen::type_id::create("l_smoke_test_gen");
      mem_l_bfm.execute_test(mem_test_gen);
   end
   always @ (posedge clk) begin
      if (!reset_l) begin
	 cyc_cnt <= 0;
      end
      else begin
	 cyc_cnt <= cyc_cnt + 1;
      end
   end  
endmodule  
/* -----\/----- EXCLUDED -----\/-----
module top
(
 input clk,
 input reset_l
 );
   test_core_wrapper tile( .* );
endmodule  
 -----/\----- EXCLUDED -----/\----- */
```
avoids this bug:
```
- V e r i l a t i o n   R e p o r t: Verilator 5.029 devel rev v5.028-201-ga3d0cc652
- Verilator: Built from 0.016 MB sources in 11 modules, into 0.078 MB in 41 C++ files needing 0.000 MB
- Verilator: Walltime 0.021 s (elab=0.001, cvt=0.007, bld=0.000); cpu 0.021 s on 1 threads; alloced 8.031 MB
```

P.S. Just to be clear: I am not a fan of UVM, factory pattern, etc.