verilator/verilator #5539: Can Verilator compile only C++ files without Verilog files

alpha-beta-user, 2024-10-15:
I have a quick question regarding your project. Is it possible to use Verilator to compile only C++ files without having any Verilog files in the project? I'm looking for a way to utilize Verilator with just C++ code.

Because I encountered errors when trying to compile several pure C++ files, it seems that Verilator requires Verilog files to be present.

Thank you in advance for your help!

wsnyder, 2024-10-15:
No, I'm not understanding, what would be the point?

alpha-beta-user, 2024-10-15:
> No, I'm not understanding, what would be the point?

I want to ultilize the mutithread. my origin work solwly.

wsnyder, 2024-10-15:
Verilator wont partition C++ routines into threads.

Either use raw C++ threads without Verilator, or make a small Verilog file that does nothing (or calls your C) to appease Verilator.

closed 2024-10-15 by wsnyder

alpha-beta-user, 2024-10-15:
> Verilator wont partition C++ routines into threads.
> 
> Either use raw C++ threads without Verilator, or make a small Verilog file that does nothing (or calls your C) to appease Verilator.

Thank you for your reply.

I also have a question. If I have some systemc models, two(for example) systemc models is built by verilator using multithread, some is created myself. if I use just systemc methods to link them into a new project, is it multithread?

wsnyder, 2024-10-15:
The SystemC models Verilator builds will use threads underneath them, but the SystemC kernel is not multithreaded so nothing else will be.