verilator/verilator #5546: Long module names cause simulation to fail. Is there some limit?

badanf, 2024-10-18:
Hello, I am trying to simulate HLS-generated model in Verilator. In most cases it works well, but some generated models are failing with following message:
`*** stack smashing detected ***: terminated`
When I add -fno-stack-protector to compile options to remove this check, I get (unsurprisingly) segmentation fault. When I traced the error I see following trace:
```
Thread 1 "Vtop" received signal SIGSEGV, Segmentation fault.
0x0000555555573f24 in VL_CVT_PACK_STR_NW[abi:cxx11](int, unsigned int const*) ()
(gdb) backtrace
#0  0x0000555555573f24 in VL_CVT_PACK_STR_NW[abi:cxx11](int, unsigned int const*) ()
#1  0x726874374c5a5f70 in ?? ()
#2  0x0000555555fc0bb0 in ?? ()
#3  0x0000000000000136 in ?? ()
#4  0x0000000000000136 in ?? ()
#5  0x00005555555783ad in VlDeleter::deleteAll() ()
#6  0x00005555555b2163 in Vtop::eval_step() ()
#7  0x000055555556fe8f in main ()
```
I tried to investigate difference between working and not working generated designs and I found the only difference is missing prefix from the module names (and instances names) in working models.
And indeed with simple replace on module name prefix on RTL code to make it shorter, I was able to make failing model work. For context we are talking about 125 characters on longest module name I was able to find. After removing prefix of length about 20 characters it started to work. Even from the trace log it seems the problem is with some string handling inside the model.

I tried this on v5.028 and the newest master with some result. I am using verilator through cocotb, but it seems the issue is with Verilator (correct me if I am wrong)

So my questions is, is there some limit on how long module names can be? I wasn't able to find any. Is there some way to make it work even with such long names - for example to limit size of some generated strings used in the verilated model?
Thanks!

wsnyder, 2024-10-18:
Verilator is supposed to hash long names shorter to avoid trouble.

Can you please post the backtrack after using `--runtime-debug`?

Also, if you can, a small test case.

badanf, 2024-10-21:
So I looked into the issue, since the string should not be the problem, and indeed I found the culprit.
I tried to find what else is connected with module names and I realized the only other place where it is used are actually $readmemh accesses to initialize memories. And since all the .dat files read by these functions are in the folders named after the modules they are supposed to be read in, shorter module names did really help, but it was not the issue.

It seems that there is some limit on how long the path in $readmemh can be, probably because of some string handling functions.
This is a trace after using --runtime-debug:
```
==8659==ERROR: AddressSanitizer: stack-buffer-overflow on address 0x7ffda7a3d601 at pc 0x558c6b1d86e1 bp 0x7ffda7a3d470 sp 0x7ffda7a3d460
WRITE of size 1 at 0x7ffda7a3d601 thread T0
    #0 0x558c6b1d86e0 in VL_CVT_PACK_STR_NW[abi:cxx11](int, unsigned int const*) /usr/local/share/verilator/include/verilated.cpp:1929
    #1 0x558c6b013821 in Vtop___024root___eval_initial__TOP(Vtop___024root*) ~/cocotb/sim_build/Vtop___024root__DepSet_heccd7ead__0__Slow.cpp:847
    #2 0x558c6b02f47e in Vtop___024root___eval_initial(Vtop___024root*) ~/cocotb/sim_build/Vtop___024root__DepSet_heccd7ead__0__Slow.cpp:309
    #3 0x558c6b643e81 in Vtop::eval_step() ~/cocotb/sim_build/Vtop.cpp:64
    #4 0x558c6b18732e in main ~/.local/lib/python3.8/site-packages/cocotb/share/lib/verilator/verilator.cpp:129
    #5 0x7fe32afdb082 in __libc_start_main ../csu/libc-start.c:308
    #6 0x558c6b18857d in _start (~/cocotb/sim_build/Vtop+0x281757d)

Address 0x7ffda7a3d601 is located in stack of thread T0 at offset 337 in frame
    #0 0x558c6b1d826f in VL_CVT_PACK_STR_NW[abi:cxx11](int, unsigned int const*) /usr/local/share/verilator/include/verilated.cpp:1918

  This frame has 3 object(s):
    [48, 49) '<unknown>'
    [64, 65) '<unknown>'
    [80, 337) 'destout' (line 1920) <== Memory access at offset 337 overflows this variable
HINT: this may be a false positive if your program uses some custom stack unwind mechanism, swapcontext or vfork
      (longjmp and C++ exceptions *are* supported)
SUMMARY: AddressSanitizer: stack-buffer-overflow /usr/local/share/verilator/include/verilated.cpp:1929 in VL_CVT_PACK_STR_NW[abi:cxx11](int, unsigned int const*)
Shadow bytes around the buggy address:
  0x100034f3fa70: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x100034f3fa80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x100034f3fa90: 00 00 00 00 00 00 f1 f1 f1 f1 f1 f1 01 f2 01 f2
  0x100034f3faa0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x100034f3fab0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x100034f3fac0:[01]f3 f3 f3 f3 f3 f3 f3 f3 f3 00 00 00 00 00 00
  0x100034f3fad0: 00 00 00 00 00 00 f1 f1 f1 f1 f8 f2 f2 f2 f8 f2
  0x100034f3fae0: f2 f2 f8 f8 f8 f8 f2 f2 f2 f2 f8 f8 f8 f8 f2 f2
  0x100034f3faf0: f2 f2 00 00 00 00 f2 f2 f2 f2 00 00 00 00 f2 f2
  0x100034f3fb00: f2 f2 00 00 00 00 f2 f2 f2 f2 00 00 00 00 f2 f2
  0x100034f3fb10: f2 f2 00 00 00 00 f2 f2 f2 f2 00 00 00 00 f2 f2
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==8659==ABORTING
```
Actually this is much better for me, because there is easy workaround on my side, but just to let you know.

wsnyder, 2024-10-21:
Can you please edit verilated.cpp VL_CVT_PACK_STR_NW to print the lwords arguments, and also value of VL_VALUE_STRING_MAX_CHARS?

Verilator is supposed to warn when strings exceed that length, but perhaps is not.

badanf, 2024-10-22:
It is strange, because it doesn't seem to be so long:
```
Lwords: 39
VL_VALUE_STRING_MAX_CHARS: 256
lwp: tad.kcolbmem/zsci_q0q_0_slh_UAVM_1_00_020z7cx_omhugto1_nnif_negpi_neg_edoc/00_020z7cx_omhugto1_nnif/detareneg-nnif/spi-nnif/btococ/3nI/stcejorP/nadabf/emoh/
Lwords: 39
VL_VALUE_STRING_MAX_CHARS: 256
lwp: tad.kcolbmem/g42hbs7s_1_slh_UAVM_1_00_020z7cx_omhugto1_nnif_negpi_neg_edoc/00_020z7cx_omhugto1_nnif/detareneg-nnif/spi-nnif/btococ/3nI/stcejorP/nadabf/emoh/
Lwords: 81
VL_VALUE_STRING_MAX_CHARS: 256
lwp: tad.R1_OTUA_MOR_0_shserht7LZ_p_hctaB_maertS_etavitcA_rotceV_xirtaM_01_slh_UAVM_1_00_020z7cx_omhugto1_nnif/golirev/ldh/pi/lpmi/1los/01_slh_UAVM_1_00_020z7cx_omhugto1_nnif_tcejorp/0irhy9xn_01_slh_UAVM_1_00_020z7cx_omhugto1_nnif_negpi_neg_edoc/00_020z7cx_omhugto1_nnif/detareneg-nnif/spi-nnif/btococ/3nI/stcejorP/nadabf/emoh/
=================================================================
==29511==ERROR: AddressSanitizer: stack-buffer-overflow on address 0x7ffecd6e4391 at pc 0x55bcd379c27c bp 0x7ffecd6e4200 sp 0x7ffecd6e41f0
WRITE of size 1 at 0x7ffecd6e4391 thread T0
    #0 0x55bcd379c27b in VL_CVT_PACK_STR_NW[abi:cxx11](int, unsigned int const*) /usr/local/share/verilator/include/verilated.cpp:1933
    #1 0x55bcd35bd9ed in Vtop___024root___eval_initial__TOP(Vtop___024root*) ~/cocotb/sim_build/Vtop___024root__DepSet_heccd7ead__0__Slow.cpp:925
    #2 0x55bcd35d92ae in Vtop___024root___eval_initial(Vtop___024root*) ~/cocotb/sim_build/Vtop___024root__DepSet_heccd7ead__0__Slow.cpp:309
    #3 0x55bcd3c07a81 in Vtop::eval_step() ~/cocotb/sim_build/Vtop.cpp:64
    #4 0x55bcd374ae9e in main ~/.local/lib/python3.8/site-packages/cocotb/share/lib/verilator/verilator.cpp:129
    #5 0x7fe676396082 in __libc_start_main ../csu/libc-start.c:308
    #6 0x55bcd374c0ed in _start (~/cocotb/sim_build/Vtop+0x29a40ed)

Address 0x7ffecd6e4391 is located in stack of thread T0 at offset 337 in frame
    #0 0x55bcd379bddf in VL_CVT_PACK_STR_NW[abi:cxx11](int, unsigned int const*) /usr/local/share/verilator/include/verilated.cpp:1918

  This frame has 3 object(s):
    [48, 49) '<unknown>'
    [64, 65) '<unknown>'
    [80, 337) 'destout' (line 1920) <== Memory access at offset 337 overflows this variable
HINT: this may be a false positive if your program uses some custom stack unwind mechanism, swapcontext or vfork
      (longjmp and C++ exceptions *are* supported)
SUMMARY: AddressSanitizer: stack-buffer-overflow /usr/local/share/verilator/include/verilated.cpp:1933 in VL_CVT_PACK_STR_NW[abi:cxx11](int, unsigned int const*)
Shadow bytes around the buggy address:
  0x100059ad4820: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x100059ad4830: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x100059ad4840: 00 00 00 00 00 00 00 00 f1 f1 f1 f1 f1 f1 01 f2
  0x100059ad4850: 01 f2 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x100059ad4860: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x100059ad4870: 00 00[01]f3 f3 f3 f3 f3 f3 f3 f3 f3 00 00 00 00
  0x100059ad4880: 00 00 00 00 00 00 00 00 f1 f1 f1 f1 f8 f2 f2 f2
  0x100059ad4890: f8 f2 f2 f2 f8 f8 f8 f8 f2 f2 f2 f2 f8 f8 f8 f8
  0x100059ad48a0: f2 f2 f2 f2 00 00 00 00 f2 f2 f2 f2 00 00 00 00
  0x100059ad48b0: f2 f2 f2 f2 00 00 00 00 f2 f2 f2 f2 00 00 00 00
  0x100059ad48c0: f2 f2 f2 f2 00 00 00 00 f2 f2 f2 f2 00 00 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==29511==ABORTING

```

wsnyder, 2024-10-22:
Thanks for the debug, I've removed the constraint.

closed 2024-10-22 by wsnyder