verilator/verilator #5552: struct literal internal error

toddstrader, 2024-10-23:
@wsnyder we're running into a problem with this commit causing a lot of internal errors when we build:
```
commit 2fc94557a09f2d91d9f4b14752090af5cef9dea7 (HEAD -> master)
Author: Wilson Snyder <wsnyder@wsnyder.org>
Date:   Fri Oct 11 22:37:48 2024 -0400

    Fix error on enums with x/z using .name(), and internal refactoring
```

I haven't been able to reproduce in a `test_regress` test yet but am still working on it.  What I know so far is that `WidthVisitor::visit(AstPin*)` is failing on:
```
                    patternp->childDTypep(setDtp);
```
with:
```
%Error: Internal Error: foo.sv:129:47: ../V3Ast.cpp:418: Adding to non-empty, non-list op1
                                                                                          : ... note: In instance 'bar.baz'
-node: PATTERN 0x55555abd3a40 <e129620> {ez129bv} @dt=0@
  129 |         .struct_param(Some_pkg::struct_t'{
```

I've tried to reproduce with a struct literal assignment to a parameter but no luck so far with that.  When I look at one of our failing cases with the debugger I find that we're trying to set this data type:
```
$4 = (AstNode *) 0x55555b143c20
(rr) p AstNode::dumpGdb(newp)                                                                                                                                                                                                                                                                                                                                                                                                           
This=REFDTYPE 0x55555b143c20 back=0 headtail=this
  Fileline = qux.sv:29:27: 
  REFDTYPE 0x55555b143c20 <e305108#> {dp29bb} @dt=0x55555719ee00@(w97)  struct_t refdt=0x55555719ee00(w97) -> typedef=0x555557266ff0 -> STRUCTDTYPE 0x55555719ee00 <e305107#> {x8an} @dt=this@(w97)  Some_pkg::struct_t [PACKED]
```
but we already have this `m_op1p`:
```
$6 = (AstNode *) 0x55555abe5ef0
(rr) p AstNode::dumpGdb(m_op1p)
This=REFDTYPE 0x55555abe5ef0 back=0x55555abd3a40 headtail=this
  Fileline = foo.sv:129:34: 
  REFDTYPE 0x55555abe5ef0 <e129618> {ez129bi} @dt=0@  struct_t -> typedef=0x555557266ff0 -> STRUCTDTYPE 0x55555719ee00 <e305107#> {x8an} @dt=this@(w97)  Some_pkg::struct_t [PACKED]
```
I don't know if this is enough information to do anything with but I wanted to bounce this off of you while I continue to try to make a reproducer in case you can see what's going on here.

wsnyder, 2024-10-23:
I'd be just guessing at a fix. Note the example may need parameters to show the issue.

Probably a good case for using the new sv-bugpoint tool:
https://antmicro.com/blog/2024/09/sv-bugpoint-for-improved-debugging-in-sv-tools/

toddstrader, 2024-10-24:
Thanks for the pointer to `sv-bugpoint`.  I've been meaning to check this out and it's super cool.  Currently it doesn't seem to know how to rip out interfaces (probably not a huge surprise) but it's a super cool tool.

However, I eventually discovered that my hand-coded reproducer did capture the bug but only with `--debug`.  Closing this now and moving over to the PR.

closed 2024-10-24 by toddstrader