verilator/verilator #5556: Deterministic feature configuration

muffgaga, 2024-10-24:
When building current versions of  `verilator` some features are detected automatically without the possibility to disable a certain configuration (e.g. disabling ccache/mold/systemc if it is installed → `./configure --without-{systemc,mold,ccache}`).

For package managers (e.g. [spack](https://github.com/spack/spack) that aim for deterministic behavior this is a problem, cf. some parts of the discussion in [spack#47168](https://github.com/spack/spack/pull/47168)).

Is this something that would be okay reasonable to integrate? We could give it a try… (no autotools (nor cmake) experts here though).

wsnyder, 2024-10-24:
Yes, a patch to configure to add `./configure --without-{systemc,mold,ccache}` is ok.  In general the same effect can likely be had by appropriate environment variable changes, but that would be much cleaner.

However, please consider that using these will affect runtime (as mentioned in the other thread). This will give a bad experience to the runtime users, as these tools are fairly critical to reasonable runtimes.  Also I don't understand why using faster compiles/linking affects reproducibility.

wsnyder, 2024-10-24:
P.S. If you want to disable these only for compiling the verilator binaries itself, perhaps we need to split the flags into runtime and buildtime versions.

haampie, 2024-10-24:
For slightly more context:

1. "if X is available, use X" makes reproducibility hard when the build environment is not entirely controlled by the package manager. This is the case for Spack because it runs on top of any os/distro using its system compilers, which means /usr/bin is often exposed in the build environment, which may influence the choices of the configure script. So for instance whether `/usr/bin/mold` is installed would influence the binaries produced by Spack -- we'd rather let the package manager pass `LDFLAGS=-fuse-ld=mold` instead.

2. Also in case of Spack, the provided compiler is actually a compiler *wrapper*, which adds various flags such as `-Wl,-rpath,...`. That means that `ccache gcc foo.c` is actually wrong, because `ccache` cannot see that more flags are added when actually invoking the compiler. The compiler wrapper is really the one that should invoke `ccache <actual compiler call>` (which we make configurable for users).

muffgaga, 2024-10-24:
@haampie if I combine your statement from "2." and your last paragraph → for the ccache "variant" the package manager (spack) would need a runtime-specific ccache flag, correct? (As you said, at buildtime spack internally decides about using ccache in its gcc wrapper.)

wsnyder, 2025-02-25:
There is no plan to add this, unless someone wants to contribute a pull to do so.

closed 2025-02-25 by wsnyder