verilator/verilator #5560: syntax error, unexpected IDENTIFIER when using interface in a class

Ziyad-Y, 2024-10-24:
I have been learning SystemVerilog with verilator. Everything I've done so far has compiled. I only started having problems when using interface.  

CODE:
```
import alu_pkg::*;

interface alu_intf;  
    operation op_code;
    logic [15:0] a,b; 
    logic [15:0] result;
    logic status;

    modport DRV(    
        output op_code, a, b, 
        input result 
                    );

endinterface
```
```
class driver;  
    alu_intf alu_if.DRV;  // No virtual interface allowed in verilator    
    utils u;
    mailbox #(transaction) drv_mbx;  
    event done;
    function new(mailbox #(transaction) mbx, event done_event);  
        this.drv_mbx = mbx;
        this.done=done_event; 
        u= new();
    endfunction 

    task main();
        transaction t;
        forever begin
            
            if (drv_mbx.try_get(t)) begin
                $display("[DRV] RCV: A=%d, B=%d, op code(%d)=%s",t.a,t.b,t.op_code, u.opc(t.op_code));
            end else begin
                if(done.triggered) begin
                    $display("Finished Simulation");
                    break;
                end
                $display("DRV <- GEN ERROR, mailbox=%d\n", drv_mbx.num());
                #10ns;
            end
            #10ns;
        end
    endtask : main
    
endclass: driver
```
I get this error
[image]

When I comment out the declaration in the driver class, and utilize the interface in the test bench module, it compiles. I only have problems when using it in the driver class. (I am aware that there is no support for virtual interface)

wsnyder, 2024-10-24:
Did you try this on another simulator?  What in IEEE indicates this is legal?

Ziyad-Y, 2024-10-25:
After further research, I've found my answer. You cannot instantiate a interface in a class without virtual. Since interface is static and class is dynamic, the interface needs to be virtual.

closed 2024-10-25 by Ziyad-Y