verilator/verilator #5566: Internal error with $countbits in property assertion

hankhsu1996, 2024-10-26:
### Issue

Thanks for taking the time to review this.

#### Summary
When using `$countbits` in a property assertion, Verilator throws an internal error indicating an unexpected `AstNode` type, while `$countones` works without issue.

#### Code Example
Here's a minimal code snippet that reproduces the issue:

```systemverilog
module Counter #(
    parameter  int VectorWidth = 8,
    localparam int CounterWidth = $clog2(VectorWidth + 1)
) (
    input  logic                    clk,
    input  logic                    rst,
    input  logic [ VectorWidth-1:0] vec,
    output logic [CounterWidth-1:0] num_zeros,
    output logic [ VectorWidth-1:0] num_ones
);

  always_comb begin
    num_zeros = '0;
    num_ones  = '0;
    for (int i = 0; i < VectorWidth; i++) begin
      if (vec[i] == 0) begin
        num_zeros++;
      end else begin
        num_ones++;
      end
    end
  end

  assert property (@(negedge clk) disable iff (rst) (num_ones == $countones(vec)));
  assert property (@(negedge clk) disable iff (rst) (num_zeros == $countbits(vec, '0)));
endmodule
```

#### Error Message
Verilator fails with the following error:

```
%Error: Internal Error: test.sv:25:78: ../V3Ast.h:2478: AstNode is not of expected type, but instead has type 'SAMPLED'
                                     : ... note: In instance 'Testbench.counter'
   25 |   assert property (@(negedge clk) disable iff (rst) (num_zeros == $countbits(vec, '0)));
      |                                                                              ^~~
... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

Interestingly, `$countones` works correctly, even though it’s simply an alias for `$countbits(expression, '1)` as defined in LRM 20.9.

#### Verilator Version
```
Verilator 5.027 devel rev v5.026-13-gd6a294b78
```

#### Additional Information
- OS and distribution: Ubuntu 22.04.4 LTS
- Verilator command line:
  ```
  verilator test.sv --cc --exe --binary --main --assert
  ```
- Tried on Git master version: No

wsnyder, 2024-10-26:
Works for me, please try the latest git master.

wsnyder, 2024-10-26:
Also note these are different widths, which might be related to why only one worked:

   output logic [CounterWidth-1:0] num_zeros,
   output logic [ VectorWidth-1:0] num_ones

hankhsu1996, 2024-10-27:
Thank you for your prompt response. 

I tried the latest commit ([469eca7](https://github.com/verilator/verilator/commit/469eca7de20557d5e10ec78f93b9c6b5d8802f18)) and fix the width issue, but I'm still encountering the same problem.

```bash
$ verilator --version
Verilator 5.029 devel rev v5.028-222-g469eca7de

$ verilator test.sv --cc --exe --binary --main --assert
%Error: Internal Error: test.sv:25:78: ../V3Ast.h:2615: AstNode is not of expected type, but instead has type 'SAMPLED'
                                     : ... note: In instance 'Counter'
   25 |   assert property (@(negedge clk) disable iff (rst) (num_zeros == $countbits(vec, '0)));
      |                                                                              ^~~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

This is the `test.sv` file I'm using now:

```systemverilog
module Counter #(
    parameter  int VectorWidth = 8,
    localparam int CounterWidth = $clog2(VectorWidth + 1)
) (
    input  logic                    clk,
    input  logic                    rst,
    input  logic [ VectorWidth-1:0] vec,
    output logic [CounterWidth-1:0] num_zeros,
    output logic [CounterWidth-1:0] num_ones
);

  always_comb begin
    num_zeros = '0;
    num_ones  = '0;
    for (int i = 0; i < VectorWidth; i++) begin
      if (vec[i] == 0) begin
        num_zeros++;
      end else begin
        num_ones++;
      end
    end
  end

  assert property (@(negedge clk) disable iff (rst) (num_ones == $countones(vec)));
  assert property (@(negedge clk) disable iff (rst) (num_zeros == $countbits(vec, '0)));

endmodule
```

I'm unsure why this issue isn’t reproducible on your end. Could it be due to a compiler difference?

wsnyder, 2024-10-27:
I'm able to make this fail on my side.

The problem is $countbits accepts counting X/Zs so needs to be processed in V3Tristate, and currently only takes a variable. For cases where it is only counting 0/1s it could be easily handled as a non-tristate case by converting earlier. As to handling the Sampled itself, I'm not sure if we can just add a Sampled to the tristate variable - @kozdra?

kozdra, 2024-10-27:
My quick workaround (a proper fix should address tristate countbits as well):
```patch
From cf94d7c0d1a08495c7d9e8a237fa7d8cae1bc186 Mon Sep 17 00:00:00 2001
From: Arkadiusz Kozdra <business@arusekk.pl>
Date: Sun, 27 Oct 2024 11:06:49 +0100
Subject: [PATCH] Fix non-tristate countbits in assertions by tighter scoping

---
 src/V3Tristate.cpp | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/src/V3Tristate.cpp b/src/V3Tristate.cpp
index b30be761d..36eb84bbf 100644
--- a/src/V3Tristate.cpp
+++ b/src/V3Tristate.cpp
@@ -1433,7 +1433,6 @@ class TristateVisitor final : public TristateBaseVisitor {
         dropop[1] = VN_IS(nodep->thsp(), Const) && VN_AS(nodep->thsp(), Const)->num().isAnyZ();
         dropop[2] = VN_IS(nodep->fhsp(), Const) && VN_AS(nodep->fhsp(), Const)->num().isAnyZ();
         UINFO(4, " COUNTBITS(" << dropop[0] << dropop[1] << dropop[2] << " " << nodep << endl);
-        const AstVarRef* const varrefp = VN_AS(nodep->lhsp(), VarRef);  // Input variable
         if (m_graphing) {
             iterateAndNextNull(nodep->lhsp());
             if (!dropop[0]) iterateAndNextNull(nodep->rhsp());
@@ -1459,6 +1458,7 @@ class TristateVisitor final : public TristateBaseVisitor {
                                                      << nodep->prettyTypeName());
                     return;
                 }
+                const AstVarRef* const varrefp = VN_AS(nodep->lhsp(), VarRef);  // Input variable
                 AstVar* const envarp = getCreateEnVarp(varrefp->varp(), false);
                 // If any drops, we need to add in the count of Zs (from __en)
                 UINFO(4, " COUNTBITS('z)-> " << nodep << endl);
-- 
2.46.2
```

wsnyder, 2024-10-27:
Thanks!  Let's push the real tristates down the road, @hankhsu1996 feel free to reopen if this doesn't solve your larger test case.

closed 2024-10-27 by wsnyder

shoulihsu, 2024-10-27:
It fixed my test case! Thanks!

hankhsu1996, 2024-10-27:
It fixed my test case! Thanks!