verilator/verilator #5567: Internal error with FST + multidriven conditional operators

sifferman, 2024-10-26:
```
Ubuntu 22.04.4 LTS
Verilator 5.029 devel rev v5.028-183-gb873c23cf
```
Hello! One of my students discovered a weird internal fault. I converted it to a minimal test case:

```systemverilog
// top.sv
module top (
    input  logic a2_i, a1_i, a0_i
);
    m m(
        .a_i({a2_i, a1_i, a0_i})
    );
endmodule

module m (
    input  logic [2:0] a_i,
    input  logic b_i,
    output logic c_o,
    output logic d_o
);
    assign c_o = (a_i != 0) ? 1 : 0;

    assign d_o =
        (c_o = 1) ? 1 :
        (c_o = 0) ? b_i : 0;
endmodule
```

```verilog-filelist
// verilator.f

top.sv
--lint-only
--trace-fst

-Wno-PINMISSING // not needed to generate fault, but simplifies the output
```
```log
verilator -f verilator.f
%Error: Verilator internal fault, sorry. Suggest trying --debug --gdbbt
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec verilator_bin -f verilator.f
```
I believe expected behavior would be to generate no error or warning.

Although, maybe it would be nice to have a warning for `=` in common places of `==`? Similar to Visual C++: <https://learn.microsoft.com/en-us/cpp/error-messages/compiler-warnings/compiler-warning-level-4-c4706?view=msvc-170>. I could make a separate issue if you think it's worth pursuing.

wsnyder, 2024-10-27:
Driver error on my side.

The problem is $countbits accepts counting X/Zs so needs to be processed in V3Tristate, and currently only takes a variable.  For cases where it is only counting 0/1s it could be easily handled as a non-tristate case by converting earlier.  As to handling the Sampled itself, I'm not sure if we can just add a Sampled to the tristate variable - @kozdra?

wsnyder, 2024-10-27:
Ignore - wrong bug.

wsnyder, 2024-10-27:
FWIW I always thought allowing '=' in general expressions was a bad idea in C and seemed unfortunate to propagate to SystemVerilog.

When I use "always_comb" I get a multidriver error, so would think that perhaps should fire here too. I'm fine additionally with a new general warning on '=' inside any expression - please file a bug and consider a pull.

As to this issue V3Life is confused, I tried disabling the optimization but that turned off too much, so needs more work.  Not obvious how to make an assertion as it's a dangling pointer.  I would need to get back to it - alternatively a pull request to fix it would be appreciated.

wsnyder, 2025-09-14:
The error itself passes on master as of some time ago, just added warning on this use of `=` when using `-Wall` or similar.

closed 2025-09-14 by wsnyder