verilator/verilator #5570: Compile error with `with` clause in cross-module function call

hankhsu1996, 2024-10-28:
**Issue Description:**
Verilator throws a compilation error when calling a function with a cross-module reference that includes a `$fatal` statement. The error only occurs when `$fatal` is present; otherwise, the code compiles without issues.

**Example Code:**
```systemverilog
module DUT;
  function automatic int find_elem(int elem, int array[]);
    int index;

    index = array.find_first_index(x) with (x == elem);
    if (index == -1) begin
      $fatal(1, "Element not found in array");
    end

    return index;
  endfunction
endmodule

module Testbench;
  DUT dut();

  initial begin
    int array[] = '{5, 4, 3, 2, 1};
    int index = dut.find_elem(2, array);
  end
endmodule
```

**Verilator Command:**
```bash
verilator test.sv --cc --exe --binary --main
```

**Error Message:**
```
In file included from Vtest__ALL.cpp:8:
Vtest___024root__DepSet_h9b2d61ff__0__Slow.cpp: In function ‘void Vtest___024root___eval_initial__TOP(Vtest___024root*)’:
Vtest___024root__DepSet_h9b2d61ff__0__Slow.cpp:31:81: error: cannot convert ‘VlQueue<unsigned int>’ to ‘IData’ {aka ‘unsigned int’} in assignment
   31 |         = __Vfunc_Testbench__DOT__dut__DOT__find_elem__0__array.find_first_index(
      |           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
      |                                                                                 |
      |                                                                                 VlQueue<unsigned int>
   32 |                                                                                 [&](
      |                                                                                 ~~~~
   33 |                                                                                 IData/*31:0*/ x__DOT__index,
      |                                                                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   34 |                                                                                 IData/*31:0*/ x) {
      |                                                                                 ~~~~~~~~~~~~~~~~~~
   35 |             return ((2U == x));
      |             ~~~~~~~~~~~~~~~~~~~                                                  
   36 |         }
      |         ~                                                                        
   37 |     );
      |     ~ 
```

**Verilator Version:**
Verilator 5.031 (devel) rev v5.030-1-g68e0cf552 (git master)

**OS and Distribution:**
WSL Ubuntu 22.04.4

**Additional Notes:**

- The error only occurs when the function is called across modules.
- Removing `$fatal` eliminates the issue.

wsnyder, 2024-10-29:
This code is illegal.  Verilator needs to add an error you are comparing a queue with an int.

If the `$error` is removed, the `if` gets optimized away so you don't get the compile error (but should still get the new lint error proposed.)

hankhsu1996, 2024-10-30:
Thank you -- I didn't notice the return type of `find_first_index` is still a queue.

PR #5950 (2025-05-20): Fix crash when comparing queue to scalar (#5570)
## Summary

Fixes #5570.

This PR adds proper validation to reject comparisons between incompatible aggregate types (e.g. queue vs scalar). Previously, Verilator would generate invalid C++ code and crash during compilation, making it hard to debug.

## What changed

- Introduced `areAggregatesComparable` to recursively verify that aggregate types are structurally comparable.
- Updated `visit_cmp_eq_gt` to call this check and emit an early error with useful context.
- Removed now-redundant legacy slicing checks in `expandBiOp`. With the new valid [... cut]
closed 2025-05-20 by wsnyder