verilator/verilator #5588: Verilator does not accept module port lists with .port_identifier

ldoolitt, 2024-11-05:
IEEE Std 1354-2001 Version C gives the following BNF:
```
port ::=
  [ port expression ]
  | .port_identifier([port_expression])
```
And p.187 includes the example
```
module same_port (.a(i), .b(i)); // Name 'i' is declared inside the
                    // module as a inout port. Names 'a' and 'b' are
                    // defined for port connections.
inout i;
wire i;
endmodule
```
where I've added the last three lines to complete the story.
Feeding that to verilator v5.030 yields
```
$ verilator --lint-only same_port.v
%Error: same_port.v:1:19: syntax error, unexpected '.', expecting '['
    1 | module same_port (.a(i), .b(i));
      |                   ^
%Error: same_port.v:6:1: syntax error, unexpected endmodule
    4 | endmodule
      | ^~~~~~~~~
%Error: Exiting due to 2 error(s)
```
Vivado and iverilog both handle this case correctly.

ldoolitt, 2024-11-05:
Yosys also fails this test.
See its issue 4708 https://github.com/YosysHQ/yosys/issues/4708

wsnyder, 2024-11-05:
This is a little-used language feature that I would strongly recommend not relying on.  As to adding it, we already have an issue, see #2844.

closed 2024-11-05 by wsnyder

ldoolitt, 2024-11-05:
OK.  Thanks for the link to issue 2844.
I did look for previous submissions, but clearly didn't use the right keywords.  I'll post additional comments there.
I'd argue that "not planned", while literally correct today, is misleading.  It's totally valid to close this one because it's a duplicate.
Issue 2844 is still open.

wsnyder, 2024-11-05:
As to not planned, github literally says next to that drop down "duplicate" ;)