verilator/verilator #5590: PeakRDL generated registers for AXI Lite syntax broke between 5.028 -> 5.031

drewranck, 2024-11-07:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

--------------------------
Code snippet:
```
    // AXI4-Lite Response Logic
    struct {
        logic is_wr;
        logic err;
        logic [31:0] rdata;
    } axil_resp_buffer[2];

    logic [1:0] axil_resp_wptr;
    logic [1:0] axil_resp_rptr;

    always_ff @(posedge clk) begin
        if(rst) begin
            for(int i=0; i<2; i++) begin
                axil_resp_buffer[i].is_wr <= '0;
                axil_resp_buffer[i].err <= '0;
                axil_resp_buffer[i].rdata <= '0;
            end

```
```  
%Error: Internal Error: ............... foo_csrs.sv:166:37: ../V3Delayed.cpp:344: Malformed LHS in NBA
  166 |                 axil_resp_buffer[i].is_wr <= '0;
      |                                     ^~~~~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```
--------------------------

What 'verilator' command line do we use to run your example?
It's a pretty long command, I will try to distill a smaller example that can (eventually) go in test_regress

What 'verilator --version' are you using?  Did you try it with the git master version?
Broken on 5.031, Works on 5.028. I'm sadly not able to build the tagged 5.030 on my machine to narrow it down.

What OS and distribution are you using?
Ubuntu 22

May we assist you in trying to fix this in Verilator yourself?
I'll try to figure it out once I get a smaller example. Likely can also be worked around if I bring-my-own templates to PeakRDL, or post-process the code they generate, but this syntax didn't really seem LRM illegal, just probably not common in Verilator test cases.

(Please avoid attaching screenshots that show text - you can convert images to text using e.g. https://ocr.space)

wsnyder, 2024-11-07:
Awaiting test case.

gezalore, 2024-11-07:
This is due to my recent refactor of V3Delayed, probably worked by accident before, using a shadow variable for the whole array, which of course can be extremely slow.

drewranck, 2024-11-08:
Confirmed that v5.030 is also affected.

PR #5603 (2024-11-10): Fix NBAs to unpacked arrays of unpacked structs (#5590)
This happened to work before #5516, by creating a whole shadow copy of the entire array. Revert back to that behaviour for now, it will be slow, but works still.

Fixes #5590
commit 863abdb1 (2024-11-10): Fix NBAs to unpacked arrays of unpacked structs (#5603)

This happened to work before #5516, by creating a whole shadow copy of
the entire array. Revert back to that behaviour for now, it will be
slow, but works still.

Fixes #5590
closed 2024-11-10 by gezalore