verilator/verilator #5593: Use of localparams in array initialization

gregdavill, 2024-11-08:
I'm trying to have verilator compile an SoC project, the AHB/AXI buses make use of the following pattern to assign interfaces to crossbars/arbiters, etc.

> Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

```verilog
module test;
  localparam int unsigned SPI_AHB_INDEX = 0;
  localparam int unsigned I2C_AHB_INDEX = 1;

  localparam bit AHB_USE_IO_BASE[2] = '{
    SPI_AHB_INDEX: 1'b0,
    I2C_AHB_INDEX: 1'b0};

  localparam logic [31:0] AHB_ADDR[2] = '{
    SPI_AHB_INDEX: 32'h8001000,
    I2C_AHB_INDEX: 32'h8002000};

endmodule
```

> What 'verilator' command line do we use to run your example?

```
$ verilator --lint-only test.sv
%Error: test.sv:6:5: Assignment pattern key not supported/understood: TEXT
                   : ... note: In instance 'test'
    6 |     SPI_AHB_INDEX: 1'b0,
      |     ^~~~~~~~~~~~~
%Error: test.sv:7:5: Assignment pattern key not supported/understood: TEXT
                   : ... note: In instance 'test'
    7 |     I2C_AHB_INDEX: 1'b0};
      |     ^~~~~~~~~~~~~
%Error: test.sv:10:5: Assignment pattern key not supported/understood: TEXT
                    : ... note: In instance 'test'
   10 |     SPI_AHB_INDEX: 32'h8001000,
      |     ^~~~~~~~~~~~~
%Error: test.sv:11:5: Assignment pattern key not supported/understood: TEXT
                    : ... note: In instance 'test'
   11 |     I2C_AHB_INDEX: 32'h8002000};
      |     ^~~~~~~~~~~~~
%Error: Exiting due to 4 error(s)
```

> What 'verilator --version' are you using?  Did you try it with the git master version?

```
$ verilator --version
Verilator 5.031 devel rev v5.030-23-g2e1128b41
```

> What OS and distribution are you using?

Ubuntu 24.04 (via wsl)

> May we assist you in trying to fix this in Verilator yourself?

Yes, I see the error is being generated in V3Width:patVectorMap
At this stage the localparams appear as a Text object, And this function is only handling Const objects. 
I haven't dug much deeper into the code yet.

gregdavill, 2024-11-08:
I do see on the language limitations this is listed as not supported.

https://verilator.org/guide/latest/languages.html#language-limitations
> ‘{} operator
> Assignment patterns with an order based, default, constant integer (array) or member identifier (struct/union) keys are supported. Data type keys and keys computed from a constant expression are not supported.

So please change this from a bug to a feature request.

wsnyder, 2024-11-08:
It's a Text node because until the data types are known, we don't know if "FOO" is a member name, or value (for this array), or otherwise.  

I think the fix is to have AstPatMember have a child that is the decoded text's nodep .  V3LinkDot would convert the Text appropriately and put it in there, and not signal an error if it is not legal, plus keep existing Text.  That lets us defer deciding what the parameter key means.  So in this case that new node would in V3LinkDot become a AstVarRef to the parameter, then V3Param would make it a AstConst, and V3Width can use the const.

Can you attempt a patch?

gregdavill, 2024-11-09:
Thanks for outlining the general approach! I've written a patch, and a testcase that fails on the current head, and this patch fixes. 
I deviated a bit on the last points you made, I couldn't quite see how to add a new AstVarRef into the PatMember/tree so that it would be picked up by Param/Const Visitors.

wsnyder, 2024-11-14:
Thanks for your work on contributing this.