verilator/verilator #5595: Examples of converting from SV->AST->SV

rmcnultytt, 2024-11-08:
The documentation covers converting from SV to Json formatted AST form, but are there any references or examples that convert from the AST form back to SystemVerilog? This is likely outside the scope of Verilator itself, but didn't want to reinvent the wheel if there are already solutions/references that convert back and forth between SV and AST forms.

wsnyder, 2024-11-08:
V3EmitV.cpp creates Verilog from AST.  This is only legal Verilog in certain very limited cases needed for `--hierarchical`, there is no current need/intent to completely round trip back to Verilog.

MrWater98, 2025-02-20:
I just wanted to say that even though this is outside the scope of Verilator, I find this issue pretty interesting. If Verilog had some kind of unified intermediate format? It could make standardized editing a lot easier, especially for things like custom refactoring.