verilator/verilator #5601: An array of constant arrays not marked as constant

gregdavill, 2024-11-09:
> Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
```verilog
module array_2d ();

  localparam bit MATRIX[2][2] = '{'{1'b1, 1'b0}, '{1'b1, 1'b1}};

  genvar i, j;
  generate
    for (i = 0; i < 1; i++)
      for (j = 0; j < 1; j++) begin
        if (MATRIX[i][j]) begin : gen_matrix
        end
      end
  endgenerate

endmodule;
```

> What 'verilator' command line do we use to run your example?

```
$ verilator_bin --lint-only test_2d.sv
%Error: test_2d.sv:9:13: Expecting expression to be constant, but variable isn't const: 'MATRIX'
                       : ... note: In instance 'array_2d'
    9 |         if (MATRIX[i][j]) begin : gen_matrix
      |             ^~~~~~
%Error: test_2d.sv:9:22: Generate If condition must evaluate to constant
                       : ... note: In instance 'array_2d'
    9 |         if (MATRIX[i][j]) begin : gen_matrix
      |                      ^
%Error: Exiting due to 2 error(s)
```

> What 'verilator --version' are you using?  Did you try it with the git master version?

```
$ verilator --version
Verilator 5.031 devel rev v5.030-23-g2e1128b41
```

> What OS and distribution are you using?

Ubuntu 24.04

> May we assist you in trying to fix this in Verilator yourself?

Yes, I've taken a look through the V3Const.cpp:2766 and noticed this, so it's maybe a known issue?
```c
                } else if (m_params && VN_IS(valuep, InitArray)) {
                    // Allow parameters to pass arrays
                    // Earlier recursion of InitArray made sure each array value is constant
                    // This exception is fairly fragile, i.e. doesn't
                    // support arrays of arrays or other stuff
                    AstNode* const newp = valuep->cloneTree(false);
```

wsnyder, 2024-11-09:
It's known but I don't see a bug requesting this get supported.  Might you give it an attempt to fix after your current one is done?

gregdavill, 2024-11-10:
Sure. I can give it a go.

Do you have any architectural tips, of where this would be done?

wsnyder, 2024-11-10:
Check that with --debug the .tree file looks correct for the parameter.  Assuming it looks right, we would need to improve only V3Simulate to understand how to store and process the array-in-array.

gregdavill, 2024-11-10:
I think I've traced down the issue

With this example module
```verilog
module array_2d ();
  localparam bit MATRIX[2][2] = '{'{1'b1, 1'b0}, '{1'b1, 1'b1}};

  genvar i, j;
  for (i = 0; i < 2; i++)
    for (j = 0; j < 2; j++)
      initial $display("%0d", MATRIX[i][j]);

endmodule
```

The initial tree is correct:
```
    1:2:2:2: INITIAL 0x5dc5bf923bc0 {d7ah}
    1:2:2:2:2: DISPLAY 0x5dc5bf9239d0 {d7ap}
    1:2:2:2:2:1: SFORMATF 0x5dc5bf9376e0 {d7ap} @dt=0x5dc5bf92f3f0@(G/str)  %0d
    1:2:2:2:2:1:1: SELBIT 0x5dc5bf9233c0 {d7bo} @dt=0@
    1:2:2:2:2:1:1:1: SELBIT 0x5dc5bf9231d0 {d7bl} @dt=0@
    1:2:2:2:2:1:1:1:1: VARREF 0x5dc5bf928f80 {d7bf} @dt=0@  MATRIX [RV] <- VAR 0x5dc5bf935a40 {d2as} @dt=0x5dc5bf915660@(w1)u[0:1]  MATRIX [VSTATIC]  LPARAM
    1:2:2:2:2:1:1:1:2: CONST 0x5dc5bf9214f0 {d5ad} @dt=0x5dc5bf915d00@(G/swu32/1)  ?32?sh0
    1:2:2:2:2:1:1:2: CONST 0x5dc5bf927820 {d6af} @dt=0x5dc5bf915d00@(G/swu32/1)  ?32?sh1
```

The SELBIT nodes are transformed to ARRAYSEL in the linkdotparam step, and duplicated per the generate loops
```
    1:2:2:2: INITIAL 0x5dc5bf91e170 {d7ah}
    1:2:2:2:2: DISPLAY 0x5dc5bf91d540 {d7ap}
    1:2:2:2:2:1: SFORMATF 0x5dc5bf92e910 {d7ap} @dt=0x5dc5bf92f3f0@(G/str)  %0d
    1:2:2:2:2:1:1: ARRAYSEL 0x5dc5bf9205c0 {d7bo} @dt=0x5dc5bf934870@(G/w1)
    1:2:2:2:2:1:1:1: ARRAYSEL 0x5dc5bf938230 {d7bl} @dt=0x5dc5bf91d720@(w1)u[0:1]
    1:2:2:2:2:1:1:1:1: VARREF 0x5dc5bf937900 {d7bf} @dt=0x5dc5bf915660@(w1)u[0:1]  MATRIX [RV] <- VAR 0x5dc5bf935a40 {d2as} @dt=0x5dc5bf915660@(w1)u[0:1]  MATRIX [VSTATIC]  LPARAM
    1:2:2:2:2:1:1:1:2: CONST 0x5dc5bf934a60 {d5ad} @dt=0x5dc5bf9131c0@(G/sw1)  1'h0
    1:2:2:2:2:1:1:2: CONST 0x5dc5bf926980 {d6af} @dt=0x5dc5bf9131c0@(G/sw1)  1'h0
```

The array init looks good too:
```
    1:2: VAR 0x5dc5bf935a40 {d2as} @dt=0x5dc5bf915660@(w1)u[0:1]  MATRIX [VSTATIC]  LPARAM
    1:2:3: INITARRAY 0x5dc5bf936500 {d2bh} @dt=0x5dc5bf915660@(w1)u[0:1] [0]=(B) [1]=(C)
    1:2:3:2: INITITEM 0x5dc5bf90c9e0 {d2bh}
    1:2:3:2:1: INITARRAY 0x5dc5bf936c40 {d2bj} @dt=0x5dc5bf91d720@(w1)u[0:1] [0]=(D) [1]=(E)
    1:2:3:2:1:2: INITITEM 0x5dc5bf91ce90 {d2bj}
    1:2:3:2:1:2:1: CONST 0x5dc5bf934d00 {d2bl} @dt=0x5dc5bf91b350@(G/w1)  1'h1
    1:2:3:2:1:2: INITITEM 0x5dc5bf91b620 {d2bj}
    1:2:3:2:1:2:1: CONST 0x5dc5bf934ec0 {d2br} @dt=0x5dc5bf91b350@(G/w1)  1'h0
    1:2:3:2: INITITEM 0x5dc5bf91f080 {d2bh}
    1:2:3:2:1: INITARRAY 0x5dc5bf936160 {d2by} @dt=0x5dc5bf91d720@(w1)u[0:1] [0]=(F) [1]=(G)
    1:2:3:2:1:2: INITITEM 0x5dc5bf910dc0 {d2by}
    1:2:3:2:1:2:1: CONST 0x5dc5bf9352d0 {d2ca} @dt=0x5dc5bf91b350@(G/w1)  1'h1
    1:2:3:2:1:2: INITITEM 0x5dc5bf938cf0 {d2by}
    1:2:3:2:1:2:1: CONST 0x5dc5bf9354d0 {d2cg} @dt=0x5dc5bf91b350@(G/w1)  1'h1
```

When we go to resolve the ARRAYSEL into a constant V3Const's : ConstVisitor(AstNodeVarRef) the logic isn't handling the nested case. 
I wrote a quick patch that when converting a VarRef to const if nested, iterate over multiple INITARRAYS per dimension. which seems to handle my failing example, I don't think V3Simulate needs any adjustment for this.

closed 2025-03-09 by wsnyder