verilator/verilator #5604: Looks like tracing_off does not work with Hierarchy blocks?

MTXDavid, 2024-11-12:
When i trying to use tracing_off to reduce vcd size, it does not work.
I just want to hierarchy alu block in .vlt file and reduce vcd file size.

hier_block -module "alu"
tracing_off -module "alu.v"

Theare are two modules in the DUT, alu and led, the corresponding verilog file - alu.v, led.v

module top(
  input clk,
  input rst_n,
  input en,
  output reg [15:0] led,

  input  operation_t  op_in,
  input  [WIDTH-1:0]  a_in,
  input  [WIDTH-1:0]  b_in,
  input               in_valid,

  output logic [WIDTH-1:0]  out,
  output logic              out_valid  
  
  alu u_alu(clk, .rst_n, .op_in, .a_in, .a_in, .in_valid, .out, .out_valid);
  led u_led(.clk, .rst_n, .en, .led);
);
endmodule

module alu #(
        parameter WIDTH = 6
) (
        input clk,
        input rst_n,

        input  operation_t  op_in,
        input  [WIDTH-1:0]  a_in,
        input  [WIDTH-1:0]  b_in,
        input               in_valid,

        output logic [WIDTH-1:0]  out,
        output logic              out_valid
);

        operation_t  op_in_r;
        logic  [WIDTH-1:0]  a_in_r;
        logic  [WIDTH-1:0]  b_in_r;
        logic               in_valid_r;
        logic  [WIDTH-1:0]  result;

        // Register all inputs
        always_ff @ (posedge clk, posedge rst_n) begin
                if (rst_n) begin
                        op_in_r     <= nop;      // 这里需要改称nop，否则会报Error
                        a_in_r      <= '0;
                        b_in_r      <= '0;
                        in_valid_r  <= '0;
                end else begin
                        op_in_r    <= op_in;
                        a_in_r     <= a_in;
                        b_in_r     <= b_in;
                        in_valid_r <= in_valid;
                end
        end

        // Compute the result
        always_comb begin
                result = '0;
                if (in_valid_r) begin
                        case (op_in_r)
                                add: result = a_in_r + b_in_r;
                                sub: result = a_in_r + (~b_in_r+1'b1);
                                default: result = '0;
                        endcase
                end
        end

        // Register outputs
        always_ff @ (posedge clk, posedge rst_n) begin
                if (rst_n) begin
                        out       <= '0;
                        out_valid <= '0;
                end else begin
                        out       <= result;
                        out_valid <= in_valid_r;
                end
        end

endmodule;

module led(
  input clk,
  input rst_n,
  input en,
  output reg [15:0] led
);
  reg [31:0] count;
  always @(posedge clk) begin
    if (!rst_n) begin 
        led <= 16'b0000_0000_0000_0001;
    end
    else if (count == 5) begin
        led <= {led[14:0], led[15]};
    end
  end

  always @(posedge clk) begin
    if (!rst_n) begin 
        count <= 0; 
    end
    else if(count >= 5) begin
        count <= 32'b0;
    end
    else if(en) begin
        count <= count + 1;
    end
  end

endmodule

wsnyder, 2024-11-12:
Your example is not valid verilog, so I cannot test it.  I suspect you are using an older version and putting the .vlt control file after the .v files on the command line?  Either put it first, or use the most recent version.

MTXDavid, 2024-11-13:
Do like below?
/bin/verilator --top-module top --trace tracing_off.vlt?

MTXDavid, 2024-11-13:
> Your example is not valid verilog, so I cannot test it. I suspect you are using an older version and putting the .vlt control file after the .v files on the command line? Either put it first, or use the most recent version.

Just like below?
/bin/verilator --top-module top --trace tracing_off.vlt?

closed 2024-11-26 by wsnyder