verilator/verilator #5610: lint_off not turning off warnings in some file

kareefardi, 2024-11-14:
Verilator version:

```
Verilator 5.031 devel rev v5.030-71-gd4a8cbb1d
```

Test case: [test-case.tar.gz](https://github.com/user-attachments/files/17756938/test-case.tar.gz)

Running this command, Verilator still reports warnings in the lint_off files  
```
verilator -Wall --lint-only config.vlt EF_SRAM_1024x32_stub.v EF_SRAM_1024x32.v
```

config.vlt:
```
`verilator_config

lint_off -file "*stub*"
lint_off -file "*EF_SRAM_1024x32.v"
```

More interesting if I used `lint_off -file "*"`, all the warnings are ignored except for one. 
```
%Warning-TIMESCALEMOD: EF_SRAM_1024x32_stub.v:28:8: Timescale missing on this module as other modules have it (IEEE 1800-2023 3.14.2.3)
   28 | module EF_SRAM_1024x32_macro (DO, ScanOutCC, AD, BEN, CLKin, DI, EN, R_WB, ScanInCC, ScanInDL, ScanInDR, SM, TM, WLBI, WLOFF, vgnd, vnb, vpb, vpwra, vpwrac, vpwrm, vpwrp, vpwrpc);
      |        ^~~~~~~~~~~~~~~~~~~~~
                       EF_SRAM_1024x32.v:35:8: ... Location of module with timescale
   35 | module EF_SRAM_1024x32 (DO, ScanOutCC, AD, BEN, CLKin, DI, EN, R_WB, ScanInCC, ScanInDL, ScanInDR, SM, TM, WLBI, WLOFF,
      |        ^~~~~~~~~~~~~~~
                       ... For warning description see https://verilator.org/warn/TIMESCALEMOD?v=5.031
                       ... Use "/* verilator lint_off TIMESCALEMOD */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
```

Possibly, wrong usage at my end. Still would like to know what would the correct usage be.

wsnyder, 2024-11-25:
See also #5368

paul-demo, 2025-05-29:
I recently tried upgrading from 5.024 to 5.036 and also encountered that .vlt waivers which worked in 5.024 are not working in 5.036. Specifically things like this for missing modules now do not successfully waive the lint warnings in 5.036:

```{text}
`verilator config
lint_off -file "*.sv" -match "Cannot find file containing module: 'some_module_name'"
```

paul-demo, 2025-05-29:
It seems the code which decides "isError" changed in 5.028, possibly introducing unintended bugs/changes at that time?

https://github.com/verilator/verilator/commit/1b15509a9ca6c448e9e83d6046664b5c5c79accb
https://github.com/verilator/verilator/issues/5351
https://github.com/verilator/verilator/pull/5363