verilator/verilator #5613: Installation issue "/bin/sh: 1: cd: can't cd to ./bin"

PedroAntunes178, 2024-11-15:
Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
```
------------------------------------------------------------
making verilator in src
make -C src 
make[1]: Entering directory '/home/pantunes/verilator/src'
make -C obj_dbg -j 1  TGT=../../bin/verilator_bin_dbg VL_DEBUG=1 -f ../Makefile_obj serial
make[2]: Entering directory '/home/pantunes/verilator/src/obj_dbg'
make[2]: Nothing to be done for 'serial'.
make[2]: Leaving directory '/home/pantunes/verilator/src/obj_dbg'
make -C obj_dbg       TGT=../../bin/verilator_bin_dbg VL_DEBUG=1 -f ../Makefile_obj
make[2]: Entering directory '/home/pantunes/verilator/src/obj_dbg'
      Compile flags:  g++ -Og -ggdb -gz -DVL_DEBUG -D_GLIBCXX_DEBUG -DVERILATOR_INTERNAL_ -MMD -I. -I.. -I.. -I../../include -I../../include -MP -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/opt/verilator/v5.030/share/verilator"
make[2]: Leaving directory '/home/pantunes/verilator/src/obj_dbg'
make -C obj_dbg       TGT=../../bin/verilator_coverage_bin_dbg VL_DEBUG=1 VL_VLCOV=1 -f ../Makefile_obj serial_vlcov
make[2]: Entering directory '/home/pantunes/verilator/src/obj_dbg'
make[2]: Nothing to be done for 'serial_vlcov'.
make[2]: Leaving directory '/home/pantunes/verilator/src/obj_dbg'
make -C obj_dbg       TGT=../../bin/verilator_coverage_bin_dbg VL_DEBUG=1 VL_VLCOV=1 -f ../Makefile_obj
make[2]: Entering directory '/home/pantunes/verilator/src/obj_dbg'
      Compile flags:  g++ -Og -ggdb -gz -DVL_DEBUG -D_GLIBCXX_DEBUG -DVERILATOR_INTERNAL_ -MMD -I. -I.. -I.. -I../../include -I../../include -MP -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/opt/verilator/v5.030/share/verilator"
make[2]: Leaving directory '/home/pantunes/verilator/src/obj_dbg'
make -C obj_opt -j 1  TGT=../../bin/verilator_bin -f ../Makefile_obj serial
make[2]: Entering directory '/home/pantunes/verilator/src/obj_opt'
make[2]: Nothing to be done for 'serial'.
make[2]: Leaving directory '/home/pantunes/verilator/src/obj_opt'
make -C obj_opt       TGT=../../bin/verilator_bin -f ../Makefile_obj
make[2]: Entering directory '/home/pantunes/verilator/src/obj_opt'
      Compile flags:  g++ -O3 -DVERILATOR_INTERNAL_ -MMD -I. -I.. -I.. -I../../include -I../../include -MP -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/opt/verilator/v5.030/share/verilator"
make[2]: Leaving directory '/home/pantunes/verilator/src/obj_opt'
make[1]: Leaving directory '/home/pantunes/verilator/src'
/bin/sh ./src/mkinstalldirs /opt/verilator/v5.030/share/verilator/bin
/bin/sh ./src/mkinstalldirs /opt/verilator/v5.030/bin
cd ./bin; \
for p in verilator verilator_coverage verilator_gantt verilator_profcfunc  ; do \
  /usr/bin/install -c $p /opt/verilator/v5.030/bin/$p; \
done
/bin/sh: 1: cd: can't cd to ./bin
/usr/bin/install: cannot stat 'verilator': No such file or directory
/usr/bin/install: cannot stat 'verilator_coverage': No such file or directory
/usr/bin/install: cannot stat 'verilator_gantt': No such file or directory
/usr/bin/install: cannot stat 'verilator_profcfunc': No such file or directory
make: *** [Makefile:267: installbin] Error 1
```

What 'verilator' command line do we use to run your example?
`sudo make install`

What 'verilator --version' are you using?  Did you try it with the git master version?
5.030

What OS and distribution are you using?
Ubuntu 22.04.4 LTS

May we assist you in trying to fix this in Verilator yourself?
Yes

Thank you for any help :)

PedroAntunes178, 2024-11-15:
Additional info:
`./configure --prefix=/opt/verilator/v5.030`

wsnyder, 2024-11-15:
Probably you need to mkdir "/opt/verilator/v5.030"

Anyhow, this is something wrong at your end as opposed to a Verilator issue, so closing, but feel free to follow up.

closed 2024-11-15 by wsnyder

PedroAntunes178, 2024-11-18:
I also considered it might be something on my end. I double-checked if I had miswritten the directory, but nah, the directory is there, and the bin and share folders are created by the install command. 

However, it might also be something on the Verilator Makefiles. Using relative paths, such as `./bin`, is not a good practice.
I'll try looking into the makefile (not sure when the time will allow) and seeing if I can write something like `$PWD/bin`.

I'll keep this issue updated if I manage to fix it.