verilator/verilator #5614: Variable index into queue array

Aidan-McNay, 2024-11-15:
(Verified on v5.028, both on MacOS and RHEL)

I'm having some issues when attempting to use a variable index (ex. in a for-loop) to push into a specific queue in the array. It appears as though pushing into a specific queue additionally pushes all previous elements from other queues in the array

Minimal Reproducible Example:
```verilog
module top;
  int queues [3][$];

  initial begin
    for( int i = 0; i < 3; i = i + 1) begin
      queues[i].push_back(i);
    end
    $display("%p", queues);
    $finish();
  end
endmodule
```
I'm compiling/running the above (saved as `tmp.v`) with the commands
```bash
verilator --binary tmp.v
./obj_dir/tmp
```
My expected output is 1 element per queue in the array:
```plain
'{'{'h0} , '{'h1} , '{'h2} } 
```
However, the actual output I'm getting is
```plain
'{'{'h0} , '{'h0, 'h1} , '{'h0, 'h1, 'h2} } 
```

Is there something I'm doing wrong, or could contribute to a fix? Thanks in advance!

wsnyder, 2024-11-16:
Please try this on the mainline version.

Aidan-McNay, 2024-11-16:
Whoops - that had the indented behavior 😅, thanks for the help!

Aidan-McNay, 2025-02-08:
Reopening - for some reason, this issue has re-surfaced when switching from 5.030 to 5.032. I discovered an additional, similar bug, where queues in arrays can be "stuck" after using another one (maybe due to the "current" value of the variable index?). Not quite sure how to best approach

```verilog
module top;
  int queues [3][$];

  initial begin
    for( int i = 0; i < 3; i = i + 1 ) begin
      queues[i].push_back(7);
      $display("Queue With Element: %p", queues[0]);

      queues[i].pop_front();
      $display("Queue Without Element: %p", queues[0]);
    end
    $finish();
  end
endmodule
```
```plain
Queue With Element: '{'h7} 
Queue Without Element: '{}
Queue With Element: '{}
Queue Without Element: '{}
Queue With Element: '{}
Queue Without Element: '{}
```

Aidan-McNay, 2025-02-08:
Hmm - looks like this is the main cause of the issue? (Specifically the difference in `body`)

### `Vtmp___024root__DepSet_h669aa0fa__0__Slow.cpp` - 5.032 (incorrect)
```c++
VL_ATTR_COLD void Vtmp___024root___eval_initial__TOP(Vtmp___024root* vlSelf) {
    (void)vlSelf;  // Prevent unused variable warning
    Vtmp__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vtmp___024root___eval_initial__TOP\n"); );
    auto& vlSelfRef = std::ref(*vlSelf).get();
    // Init
    std::string __Vtemp_1;
    // Body
    vlSelfRef.top__DOT____Vlvbound_hca9b178d__0.push_back(0U);
    vlSelfRef.top__DOT__queues[0U] = vlSelfRef.top__DOT____Vlvbound_hca9b178d__0;
    vlSelfRef.top__DOT____Vlvbound_hca9b178d__0.push_back(1U);
    vlSelfRef.top__DOT__queues[1U] = vlSelfRef.top__DOT____Vlvbound_hca9b178d__0;
    vlSelfRef.top__DOT____Vlvbound_hca9b178d__0.push_back(2U);
    vlSelfRef.top__DOT__queues[2U] = vlSelfRef.top__DOT____Vlvbound_hca9b178d__0;
    __Vtemp_1 = VL_TO_STRING(vlSelfRef.top__DOT__queues);
    VL_WRITEF_NX("%@\n",0,-1,&(__Vtemp_1));
    VL_FINISH_MT("tmp.v", 9, "");
}
```

### `Vtmp___024root__DepSet_h669aa0fa__0__Slow.cpp` - 5.030 (correct)
```c++
VL_ATTR_COLD void Vtmp___024root___eval_initial__TOP(Vtmp___024root* vlSelf) {
    (void)vlSelf;  // Prevent unused variable warning
    Vtmp__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vtmp___024root___eval_initial__TOP\n"); );
    auto &vlSelfRef = std::ref(*vlSelf).get();
    // Init
    VlQueue<IData/*31:0*/> top__DOT____Vlvbound_hee7f1cc2__0;
    top__DOT____Vlvbound_hee7f1cc2__0.atDefault() = 0;
    std::string __Vtemp_1;
    // Body
    top__DOT____Vlvbound_hee7f1cc2__0 = vlSelfRef.top__DOT__queues
        [0U];
    top__DOT____Vlvbound_hee7f1cc2__0.push_back(0U);
    vlSelfRef.top__DOT__queues[0U] = top__DOT____Vlvbound_hee7f1cc2__0;
    top__DOT____Vlvbound_hee7f1cc2__0 = vlSelfRef.top__DOT__queues
        [1U];
    top__DOT____Vlvbound_hee7f1cc2__0.push_back(1U);
    vlSelfRef.top__DOT__queues[1U] = top__DOT____Vlvbound_hee7f1cc2__0;
    top__DOT____Vlvbound_hee7f1cc2__0 = vlSelfRef.top__DOT__queues
        [2U];
    top__DOT____Vlvbound_hee7f1cc2__0.push_back(2U);
    vlSelfRef.top__DOT__queues[2U] = top__DOT____Vlvbound_hee7f1cc2__0;
    __Vtemp_1 = VL_TO_STRING(vlSelfRef.top__DOT__queues);
    VL_WRITEF_NX("%@\n",0,-1,&(__Vtemp_1));
    VL_FINISH_MT("tmp.v", 9, "");
}
```

wsnyder, 2025-02-24:
@Aidan-McNay I still see this passing on master, please post the test case causing your issue.

wsnyder, 2025-02-24:
(and check you are using the latest master github branch, thanks)

closed 2025-03-16 by wsnyder