verilator/verilator #5615: Invalid result of Combinational Circuit

ngminhthanh12a3, 2024-11-16:
I have an issue with designing and simulating a Control module of a Single Cycle RISC-V.

Outputs must change due to the opcode input. However, all signals still remain unchanged when they reach high, following the bellow figure. This is the result I've got on using Verilator.

[image]

On using Quartus, all signals act normally.

[image]

My verilog code and cpp testbench for this module is as follows:

```v
module Control (
    input [6:0] opcode,
    output reg branch,
    output reg memRead,
    output reg memtoReg,
    output reg [1:0] ALUOp,
    output reg memWrite,
    output reg ALUSrc,
    output reg regWrite
);

    // TODO: implement your Control here
    // Hint: follow the Architecture (figure in spec) to set output signal
    parameter   OP      = 7'b0110011,
                OP_IMM  = 7'b0010011,
                LOAD    = 7'b0000011,
                STORE   = 7'b0100011,
                BRANCH  = 7'b1100011;

    
    always @(*) begin
        case (opcode)
            OP: begin
                branch = 1'b0;
                memRead = 1'b0;
                memtoReg = 1'b0;
                ALUOp = 2'b01;
                memWrite = 1'b0;
                ALUSrc = 1'b0;
                regWrite = 1'b1;
            end
            OP_IMM: begin
                branch = 1'b0;
                memRead = 1'b0;
                memtoReg = 1'b0;
                ALUOp = 2'b00;
                memWrite = 1'b0;
                ALUSrc = 1'b1;
                regWrite = 1'b1;
            end
            LOAD: begin
                branch = 1'b0;
                memRead = 1'b1;
                memtoReg = 1'b1;
                ALUOp = 2'b00;
                memWrite = 1'b0;
                ALUSrc = 1'b1;
                regWrite = 1'b1;
            end
            STORE: begin
                branch = 1'b0;
                memRead = 1'b0;
                memtoReg = 1'b0;
                ALUOp = 2'b01;
                memWrite = 1'b1;
                ALUSrc = 1'b1;
                regWrite = 1'b0;
            end
            BRANCH: begin
                branch = 1'b1;
                memRead = 1'b0;
                memtoReg = 1'b0;
                ALUOp = 2'b11;
                memWrite = 1'b0;
                ALUSrc = 1'b0;
                regWrite = 1'b0;
            end
            default: begin
                branch = 1'bz;
                memRead = 1'bz;
                memtoReg = 1'bz;
                ALUOp = 2'bzz;
                memWrite = 1'bz;
                ALUSrc = 1'bz;
                regWrite = 1'bz;
            end
        endcase
    end
endmodule
```

```cpp

#include <verilated.h>
#include <verilated_vcd_c.h>
#include <stdio.h>
#include <vector>
#include <time.h>
#include <cmath>
#include <iostream>
#include <VControl.h>
#include <assert.h>

using namespace std;

#define IS_SIM_TIME_IN_RST(sim_time) (sim_time >= 3 && sim_time < 6)
#define MAX_SIM_TIME 10
#define VERIF_START_TIME 7
vluint64_t sim_time = 0;
vluint64_t tx_data_gen_time = 0;

class ControlInTx
{
public:
    enum OPCODE
    {
        OP = 0b0110011,
        OP_IMM = 0b0010011,
        LOAD = 0b0000011,
        STORE = 0b0100011,
        BRANCH = 0b1100011
    } opcode;
};

const vluint8_t ref_opcode[] = {ControlInTx::OP, ControlInTx::OP_IMM, ControlInTx::LOAD, ControlInTx::STORE, ControlInTx::BRANCH};

typedef struct
{
    uint8_t branch : 1;
    uint8_t memRead : 1;
    uint8_t memtoReg : 1;
    uint8_t ALUOp : 2;
    uint8_t memWrite : 1;
    uint8_t ALUSrc : 1;
    uint8_t regWrite : 1;
} ControlOutBF;

ControlOutBF ref_control_out[] = {{0, 0, 0, 1, 0, 0, 1}, {0, 0, 0, 0, 0, 1, 1}, {0, 1, 1, 0, 0, 1, 1}, {0, 0, 0, 1, 1, 1, 0}, {1, 0, 0, 3, 0, 0, 0}};

class ControlOutTx
{
public:
    // TODO
    ControlOutBF controlOutBF;
};

class ControlScb
{
private:
    std::deque<ControlInTx *> in_q;

public:
    // Input interface monitor port
    void writeIn(ControlInTx *tx)
    {
        // Push the received transaction item into a queue for later
        in_q.push_back(tx);
    }

    // Output interface monitor port
    void writeOut(ControlOutTx *tx)
    {
        // We should never get any data from the output interface
        // before an input gets driven to the input interface
        if (in_q.empty())
        {
            std::cout << "Fatal Error in ControlScb: empty ControlInTx queue" << std::endl;
            exit(1);
        }

        // Grab the transaction item from the front of the input item queue
        ControlInTx *in;
        in = in_q.front();
        in_q.pop_front();
        std::cout << "OPCODE reg = " << (uint32_t)in->opcode << std::endl;
        std::cout << "All reg = " << (uint32_t) * ((uint8_t *)(&tx->controlOutBF)) << std::endl;

        switch (in->opcode)
        {
        case ControlInTx::OP:

            break;
        case ControlInTx::OP_IMM:
            break;
        case ControlInTx::LOAD:
            break;
        case ControlInTx::STORE:
            break;
        case ControlInTx::BRANCH:
            break;

        default:
            break;
        }

        delete in;
        delete tx;
    }
};

class ControlInDrv
{
private:
    VControl *dut;

public:
    ControlInDrv(VControl *dut)
    {
        this->dut = dut;
    }

    void drive(ControlInTx *tx)
    {
        dut->opcode = tx->opcode;

        delete tx;

        dut->eval();
    }
};

class ControlInMon
{
private:
    VControl *dut;
    ControlScb *scb;

public:
    ControlInMon(VControl *dut, ControlScb *scb)
    {
        this->dut = dut;
        this->scb = scb;
    }
    void monitor()
    {
        ControlInTx *tx = new ControlInTx();

        tx->opcode = ControlInTx::OPCODE(dut->opcode);

        scb->writeIn(tx);
    }
};

class ControlOutMon
{
private:
    VControl *dut;
    ControlScb *scb;

public:
    ControlOutMon(VControl *dut, ControlScb *scb)
    {
        this->dut = dut;
        this->scb = scb;
    }
    void monitor()
    {
        ControlOutTx *tx = new ControlOutTx();

        tx->controlOutBF.branch = dut->branch;
        tx->controlOutBF.memRead = dut->memRead;
        tx->controlOutBF.memtoReg = dut->memtoReg;
        tx->controlOutBF.ALUOp = dut->ALUOp;
        tx->controlOutBF.memWrite = dut->memWrite;
        tx->controlOutBF.ALUSrc = dut->ALUSrc;
        tx->controlOutBF.regWrite = dut->regWrite;

        scb->writeOut(tx);
    }
};

ControlInTx *rndAluInTx()
{
    ControlInTx *tx = new ControlInTx();

    tx->opcode = ControlInTx::OPCODE(ref_opcode[(tx_data_gen_time) % sizeof(ref_opcode)]);

    tx_data_gen_time += 1;
    return tx;
}

int main(int argc, char **argv)
{
    srand(time(NULL));
    Verilated::commandArgs(argc, argv);
    VControl *dut = new VControl;

    Verilated::traceEverOn(true);
    VerilatedVcdC *m_trace = new VerilatedVcdC;
    dut->trace(m_trace, 5);
    m_trace->open("waveform.vcd");

    ControlInTx *tx;

    // Here we create the driver, scoreboard, input and output monitor blocks
    ControlInDrv *drv = new ControlInDrv(dut);
    ControlScb *scb = new ControlScb();
    ControlInMon *inMon = new ControlInMon(dut, scb);
    ControlOutMon *outMon = new ControlOutMon(dut, scb);

    while (sim_time < MAX_SIM_TIME)
    {
        tx = rndAluInTx();
        // Generate a randomised transaction item of type AluInTx

        // Pass the transaction item to the ALU input interface driver,
        // which drives the input interface based on the info in the
        // transaction item
        drv->drive(tx);

        // Monitor the input interface
        inMon->monitor();

        // Monitor the output interface
        outMon->monitor();

        // end of positive edge processing

        m_trace->dump(sim_time);
        sim_time++;
    }

    m_trace->close();
    delete dut;
    delete outMon;
    delete inMon;
    delete scb;
    delete drv;
    exit(EXIT_SUCCESS);
    return 0;
}
```

# Verilator command

```bash
verilator -Wall -j 0 --trace --x-assign unique --x-initial unique -cc Control.v --exe tb_Control.cpp -Wno-ENUMVALUE
```

# Verilator version:

Verilator 5.028 2024-08-21 rev v5.028

# OS

Ubuntu 24.04.1 LTS

wsnyder, 2024-11-16:
Using z like this is unusual (presumably you wanted x's), and because Verilator is a mostly 2-state simulator it misinterprets this as a tristate driver circuit, and gets confused.  I'll look at a fix, but I strongly suspect your intent is not to be building tristate drivers, so use x's instead.

ngminhthanh12a3, 2024-11-17:
Thank you! I've tried using x's, and it works now!

```v
module Control (
    input [6:0] opcode,
    output reg branch,
    output reg memRead,
    output reg memtoReg,
    output reg [1:0] ALUOp,
    output reg memWrite,
    output reg ALUSrc,
    output reg regWrite
);

    always @(*) begin
        case (opcode)
            // ...
            default: begin
                branch = 1'bx;
                memRead = 1'bx;
                memtoReg = 1'bx;
                ALUOp = 2'bxx;
                memWrite = 1'bx;
                ALUSrc = 1'bx;
                regWrite = 1'bx;
            end
        endcase
    end
endmodule
```

closed 2024-11-17 by ngminhthanh12a3