verilator/verilator #5617: SystemVerilog "Expression Evaluator"

ChrisLuck, 2024-11-18:
Dear Verilator developers,

I am looking for a tool that can calculate the result of SystemVerilog expressions like for example `5'b1 + 5'b11`.

It can be achieved by using Verilator in this fashion of course:
~~~
module top;

logic [4:0] result = 5'b1 + 5'b11;   //problem: the resulting type needs to be known beforehand (width of LHS)

initial begin
    $display("Result: %b", result); $finish;
end

endmodule
~~~

So now I am wondering: Is there a more direct way to use Verilator for just this special "expression evaluation" purpose?
Basically some operation mode where the user can provide the right hand side of an assignment and Verilator would calculate the resulting value and type?

wsnyder, 2024-11-18:
Verilator is compiled so not a good fit for this. Try one of the parser packages:

https://github.com/dalance/sv-parser
https://github.com/chipsalliance/Surelog
https://github.com/MikePopoloski/slang

It might be Icarus Verilog, which has interpreted simulation, is also a possibility.

ChrisLuck, 2024-11-21:
Thank you so much for this answer. The [slang](https://github.com/MikePopoloski/slang) hint is pure gold, it does exactly what I was looking for!
The below code block shows how to solve the simple expression from my example above.

~~~python
# This example shows how to evaluate arbitrary SystemVerilog expressions using "slang",
# or more precisely "pyslang", its python wrapper.

import pyslang
session = pyslang.ScriptSession()
session.eval("logic [4:0] result = 5'b1 + 5'b11") # can be called multiple times to declare parameters for example
result = session.eval("result;")
print(result)
~~~

Result: `5'b100`