verilator/verilator #5623: Error: ‘vlProcess’ not declared in coroutine

hankhsu1996, 2024-11-22:
### Issue Description
The error only occurs under certain conditions, and unfortunately, I cannot share the code due to confidentiality. This makes it difficult to reproduce. However, I want to know if anyone has encountered this error before and knows how to resolve it.

Additionally, I noticed that this coroutine does not include vlProcess as an argument, whereas other coroutines do. I am unsure why this discrepancy exists.

Interestingly, the issue seems to depend on the size of the testbench. For example, my testbench currently runs 27 tests. If I comment out **any** of the tests, the issue no longer occurs. This suggests the problem might be triggered when the testbench reaches a certain size or complexity.

### Command Line Flags
```bash
verilator --top-module my_tb --cc --exe --build --main --Mdir /my_mdir --threads 1 --assert --timing --trace-fst -Wall -Wno-PINCONNECTEMPTY -Wno-UNUSEDSIGNAL -Wno-UNUSEDPARAM -Wno-DECLFILENAME -Wno-GENUNNAMED -Wno-MODDUP -Wno-ALWCOMBORDER -Wno-WIDTHEXPAND -Wno-UNOPTFLAT -Wno-BLKANDNBLK -Wno-MULTIDRIVEN -Wno-IMPURE -Wno-WIDTHCONCAT
```

### Verilator Version
Verilator 5.031 devel rev v5.030-79-g58dae8f93 (latest)

### Error Output
```
Vmy_tb_my_tb__DepSet_hf2d76328__0.cpp:1373:77: error: ‘vlProcess’ was not declared in this scope; did you mean ‘VlProcess’?
 1373 |                                                                             vlProcess,
      |                                                                             ^~~~~~~~~
      |                                                                             VlProcess
Vmy_tb_my_tb__DepSet_hf2d76328__0.cpp:1383:57: error: ‘vlProcess’ was not declared in this scope; did you mean ‘VlProcess’?
 1383 |                                                         vlProcess,
      |                                                         ^~~~~~~~~
      |                                                         VlProcess
Vmy_tb_my_tb__DepSet_hf2d76328__0.cpp:1703:5: error: ‘vlProcess’ was not declared in this scope; did you mean ‘VlProcess’?
 1703 |     vlProcess->state(VlProcess::FINISHED);
      |     ^~~~~~~~~
      |     VlProcess
Vmy_tb_my_tb__DepSet_hf2d76328__0.cpp:1705:1: warning: control reaches end of non-void function [-Wreturn-type]
 1705 | }
      | ^
make[1]: *** [/usr/local/share/verilator/include/verilated.mk:269: Vmy_tb_my_tb__DepSet_hf2d76328__0.o] Error 1
make[1]: Leaving directory 'my_dir'
%Error: make -C my_dir -f Vmy_tb.mk -j 1 exited with 2
```

### Additional Context
The coroutine in question:
```cpp
VlCoroutine Vmy_tb_my_tb___eval_initial__TOP__my_tb__Vtiming__1__1(Vmy_tb_my_tb* vlSelf) { 
    ...
    while (VL_LTS_III(32, 0U, __Vtask_advance_clock__642__unnamedblk1_1__DOT____Vrepeat0)) {
        co_await vlSymsp->TOP.__VtrigSched_h87bff3fb__0.trigger(
            0U, 
            vlProcess, 
            "@(posedge my_tb.clk)", 
            "my_dir/my_tb.sv", 
            131
        );
        vlSymsp->TOP.__Vm_traceActivity[0x96U] = 1U;
        __Vtask_advance_clock__642__unnamedblk1_1__DOT____Vrepeat0 = 
            (__Vtask_advance_clock__642__unnamedblk1_1__DOT____Vrepeat0 - (IData)(1U));
    }
}
```

I don’t know why this coroutine does not include vlProcess as an argument, while I see that many other coroutines do. Any insight into this would also be appreciated.

If you’ve encountered this error or have suggestions, I’d appreciate any guidance on how to resolve it.

hankhsu1996, 2024-11-22:
I tried adding `no_inline_task` to the `advance_clock` task, and surprisingly, it fixed the issue. Honestly, I’m not sure why it works, but thank you! 🙂

wsnyder, 2024-11-22:
We'd still like a test case to fix this, if you can make one.

hankhsu1996, 2024-11-23:
This issue only occurs when:

1. There is a fork (hinted by vlProcess).
2. The testbench is sufficiently large.
To reproduce the issue, I wrote a Python script that generates a large testbench using the `advance_clock` task and fork statements. Here is the Python script:

```python
"""
Script to generate a SystemVerilog testbench with multiple test tasks.
Each test includes a counter, timeout logic, and a validation loop.
"""

# Number of tests to generate
NUM_TESTS = 200

# Testbench header template
HEADER = """\
module Testbench;

  logic clk;
  logic reset;

  // Clock driver
  initial begin
    clk = 0;
    forever begin
      #5 clk = ~clk;
    end
  end

  task automatic advance_clock(int n = 1);
    repeat (n) @(posedge clk);
  endtask

"""

# Testbench footer template
FOOTER = """
  initial begin
"""

# Test task template
TEST_TASK_TEMPLATE = """
  task automatic test_{num}();
    int counter = 0;
    int expected_value = {num};

    // Timeout wait
    fork
      begin
        advance_clock(10000);
        $error("Timeout");
      end
    join_none
    wait (counter == expected_value);
    disable fork;

    while (counter < expected_value) begin
      advance_clock();
      counter++;
    end
  endtask
"""

# Call template for invoking each test task
CALL_TEMPLATE = "    test_{num}();\n"

# Footer end
FOOTER_END = """
    $finish;
  end

endmodule
"""

def generate_testbench(filename, num_tests):
    """
    Generates a SystemVerilog testbench with the specified number of tests.

    Args:
        filename (str): The output file name for the generated testbench.
        num_tests (int): The number of test tasks to generate.
    """
    with open(filename, "w", encoding="utf-8") as file:
        # Write the header
        file.write(HEADER)

        # Generate the test tasks
        for i in range(1, num_tests + 1):
            file.write(TEST_TASK_TEMPLATE.format(num=i))

        # Write the initial block with test calls
        file.write(FOOTER)
        for i in range(1, num_tests + 1):
            file.write(CALL_TEMPLATE.format(num=i))
        file.write(FOOTER_END)

if __name__ == "__main__":
    OUTPUT_FILENAME = "generated_testbench.sv"
    generate_testbench(OUTPUT_FILENAME, NUM_TESTS)
    print(f"Generated testbench with {NUM_TESTS} tests using fork structure: '{OUTPUT_FILENAME}'")

```

After running the Python script, it generates a `generated_testbench.sv`. Then, by running Verilator with the following command:

```bash
verilator generated_testbench.sv --cc --exe --binary --main
```

I can consistently reproduce the error with this script now.

wsnyder, 2024-11-24:
@hankhsu1996 thanks for the great test!

@kbieganski might you get a chance to debug this please?  I converted it to test_regress format (test_regress/t/t_fork_advance.py):

```
#!/usr/bin/env python3
# DESCRIPTION: Verilator: Verilog Test driver/expect definition
#
# Copyright 2024 by Wilson Snyder. This program is free software; you
# can redistribute it and/or modify it under the terms of either the GNU
# Lesser General Public License Version 3 or the Perl Artistic License
# Version 2.0.
# SPDX-License-Identifier: LGPL-3.0-only OR Artistic-2.0

import vltest_bootstrap

test.scenarios('simulator')
test.top_filename = test.obj_dir + "/t_gate_tree.v"

# Number of tests to generate
NUM_TESTS = 200

# Testbench header template
HEADER = """\
module Testbench;

  logic clk;
  logic reset;

  // Clock driver
  initial begin
    clk = 0;
    forever begin
      #5 clk = ~clk;
    end
  end

  task automatic advance_clock(int n = 1);
    repeat (n) @(posedge clk);
  endtask

"""

# Test task template
TEST_TASK_TEMPLATE = """
  task automatic test_{num}();
    int counter = 0;
    int expected_value = {num};

    // Timeout wait
    fork
      begin
        advance_clock(10000);
        $error("Timeout");
      end
    join_none
    wait (counter == expected_value);
    disable fork;

    while (counter < expected_value) begin
      advance_clock();
      counter++;
    end
  endtask
"""

# Testbench footer template
FOOTER = "  initial begin"

# Call template for invoking each test task
CALL_TEMPLATE = "    test_{num}();\n"

# Footer end
FOOTER_END = """
    $finish;
  end

endmodule
"""

def gen(filename, num_tests):
    """
    Generates a SystemVerilog testbench with the specified number of tests.

    Args:
        filename (str): The output file name for the generated testbench.
        num_tests (int): The number of test tasks to generate.
    """
    with open(filename, 'w', encoding="utf-8") as fh:
        fh.write("// Generated by t_fork_advance.py\n")

        # Write the header
        fh.write(HEADER)

        # Generate the test tasks
        for i in range(1, num_tests + 1):
            fh.write(TEST_TASK_TEMPLATE.format(num=i))

        # Write the initial block with test calls
        fh.write(FOOTER)
        for i in range(1, num_tests + 1):
            fh.write(CALL_TEMPLATE.format(num=i))
        fh.write(FOOTER_END)

gen(test.top_filename, NUM_TESTS)

test.compile(verilator_flags2=["--binary"])

test.execute()

test.passes()

```

closed 2024-12-02 by wsnyder