verilator/verilator #5633: Interface Array Failing to Compile

mrstrike007, 2024-11-26:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

```
`timescale 1ns / 1ps

interface generic_interface(
                            input clk,
                            input rst
                           );
    logic        valid;
    logic        ready;
    logic [31:0] data;

endinterface

class genericClass #(
                     int NUM_INTF = 5
                    );

    virtual generic_interface vif [NUM_INTF];

endclass

module tb_top();

    localparam int NUM_INTF = 6;

    genericClass #(NUM_INTF) genCls;

    logic tb_clk, tb_rst;

    generic_interface gen_if[NUM_INTF] (tb_clk, tb_rst);

    initial begin
        tb_clk = '0;
        tb_rst = '0;

        genCls = new();

        genCls.vif = gen_if;

        $finish();
    end

endmodule
```

```
The internal error shows the following: 

%Error: Internal Error: tb_top.sv:37:22: ../V3Broken.cpp:168: Broken link in node (or something without maybePointedTo): 'm_varp && !m_varp->brokeExists()' @ ./V3Ast__gen_impl.h:4694
                                       : ... note: In instance 'tb_top'
-node: VARREF 0x192ae00 <e1694> {d37aw} @dt=0x194d9d0@(w0)uc[0:5]  gen_if__Viftop [RV] <- VAR 0x19565d0 <e1833#> {d29ax} @dt=0x194d9d0@(w0)uc[0:5]  gen_if__Viftop [VSTATIC]  IFACEREF
   37 |         genCls.vif = gen_if;
      |                      ^~~~~~
- V3Ast.cpp:1323:     Dumping obj_dir/Vtb_top_990_final.tree
- V3StatsReport.cpp:231:statsReport:
```

Please note that when working with a single interface rather than an array of interfaces the code compiles.

What 'verilator' command line do we use to run your example?

`verilator --top-module tb_top --debug --binary --assert -Wall -Wno-fatal --timescale 1ns/1ps --trace --trace-structs  ./tb_top.sv`

What 'verilator --version' are you using?  Did you try it with the git master version?

`Verilator 5.030 2024-10-27 rev v5.030-45-g2cb1a8de7`

Have not tried at latest master head.

What OS and distribution are you using?

Fedora

May we assist you in trying to fix this in Verilator yourself?

Yes.

(Please avoid attaching screenshots that show text - you can convert images to text using e.g. https://ocr.space)

wsnyder, 2024-11-27:
Perhaps related to #5270. 

Thanks for the report. From this and other virtual interface issues these may take some time to get to resolved, so pull requests to fix these welcome.

nbstrike, 2024-11-27:
@wsnyder -- yes this seems to suffer the same failure as #5270.

I looked at this earlier with `--dumpi-tree 9` and the error seems to come about as a result of the `dearrayAll` pass, where the array of interfaces is dearray'ed (exploded?) and at that point new AstNodes are created for the array contents, however references from the initial process are not fixed up to point at the newly created AstNodes.

I'm not too familiar with the innards of verilator, I did poke around a bit and made a few observations:

- If the interfaces were being assigned to an instantiated module it seems like I could have fixed it up by visiting the tree under `newp->pinsp()` (which is empty in this case, but presumably could be populated by an earlier pass) -- but I suspect that is not the right answer here?
- If within the initial process I assign a signal within the interface to something (eg `_valid = gen_if[0].valid;`), that reference to the `gen_if[0]` object is an AstVarXRef which does appear to get fixed up correctly after the dearrayAll pass. That AstVarXRef seems to have been changed anticipating the dearrayAll pass earlier in the LinkDot pass -- perhaps that is an appropriate model here? (although I'm not too clear how it should be implemented);
- if I comment out the dearrayAll pass then I get further (it does crash eventually, presumably my hacking broke something else).

Given that I don't know if you're able to suggest a way forward, or perhaps where else I should look?

wsnyder, 2024-11-27:
All the references to the exploded arrays should have their pointers cleared, so if they are not that is likely the issue.  Then yes, the later LinkDot stage should resolve the AstVarXRef's to the correct now-dearrayed instances (FWIW this process is a bit of a hack, sorry).

PR #5869 (2025-03-21): Fix interface array assignments (#5270) (#5633)
Handle assignments of interface arrays to virtual interfaces and adds a test case.

resolves #5633, resolves #5270.
closed 2025-03-21 by wsnyder