verilator/verilator #5634: Importing Package within a Class Does Not Fail Compile

mrstrike007, 2024-11-26:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

Section 26.3 of the 2017 LRM states the following "It shall be illegal to have an import statement directly within a class scope.". This fails in Questa/ModelSim with the expected error, yet in Verilator the following code compiles:

```
`timescale 1ns / 1ps

package pkg;
    typedef enum int {RED, GREEN, BLUE} colors_e_t;
endpackage

class genericClass #(
                     int NUM_INTF = 5
                    );

    import pkg::*;

    colors_e_t colors = RED;

endclass

module tb_top();

    logic tb_clk, tb_rst;

    genericClass #(5) genCls;

    initial begin
        tb_clk = '0;
        tb_rst = '0;

        genCls = new();

        $finish();
    end

endmodule
```

What 'verilator' command line do we use to run your example?

`verilator --top-module tb_top  --binary --assert -Wall -Wno-fatal --timescale 1ns/1ps --trace --trace-structs  ./tb_top.sv`

What 'verilator --version' are you using?  Did you try it with the git master version?

`Verilator 5.030 2024-10-27 rev v5.030-45-g2cb1a8de7`

Have not tried on master head.

What OS and distribution are you using?

Fedora.

May we assist you in trying to fix this in Verilator yourself?

Yes.

(Please avoid attaching screenshots that show text - you can convert images to text using e.g. https://ocr.space)

wsnyder, 2024-11-27:
Thanks for the test - a pull to add this error would be appreciated, I'd suggest adding the check in V3LinkParse.

mrstrike007, 2024-11-27:
Just to clarify. Was that suggestion directed at me or to a member?

wsnyder, 2024-11-27:
To you, @mrstrike007, as you indicated you might be willing to fix it, thanks for considering!

PR #5679 (2025-01-07): Add lint error on importing package within a class (#5634)
Add check for import statements directly within the class scope as these are illegal according to Section 26.3 of the 2017 LRM, along with a testcase to validate  the functionality.

Resolves #5634.
closed 2025-01-07 by wsnyder