verilator/verilator #5638: The order of clocking block and assertion evaluation

katsui97, 2024-11-27:
The clocking block wait @(cb) seems to be released in the Observed region.
I had understood that assertion evaluation occurs after that, but in Verilator, it appears to be happening before the assertion evaluation.
Is my understanding incorrect?
------------------------------------------------------------------------
module top();

  // Gen Clock
  localparam PERIOD = 10;

  bit CLK;
  initial
  begin
    #PERIOD;
    forever
    begin
      CLK = ~CLK;
      #(PERIOD/2);
    end
  end

  logic RST_X = 1'b1;

  clocking cb @(posedge CLK);
  endclocking

  logic [3:0] r_cnt;
  always_ff @(posedge CLK, negedge RST_X)
    if(~RST_X)
      r_cnt <= '0;
    else
      r_cnt <= 4'(r_cnt + 1'b1);
  

  initial
  begin
    @(negedge CLK)
      RST_X = 1'b0;
    @(negedge CLK)
      RST_X = 1'b1;

    repeat(20)
      @(posedge CLK);

    $finish;
  end

  chk: assert property
  (
    @(posedge CLK)
    (r_cnt == 4'd5) |-> (1'b0)
  )
    $display("\nasset:  r_cnt=%1d\n", r_cnt);
  else
  begin
    $display("\nnot_asset:  r_cnt=%1d\n", r_cnt );
  end

  initial
  begin
    forever
    begin
      @(cb)
      if(r_cnt == 4'd5)
      begin
        r_cnt = '0;   // This would affect the assertion.
      end
    end
  end

endmodule

wsnyder, 2024-11-27:
@kbieganski ?

katsui97, 2024-11-30:
I noticed that my previous comment was contradictory, so I am reposting it.
I understand that the @(cb) release in a clocking block occurs in the Observed region or possibly the Re-active region.
I had thought that assertion evaluations would already be completed before the @(cb) release, but in Verilator, it seems that @(cb) is released before the assertion evaluation.
In other words, if the testbench modifies DUT values after the @(cb) release, it appears to affect the assertion results.

Additionally, when defining input and output within the clocking block, non-blocking assignments can be made in initial blocks even in Verilator.
However, in such cases, the assignments seem to be ignored and are not reflected in the DUT.
Based on this behavior, it seems that these assignments are not behaving like Re-NBA (re-scheduled non-blocking assignments).

closed 2024-12-13 by wsnyder