verilator/verilator #5647: Support `solve..before` constraints

RRozak, 2024-11-29:
`solve..before` constraints aren't currently supported.
The following code gives a warning:
```systemverilog
class Cls;
   rand int x;
   rand int y;
   constraint dynsize { 
      solve x before y;
      x < y;
   }
endclass

module t (/*AUTOARG*/);
   Cls obj;
   int res;

   initial begin
      obj = new;
      res = obj.randomize();
   end
endmodule
```
```
%Warning-CONSTRAINTIGN: ex/solve_before.sv:5:7: Constraint expression ignored (imperfect distribution)
                                              : ... note: In instance 't'
    5 |       solve x before y;
      |       ^~~~~
                        ... For warning description see https://verilator.org/warn/CONSTRAINTIGN?v=5.031
                        ... Use "/* verilator lint_off CONSTRAINTIGN */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
```

I think this constraint can be used to support queue size constraints (see https://github.com/verilator/verilator/issues/5582#issuecomment-2507463302)

RRozak, 2024-11-29:
Random generator contains a list of variables that should be randomized: https://github.com/verilator/verilator/blob/d1656712257ed73552a110bc284c695b6b82dd38/include/verilated_random.h#L217-L218
If we want to keep it in that way, we should have a few generators, one for each group. There may also be `solve..before` constraint in `randomize..with`, which may require a different division into groups than in other randomize functions.

I think it would be good to replace a generator with static functions which would take constraints and return the values that were found by a solver or at least to make a generator a function local variable instead of a class field.

wsnyder, 2024-11-29:
This approach makes sense to me.

I'd suggest see if it can be done without making the functions static, perhaps a new class (of which there is two instances corresponding to the groups before and after) would work?

RRozak, 2024-11-29:
Do you mean having a class instance for each group (because there may be more than two groups)?
I think it would be better if a generator variable wasn't a class field, because it isn't in a case of `randomize..with`. `randomize..with` uses a local copy of a generator from a class, which has one constraint more. If these two functions had their own local generators, we would be able to unify the handling of `randomize` and `randomize..with`. It should also make adding support for `std::randomize` easier.

RRozak, 2024-12-11:
> I think that this constraint can be handled in the following way:
> 
>     1. Create a graph in which constrained variables are vertices and there is an edge A -> B if there is a constraint `solve B before A`.
> 
>     2. Divide the variables into groups in such a way that all variables within a group can be randomized simultaneously.
> 
>     3. Add calls to the solver for each group and array resizing after it if any variable in a group is a size variable.

This approach sets the order during verilation and it's insufficient for `randomize...with`.
`randomize..with` should behave as normal `randomize` with additional constraints, so it should be virtual.
However, we handle it by converting it to a function that is not virtual. It works, because inside we call a virtual function that sets up constraints.
Please consider the following example:
```systemverilog
class A;
   rand int x;
   rand int y;
   constraint c1 { x > 5; };
endclass

class B extends A;
   constraint c2 { solve x before y; };
endclass

class C extends A;
endclass

module t;
   initial begin
      B b = new;
      C c = new;
      A a;
      bit pred;
      a = pred ? b : c;
      int res = a.randomize() with { x < y; };
   end
endmodule
```
We don't know what is the actual type of object `a`, so we don't know the priority of `x`. We don't know when we should solve the constraint provided in `with` clause.
We may also have `solve..before` inside `with`. In this case, `randomize..with` should have a different graph than `randomize`, be we don't even know the set of variables.

I see two solutions:
1. Go with the original approach, but generate `randomize..with` implementation for each class that inherits from the class that is the type of the handle on which that function is called. These functions would be virtual, so it would work.
2. Do something like the original approach, but in runtime. While this seems to be much harder solution, it would enable us to handle constraints `solve a[y] before x` where `y` is constrained too. We can't handle this during verilation, because we don't know what the value of `y` is. Another case is `solve before` inside `foreach` on dynamic array, like here: https://github.com/antmicro/riscv-dv/blob/2f1d7037ddb1acd2e8a1a233fedf30ff7d86dc82/src/riscv_pmp_cfg.sv#L148-L152. During verilation, we don't even know how many constrained variables there are.

RRozak, 2024-12-13:
@wsnyder, could you comment on which solution you think is better?

wsnyder, 2024-12-13:
Another idea would be to make a different randomize() virtual function for each caller site that has a with{}.  E.g. randomize_with_at_filefoo_line123().  (Really a different randomize for each unique set of with() calls, as multiple sites with same arguments can share the same function.)

I think that works with the `solve a[y] before x`.  I don't see what prevents static organization of constraints in the foreach example?  I agree they need to be solved in a specific order in that example of course.

Generally I'm a fan of our statically doing everything we can at build time, as it's faster for runtime.  If we end up with cases that must be postponed, we can, but I prefer that's not the default.

RRozak, 2024-12-16:
> Another idea would be to make a different randomize() virtual function for each caller site that has a with{}. E.g. randomize_with_at_filefoo_line123(). (Really a different randomize for each unique set of with() calls, as multiple sites with same arguments can share the same function.)

Currently it is done in that way. Each `randomize..with` function is converted to a separate function. But these functions don't work as virtual, because only one implementation for each `randomize..with` is generated. To make it work as virtual, we would need to generate a method for each class that inherits from the class of which the handle is, because the actual type of the object may be any of these classes.

I think a better example that shows that the static approach is insufficient is `solve a[y] before a[x]`, where `x` and `y` are randomized. In this case we don't know the order during verilation.

RRozak, 2026-02-20:
I am not working on it. Feel free to push it forward.