verilator/verilator #5651: Add ability for interactive control of sim execution

chili-chips-ba, 2024-12-02:
ModelSim has a really handy Tcl console from which users can command simulation start and stop, create break points, and otherwise interact with the sim in a way that facilitates debug. Something like that would be of value for opensource too. 

It seems that @wyvernSemi's VProc has the potential for adding this much-needed capability. Here is a [teaser](https://www.linkedin.com/feed/update/urn:li:activity:7269067079431925760), just to kick off the brainstorming process...

wsnyder, 2024-12-03:
Many simulators have interactive mode, as did the very original "Verilog-XL"-the-program.

FWIW most of the larger-organizations I'm aware of runs Verilator in batch-mode. Also we aren't keeping up with the bug backlog already, so I don't expect to be able to add the effort to support an interactive mode "inside" Verilator.

Instead, I'd encourage you/others to experiment with connecting TCL or using VProc, or whatever, through your own main(). I believe Verilator already has the primary hooks needed in its API, as Verilator has been wrapped with an interactive shell by at least 3 other closed-source implementations. But, if some new Verilator API extensions are needed to enable or extend functionality between that wrapper and Verilator, feel free to open issues/pulls.

BTW I also like the VProc concept of the console being simulator-agnostic, as then people don't need to reprogram their tool-interacting brains when switching simulators. Similar reasoning to wanting a common wave viewer across simulators.

So, closing not a target for long term, but as above feel free to file API issues or generally keep the community involved by following up here on progress/alternatives/etc.

closed 2024-12-03 by wsnyder

chili-chips-ba, 2024-12-03:
... the reasoning makes sense, esp. the `simulator agnostic` part that the VProc-based interactive cockpit would offer. We will then pursue this idea within wireguard-fpga project, to perhaps show it off later on. 

Just out of curiosity, are there any additional leads you can offer on _"... Verilator has been wrapped with an interactive shell by at least 3 other implementations..."_ ? We understand they are all closed-source, but still, we are hoping for some articles, posts or other public info for learning more about them...

wsnyder, 2024-12-03:
One wrapper I wrote in interactive Perl about 20 years ago, it interacted through /*verilator public*/ as that was pre DPI/VPI.  The command set was for software people to interact with the simulation of a chip, as opposed to peeking at RTL, for example it could "poke" a register by sending an address to the RTL which would do the address/data bus transactions appropriately.  It also supported a commercial simulator through their API (since discontinued) - predated DPI.

One major CPU vendor has a command line and GUI their customers download to interact with their processor model. Likewise this is using Verilator under the hood.

Another vendor likewise for something other than a CPU.

In all these the goal was a higher abstraction of something modelled, though it could still e.g. advance time, see internal state to a certain amount.

I wasn't able to find anything public, FWIW VProc seems a better starting point.

chili-chips-ba, 2024-12-03:
Thank you for these insights! [VProc](https://github.com/wyvernSemi/vproc) is indeed using modern **DPI-C**

> [image]

and includes a very convenient API clock tick function. While not looking to steal @wyvernSemi's thunder, we can't wait to see his `Verilator idea` turned into a super-handy tool for debugging digital designs, on par with leading commercial solutions, yet in open source ☂️ 

> [image]