verilator/verilator #5660: verilator --coverage issue

zhaoxuying, 2024-12-05:
Does Verilator support excluding some files when collecting coverage?

wsnyder, 2024-12-05:
https://veripool.org/guide/latest/simulating.html#line-coverage  applies to all coverage types.

closed 2024-12-05 by wsnyder

wsnyder, 2024-12-05:
Moved that to a new section to be more obvious: https://veripool.org/guide/latest/simulating.html#suppressing-coverage

zhaoxuying, 2024-12-06:
[image]
After adding the --coverage option, the following errors were reported:

/lib/x86_64-linux-gnu/Scrt1.o: in function `_start':
(.text+0x21): relocation truncated to fit: R_X86_64_GOTPCRELX against symbol `__libc_start_main@@GLIBC_2.34' defined in .text section in /lib/x86_64-linux-gnu/libc.so.6
/lib/x86_64-linux-gnu/Scrt1.o:(.eh_frame+0x20): relocation truncated to fit: R_X86_64_PC32 against `.text'
/lib/x86_64-linux-gnu/crti.o: in function `_init':
(.init+0xb): relocation truncated to fit: R_X86_64_REX_GOTPCRELX against undefined symbol `__gmon_start__'
VSimTop__ALL.a(VSimTop.o): in function `VSimTop::VSimTop(VerilatedContext*, char const*)':
VSimTop.cpp:(.text+0x13): failed to convert GOTPCREL relocation against '_ZTV7VSimTop'; relink with --no-relax
VSimTop__ALL.a(VSimTop.o): in function `VSimTop::VSimTop(char const*)':
VSimTop.cpp:(.text+0x171): relocation truncated to fit: R_X86_64_GOTPCREL against undefined symbol `TLS init function for Verilated::t_s'
VSimTop.cpp:(.text+0x198): relocation truncated to fit: R_X86_64_GOTPCREL against undefined symbol `TLS init function for Verilated::t_s'
VSimTop.cpp:(.text+0x1d3): failed to convert GOTPCREL relocation against '_ZN9Verilated14s_lastContextpE'; relink with --no-relax
VSimTop.cpp:(.text+0x1e3): failed to convert GOTPCREL relocation against '_ZGVZN9Verilated15defaultContextpEvE3s_s'; relink with --no-relax
VSimTop.cpp:(.text+0x1f0): failed to convert GOTPCREL relocation against '_ZZN9Verilated15defaultContextpEvE3s_s'; relink with --no-relax
VSimTop.cpp:(.text+0x1fb): relocation truncated to fit: R_X86_64_GOTPCREL against undefined symbol `TLS init function for Verilated::t_s'
VSimTop.cpp:(.text+0x21d): relocation truncated to fit: R_X86_64_GOTPCREL against undefined symbol `TLS init function for Verilated::t_s'
VSimTop.cpp:(.text+0x22d): failed to convert GOTPCREL relocation against '_ZGVZN9Verilated15defaultContextpEvE3s_s'; relink with --no-relax
VSimTop.cpp:(.text+0x23d): failed to convert GOTPCREL relocation against '_ZZN9Verilated15defaultContextpEvE3s_s'; relink with --no-relax
VSimTop.cpp:(.text+0x250): failed to convert GOTPCREL relocation against '_ZZN9Verilated15defaultContextpEvE3s_s'; relink with --no-relax
VSimTop.cpp:(.text+0x257): relocation truncated to fit: R_X86_64_PC32 against symbol `__dso_handle' defined in .data.rel.local section in /usr/bin/../lib/gcc/x86_64-linux-gnu/11/crtbeginS.o
VSimTop.cpp:(.text+0x263): failed to convert GOTPCREL relocation against '_ZGVZN9Verilated15defaultContextpEvE3s_s'; relink with --no-relax
VSimTop.cpp:(.text+0x277): failed to convert GOTPCREL relocation against '_ZGVZN9Verilated15defaultContextpEvE3s_s'; relink with --no-relax
VSimTop__ALL.a(VSimTop.o): in function `VSimTop::~VSimTop()':
VSimTop.cpp:(.text+0x294): failed to convert GOTPCREL relocation against '_ZTV7VSimTop'; relink with --no-relax
VSimTop__ALL.a(VSimTop.o): in function `VSimTop::nextTimeSlot()':
VSimTop.cpp:(.text+0x374): relocation truncated to fit: R_X86_64_PC32 against `.L.str'
VSimTop.cpp:(.text+0x37b): relocation truncated to fit: R_X86_64_PC32 against `.L.str.1'
VSimTop.cpp:(.text+0x382): additional relocation overflows omitted from the output
clang: error: linker command failed with exit code 1 (use -v to see invocation)

Could you give some suggestions?

wsnyder, 2024-12-06:
This is some compiler issue as opposed to something Verilator is doing wrong. Googling the error "relocation truncated to fit: R_X86_64_PC32 against `.text'" I see a lot of discussions but didn't dig in, please read through them to figure out what to fix.  We'd appreciate a bug update with the fix to help the next person with this compiler issue, thanks.

zhaoxuying, 2024-12-07:
When we compiled the Verilog code, we added some C++ code related to testing. The size of this C++ code was relatively large, which caused an address relocation truncation error. After streamlining the test code, this problem was resolved. Thank you for your help.

zhaoxuying, 2024-12-11:
Does Verilator support the ''covergroup'' and ''coverpoint'' syntax in SystemVerilog? Thank you.

wsnyder, 2024-12-11:
@zhaoxuying no covergroup etc yet, there is a open feature bug.