verilator/verilator #5662: Input combinatorial debug

toddstrader, 2024-12-05:
After a small change to a design we noticed a significant performance regression in our sims.  Using `--prof-cfuncs` we were able to track it down to one offending block in the RTL.  The resultant C++ has gone from an `__nba_sequent__` to an `__ico_sequent__` function and is now being called by not only `__eval_nba()` but also by `__eval_ico()`.  The latter is happening a lot.

Is there a simple way to determine what is causing this block to now be marked as input combinatorial?  I assume the graphviz diagrams would show me, but the design is large and I'm wondering if there's a simpler / faster way.

gezalore, 2024-12-05:
As you probably know, 'ico' contains combinational logic that is driven from a top level input, or more precisely from a signal that is:
https://github.com/verilator/verilator/blob/676fd3163590dc62c9121d0966857c2c4397251f/src/V3SchedReplicate.cpp#L153-L156

There is no simple to debug format that is currently dumped. Gross hack to figure out why something was placed into ico could be to delete the vertices from the graph that do not have the 'INPUT' driving region set, so you only dump the relevant sub-graph. That is, make this line dump a only the relevant sub-graph:
https://github.com/verilator/verilator/blob/676fd3163590dc62c9121d0966857c2c4397251f/src/V3SchedReplicate.cpp#L304
Generally ico should be fairly small, compared to the whole graph.

Also, you probably want to differentiate between the "seed" 'INPUT' vertices, i.e.: those that are marked as being in the 'INPUT' region in the constructor snippet above, and those vertices that were propagated in `propagateDrivingRegions`. If you then know which block is the problematic one, you should be able to find it's environment and see what necessitates putting it into 'ico'.

It might be worth checking if you have any circular logic in the ico region, which might cause a big slowdown. You can check the number of trigger bits in `__VicoTriggered` in the generated code, if you have a big increase, it might explain the performance difference. Also the generated `dump_triggers__ico` function contains some hints as to what are the circular signals (shown as 'hybrid' sensitivity items).

One more note of caution: V3Combine sometimes can combine ico/act/nba/etc functions with each other, so the function name is not guaranteed to accurately show which region is actually executing it. If combined, multiple region eval functions will call the same comb/sequent sub-functions.

wsnyder, 2024-12-05:
Also adding `--runtime-debug` will annotate C code with node and filelines, which might help attributing/cross referencing.

toddstrader, 2024-12-06:
Thanks all.  The `sched-replicate-propagated` graph was the key.  I was able to manually trace through the graphviz file using the secret decoder ring in `SchedReplicateVertex::dotColor()` to find what was causing the unwanted `ico`.  If I have to do this a lot I'll cook up a tool to do it.  But for now, problem solved.

closed 2024-12-06 by toddstrader