verilator/verilator #5664: V3DfgPeephole: width mismatch internal error

b-chmiel, 2024-12-06:
I'd received internal errors in V3DfgPeephole regarding mismatched width of operands.
Verilated using master as of https://github.com/verilator/verilator/commit/9656311521a7fe33054a4921986fa0cf8f6c0f5f.

## RHS mismatch internal error
Error message:
```cpp
%Error: Internal Error: t/t_dfg_peephole_bug.v:12:19: ../V3DfgPeephole.cpp:956: Mismatched RHS width
   12 |    assign o = arr | 1;
      |                   ^
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

Failing test:
```sv
// DESCRIPTION: Verilator: Verilog Test module
//
// This file ONLY is placed under the Creative Commons Public Domain, for
// any use, without warranty, 2024 by Antmicro.
// SPDX-License-Identifier: CC0-1.0

module t(
   output [1:0] o
   );

   wire [1:0] arr [1:0];
   assign o = arr | 1;
endmodule
```

```py
#!/usr/bin/env python3
# DESCRIPTION: Verilator: Verilog Test driver/expect definition
#
# Copyright 2024 by Wilson Snyder. This program is free software; you
# can redistribute it and/or modify it under the terms of either the GNU
# Lesser General Public License Version 3 or the Perl Artistic License
# Version 2.0.
# SPDX-License-Identifier: LGPL-3.0-only OR Artistic-2.0

import vltest_bootstrap

test.scenarios('vlt')

test.lint()

test.passes()
```
## LHS mismatch internal error

Error message:
```cpp
%Error: Internal Error: t/t_dfg_peephole_bug.v:13:19: ../V3DfgPeephole.cpp:955: Mismatched LHS width
   13 |    assign o = arr | i + 1;
      |                   ^
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

Failing test:
```sv
// DESCRIPTION: Verilator: Verilog Test module
//
// This file ONLY is placed under the Creative Commons Public Domain, for
// any use, without warranty, 2024 by Antmicro.
// SPDX-License-Identifier: CC0-1.0

module t(
   input [1:0] i,
   output [1:0] o
   );

   wire [1:0] arr [1:0];
   assign o = arr | i + 1;
endmodule
```

Test runner code the same as above.

gezalore, 2024-12-06:
`arr | 1` / `arr | i` is malformed (LHS is unpacked). V3Width should have errored out on this, a PR to fix it would be appreciated.

wsnyder, 2026-10-11:
#8504 added error.

closed 2026-10-11 by wsnyder