verilator/verilator #5668: FST tracing asserts at runtime with `Assertion `hierarchicalName.rfind(' ') != std::string::npos` if top level module is nameless

jwise, 2024-12-10:
I am running a large cocotb testbench that calls into Verilator, and trying to switch from outputting a VCD to outputting a FST.  I'm using Verilator from Git, sha 9656311521a7fe33054a4921986fa0cf8f6c0f5f.

Shortly after the cocotb testbench tries to turn on tracing, I get:

`tb_end_to_end_litex: /storage/home/joshua/work/verilator/include/verilated_fst_c.cpp:184: void VerilatedFst::declare(uint32_t, const char*, int, VerilatedTraceSigDirection, VerilatedTraceSigKind, VerilatedTraceSigType, bool, int, bool, int, int): Assertion `hierarchicalName.rfind(' ') != std::string::npos' failed.`

Rebuilding with `-Og -g`, I find a bunch of stuff optimized out, but enough in the backtrace to see where we are coming from:

```
#5  0x00007ffff7aabeb2 in __GI___assert_fail (assertion=0x5555557f5ce8 "hierarchicalName.rfind(' ') != std::string::npos", file=0x5555557f5b00 "/storage/home/joshua/work/verilator/include/verilated_fst_c.cpp", 
    line=184, function=0x5555557f5c48 "void VerilatedFst::declare(uint32_t, const char*, int, VerilatedTraceSigDirection, VerilatedTraceSigKind, VerilatedTraceSigType, bool, int, bool, int, int)")
    at ./assert/assert.c:101
#6  0x00005555555b3c87 in VerilatedFst::declare (this=this@entry=0x5555562d1e50, code=<optimized out>, code@entry=2, name=name@entry=0x5555557f6af3 "fpga_health", dtypenum=<optimized out>, dtypenum@entry=-1, 
    direction=<optimized out>, kind=kind@entry=VerilatedTraceSigKind::WIRE, type=VerilatedTraceSigType::LOGIC, array=false, arraynum=-1, bussed=false, msb=0, lsb=0)
    at /storage/home/joshua/work/verilator/include/verilated_fst_c.cpp:184
#7  0x00005555555b404a in VerilatedFst::declBit (this=this@entry=0x5555562d1e50, code=code@entry=2, fidx=fidx@entry=0, name=name@entry=0x5555557f6af3 "fpga_health", dtypenum=dtypenum@entry=-1, 
    direction=direction@entry=VerilatedTraceSigDirection::OUTPUT, kind=VerilatedTraceSigKind::WIRE, type=VerilatedTraceSigType::LOGIC, array=false, arraynum=-1)
    at /storage/home/joshua/work/verilator/include/verilated_fst_c.cpp:248
#8  0x0000555555732ee6 in Vtop___024root__trace_init_sub__TOP__0 (vlSelf=vlSelf@entry=0x7ffff79ea100, tracep=tracep@entry=0x5555562d1e50) at Vtop__Trace__0__Slow.cpp:20
#9  0x00005555557339e7 in Vtop___024root__trace_init_top (vlSelf=vlSelf@entry=0x7ffff79ea100, tracep=tracep@entry=0x5555562d1e50) at Vtop__Trace__0__Slow.cpp:9790
#10 0x00005555555ba126 in trace_init (voidSelf=0x7ffff79ea100, tracep=0x5555562d1e50, code=1) at Vtop.cpp:187
```

And indeed, moving up to stack frame `#6`:

```
(gdb) print hierarchicalName
$1 = "fpga_health"
```

For reference, the generated code in these stack frames is:

```c++
// #8
VL_ATTR_COLD void Vtop___024root__trace_init_sub__TOP__0(Vtop___024root* vlSelf, VerilatedFst* tracep) {
    (void)vlSelf;  // Prevent unused variable warning
    Vtop__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vtop___024root__trace_init_sub__TOP__0\n"); );
    auto& vlSelfRef = std::ref(*vlSelf).get();
    // Init
    const int c = vlSymsp->__Vm_baseCode;
    // Body
    tracep->pushPrefix("tb_end_to_end_litex", VerilatedTracePrefixType::SCOPE_MODULE);
    Vtop___024root__trace_init_sub__TOP__tb_end_to_end_litex__0(vlSelf, tracep);
    tracep->popPrefix();
    tracep->declBit(c+1,0,"fpga_health",-1, VerilatedTraceSigDirection::OUTPUT, VerilatedTraceSigKind::WIRE, VerilatedTraceSigType::LOGIC, false,-1); // line 20
    tracep->declBit(c+2,0,"fpga_test_n",-1, VerilatedTraceSigDirection::OUTPUT, VerilatedTraceSigKind::WIRE, VerilatedTraceSigType::LOGIC, false,-1);
    tracep->declBit(c+3,0,"fpga_status_n",-1, VerilatedTraceSigDirection::OUTPUT, VerilatedTraceSigKind::WIRE, VerilatedTraceSigType::LOGIC, false,-1);
    ...

// #9
VL_ATTR_COLD void Vtop___024root__trace_init_top(Vtop___024root* vlSelf, VerilatedFst* tracep) {
    (void)vlSelf;  // Prevent unused variable warning
    Vtop__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vtop___024root__trace_init_top\n"); );
    auto& vlSelfRef = std::ref(*vlSelf).get();
    // Body
    Vtop___024root__trace_init_sub__TOP__0(vlSelf, tracep); // line 9790
}

// #10 
VL_ATTR_COLD static void trace_init(void* voidSelf, VerilatedFst* tracep, uint32_t code) {
    // Callback from tracep->open()
    Vtop___024root* const __restrict vlSelf VL_ATTR_UNUSED = static_cast<Vtop___024root*>(voidSelf);
    Vtop__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    if (!vlSymsp->_vm_contextp__->calcUnusedSigs()) {
        VL_FATAL_MT(__FILE__, __LINE__, __FILE__,
            "Turning on wave traces requires Verilated::traceEverOn(true) call before time 0.");
    }
    vlSymsp->__Vm_baseCode = code;
    tracep->pushPrefix(std::string{vlSymsp->name()}, VerilatedTracePrefixType::SCOPE_MODULE);
    Vtop___024root__trace_decl_types(tracep);
    Vtop___024root__trace_init_top(vlSelf, tracep); // line 187
    tracep->popPrefix();
}
```

One distressing thing is that, in the context of `trace_init`, `vlSelf->vlSymsp->TOP.m_namep` is a zero-length string.  I went looking in the Cocotb verilator.cpp source, and indeed:

```
std::unique_ptr<Vtop> top(new Vtop(""));
```

If I change this to `Vtop("argh")`, then at least the simulator doesn't immediately crash, but cocotb would require other changes to make this work, lest I get:

```
     0.00ns ERROR    gpi                                VPI: Couldn't find root handle tb_end_to_end_litex
     0.00ns ERROR    gpi                                VPI: Toplevel instances: tb_end_to_end_litex != argh.tb_end_to_end_litex...
     0.00ns ERROR    gpi                                No root handle found
     0.00ns ERROR    cocotb                             cocotb testbench initialization failed. Exiting.
```

The fact that this worked once sort of suggests to me that this might be a regression -- and, if it's illegal to instantiate a module with an empty name, then probably there should be an assert to this effect that fires earlier?  If I should file this against cocotb instead, I can do that too.

jwise, 2024-12-10:
Ah, this is #5470 back for another round, maybe?

wsnyder, 2024-12-10:
#5470 / 48faf8d036db added a test, so it's not completely identical, if you can modify that test to show the issue it should be straightforward to fix.  Pull welcome as always too!

jamesrbailey, 2024-12-17:
I started seeing this recently as well. Commenting out the offending assertion has unblocked me and I can't see any obvious side effects, although I can't imagine this would be a correct fix.

wsnyder, 2025-01-09:
Can you modify a verilator test_regress to show the issue?  It might be ok to get rid of assertion, but without test hard to be sure & may break again.

stu-e, 2025-01-10:
Sorry I haver no idea how your regression system works or how you might test this kind of problem

wsnyder, 2025-02-25:
Closing - if someone can get a standalone test case will investigate.

closed 2025-02-25 by wsnyder

alexforencich, 2025-05-22:
I have been running in to this assertion failure sporadically as well.  Haven't been able to track down why.  I think at minimum an actual error message is needed here instead of a bare assert failure, something that actually prints out the offending name so the cause of the problem can be tracked down.  But even better would be to emit a warning, "fix up" the name to remove spaces or replace them with something else, and just continue running instead of crashing.

paul-demo, 2025-05-31:
Just encountered this for the first time as well on 5.036 (no issues with the same code on 5.024).

The issue appears to be related to packages. I threw some print statements in there and it spit out this:

```text
hierarchicalName:hamming_sec_pkg compute_bit_position__Vstatic__b_current
hierarchicalName.rfind(' '):15

hierarchicalName:clk
hierarchicalName.rfind(' '):18446744073709551615

Assertion failed: (hierarchicalName.rfind(' ') != std::string::npos), function declare, file verilated_fst_c.cpp, line 233.
```

The relevant code is that there is some package deep in the dependency tree which is included in some module:
```{systemverilog}
module axis_sync_fifo_mem
  import hamming_sec_pkg::compute_num_parity_bits;
  #(...)(...);
endmodule
```

There's nothing particularly interesting about the content of that package; here's an excerpt:
```{systemverilog}

package hamming_sec_pkg;

  // m parity bits covers w = (2^m - m - 1) useful data bits
  function int compute_num_parity_bits;
    input int w;
    begin
      // some body with if/else/for/etc. that computes m
      return m;
    end
  endfunction
  ...
endpackage
```

I don't know why it would put a space between the package and the function name.

alexforencich, 2025-05-31:
FWIW I am not using packages, so there seems to be multiple things that can cause this assertion failure.  In which case, some more defensive coding is in order instead of a bare assert.

paul-demo, 2025-05-31:
Edit: actually the one that it's complaining about is not `compute_num_parity_bits`, it's another function `compute_bit_position`. That one is used in two different ways:

```{systemverilog}

  always_comb begin
    for (int b = 0; b < WIDTH; b++) begin
      automatic int bpb = compute_bit_position(b);
      ...
    end
  end
```

and

```{systemverilog}
  generate
    genvar k;
    for (k = 0; k < WIDTH; k++) begin
      localparam int bpb = compute_bit_position(k);
      ...
    end
  endgenerate
```

alexforencich, 2025-05-31:
I am using cocotb, so adding something to the C++ side is not so straightforward.  But it is an open-source design, so the next time I run into the assert failure I'll see if I can post a link to the problematic testbench.

alexforencich, 2025-05-31:
I will note that I do use functions very occasionally, just not in packages.  I wonder if we could both be seeing some sort of issue related to how function names are handled somewhere, just not limited to packages specifically.

paul-demo, 2025-05-31:
I mean you can just find the file where the assert exists -- it's an installed, static file:

```bash
$> find /opt -name "verilated_fst_c.cpp"
/opt/homebrew/Cellar/verilator/5.036/share/verilator/include/verilated_fst_c.cpp
```

alexforencich, 2025-05-31:
Ah I see, yeah, in that case I guess I can look in to annotating it.