verilator/verilator #5672: clang sign-compare compilation warning

augierg, 2024-12-11:
```
clang++  -m64 -std=c++17 ...
In file included from /usr/local/share/verilator/include/verilated_vcd_sc.h:26:
In file included from /usr/local/share/verilator/include/verilated_sc_trace.h:24:
In file included from /usr/local/share/verilator/include/verilated.h:1023:
/usr/local/share/verilator/include/verilated_funcs.h:1591:26: fatal error: comparison of integers of different signs: 'std::size_t' (aka 'unsigned long') and 'int' [-Wsign-compare]
    for (size_t i = 0; i < q.size(); ++i) ret |= static_cast<IData>(q.at(i)) << (i * lbits);
                       ~ ^ ~~~~~~~~
1 error generated.
```

env:
```
Linux flash 5.15.0-125-generic #135-Ubuntu SMP Fri Sep 27 13:53:58 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
Verilator 5.030 2024-10-27 rev v5.030
Ubuntu clang version 14.0.0-1ubuntu1.1
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
```

wsnyder, 2024-12-12:
Are you using your own Makefile?  The default makefile Verilator builds should have Wno-sign-compare; see if it does in include/verilated.mk, and if so use that, and if not post the results of running ./configure.

augierg, 2024-12-12:
Yes, correct, I am using a traditional verilator command to generate a shared library file of my verilated RTL, then I use a custom makefile to compile a systemc environment that includes the header from verilated module + link shared library
That is where this warning shows up as an error, since we consider all compilation warnings as errors. If we want to build verilator includes with this warning disabled, I can add it to my makefile. That's what I am doing already, but I decided to report this, since it shows up in my recent upgrade from v5.026 to v5.030

wsnyder, 2024-12-12:
Ok, you need to update your Makefile then to use all the flags in verilated.mk, there are others too that may eventually turn up.

closed 2024-12-12 by wsnyder

augierg, 2024-12-12:
Thanks. Will do