verilator/verilator #5683: The `--hierarchical` option cannot find source files

awmoore-intel, 2024-12-19:
The `--hierarchical` option does not look for sources using `-I` like non-hierarchical mode when sources are not in the current directory. 

This happens on master running linux. 

Test case contains 2 sources that are 2-level deep in hierarchy and I tried 2 verilator invocations. One works and one doesn't.
```
Apptainer> more */*/*
::::::::::::::
build/src/mid.v
::::::::::::::
module mid(input clock, input reset, input a, input b, output c); /* verilator hier_block */

assign c = a & b;
endmodule
::::::::::::::
build/src/top.v
::::::::::::::
module top(input clock, input reset);

wire a,b,c;

mid mid(clock, reset, a, b, c);

endmodule
Apptainer> ../build/src/verilator -Ibuild/src  top.v mid.v --top-module top --cc 
- V e r i l a t i o n   R e p o r t:  rev v5.030 (mod)
- Verilator: Built from 0.033 MB sources in 3 modules, into 0.018 MB in 6 C++ files needing 0.000 MB
- Verilator: Walltime 0.025 s (elab=0.001, cvt=0.008, bld=0.000); cpu 0.025 s on 1 threads; alloced 9.930 MB
Apptainer> rm -rf mdir ; ../build/src/verilator -Ibuild/src  top.v mid.v --top-module top --cc  --hierarchical
make: Entering directory '/home/awmoore/pta/verilator/test/obj_dir'
make: *** No rule to make target 'top.v', needed by 'Vtop.mk'.  Stop.
make: Leaving directory '/home/awmoore/pta/verilator/test/obj_dir'
%Error: make -C obj_dir -f Vtop_hier.mk  -j 1  hier_verilation exited with 2
Apptainer> 
```

wsnyder, 2024-12-19:
When verilator reads the includes it should generate a .d file which has the appropriate path to the included files.  Perhaps that is incorrect for hierarchical?

We don't want to add to VPATH because that can break things if there are files of the same name in multiple directories, as the VPATH will change what's picked up.

As a workaround use absolute paths.

Updating your patch would be appreciated, we'll need a test case as part of the patch, and test against cmake too.

awmoore-intel, 2024-12-20:
There are no .d files created at this stage, only MkArgs.mk files. Yes, I can modify the .f file to use absolute or relative path, however, this is not required in non-hierarchical mode. 

I'm not sure how to tell make where the file is other than VPATH or be forced to use absolute or complete relative paths (instead of just filenames, like it is now).

wsnyder, 2024-12-20:
Ok, now I understand, running this instead works:

        rm -rf obj_dir
        verilator -y rtl/src  rtl/src/top.v rtl/src/mid.v --top-module top --cc
        rm -rf obj_dir
        verilator -y rtl/src  rtl/src/top.v rtl/src/mid.v --top-module top --cc  --hierarchical

First, for finding modules I suggest use -y, not -I.  -I matches gcc, but isn't supported by other simulators.  Note also -y is for finding modules in other simulators versus include files.  Verilator doesn't really care, but this is compatible with what other simulators need.

Second, if you give a filename on the command line it needs to be a path of a valid file, the -y/-I is not applied.  This also matches what most other tools support.

Adding VPATH resolved the filename for you, if you want to use that in a personal makefile it's fine, but not something we should generally do as it will confuse when the same filename exists in multiple dirs (a bad practice but people do it).

Closing as I think answered, but feel free to follow up!

closed 2024-12-20 by wsnyder

awmoore-intel, 2024-12-20:
I'm fine with closing it since it appears unsupported, (appreciate your comments), but here's a tweaked version of your last command  - one passes and one fails.

```
Apptainer> verilator -y rtl/src  top.v mid.v --top-module top --cc 
- V e r i l a t i o n   R e p o r t:  rev v5.030-163-g22009a7c0
- Verilator: Built from 0.032 MB sources in 3 modules, into 0.018 MB in 6 C++ files needing 0.000 MB
- Verilator: Walltime 0.014 s (elab=0.001, cvt=0.004, bld=0.000); cpu 0.014 s on 1 threads; alloced 8.566 MB
Apptainer> verilator -y rtl/src  top.v mid.v --top-module top --cc --hierarchical
make: Entering directory '/home/awmoore/pta/verilator/test/obj_dir'
make: *** No rule to make target 'top.v', needed by 'Vtop.mk'.  Stop.
make: Leaving directory '/home/awmoore/pta/verilator/test/obj_dir'
%Error: make -C obj_dir -f Vtop_hier.mk  -j 1  hier_verilation exited with 2
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec /home/awmoore/pta/verilator/install/bin/verilator_bin -y rtl/src top.v mid.v --top-module top --cc --hierarchical
Apptainer> 
```

wsnyder, 2024-12-20:
My text had `rtl/src/top.v`, you command had just `top.v`.

(I used rtl/ instead of build/ because cmake uses build/ and I .gitignore it)