verilator/verilator #5698: Hard-coded PCH filenames with .gch instead of $(CFG_GCH_IF_CLANG)

awmoore-intel, 2024-12-19:
For some reason my platform is not appending the .gch extension which is causing compilation errors.

wsnyder, 2024-12-20:
./configured with different compiler than ran.

Unfortunately I don't see a easy way to warn about this mistake, short of running the compiler every make just to check the version, which seems a slow thing to do.

awmoore-intel, 2024-12-20:
> ./configured with different compiler than ran.
> 
> Unfortunately I don't see a easy way to warn about this mistake, short of running the compiler every make just to check the version, which seems a slow thing to do.

Just to clarify. I don't have clang installed. I compiled with g++ and then g++ was also used by verilator.

wsnyder, 2024-12-20:
Oh, did you use a different gcc version, or same for configure and run?

If same, did the `-o` work?

awmoore-intel, 2024-12-20:
> Oh, did you use a different gcc version, or same for configure and run?
> 
> If same, did the `-o` work?

No, there's only 1 gcc installed - version `g++ (Ubuntu 11.4.0-2ubuntu1~20.04) 11.4.0`
Not sure what you mean by `-o` working. The issue is the file written did not have .gch appended, but the tools in verilator was looking for suffix `h.fast.gch`

wsnyder, 2024-12-20:
From #5699  I asked if this works?  (Edited)

    echo "int main() { return 0; }" > a.cpp
    clang++ a.cpp -o a.fast.gch

wsnyder, 2024-12-20:
Really edited...

From https://github.com/verilator/verilator/pull/5699 I asked if this works?

    echo "#include <string>" > a.h
    gcc++ a.h -o a.fast.pch

wsnyder, 2024-12-20:
Er, .gch that is.

    echo "#include <string>" > a.h
    gcc++ a.h -o a.fast.gch

awmoore-intel, 2024-12-20:
> echo "#include <string>" > a.h
> gcc++ a.h -o a.fast.gch

This compiles just fine..

Maybe it's something to do with this:
https://github.com/verilator/verilator/blob/master/configure.ac#L600

My output of g++ --help is the below and doesn't contain the pch thing

```
Usage: g++ [options] file...
Options:
  -pass-exit-codes         Exit with highest error code from a phase.
  --help                   Display this information.
  --target-help            Display target specific command line options.
  --help={common|optimizers|params|target|warnings|[^]{joined|separate|undocumented}}[,...].
                           Display specific types of command line options.
  (Use '-v --help' to display command line options of sub-processes).
  --version                Display compiler version information.
  -dumpspecs               Display all of the built in spec strings.
  -dumpversion             Display the version of the compiler.
  -dumpmachine             Display the compiler's target processor.
  -print-search-dirs       Display the directories in the compiler's search path.
  -print-libgcc-file-name  Display the name of the compiler's companion library.
  -print-file-name=<lib>   Display the full path to library <lib>.
  -print-prog-name=<prog>  Display the full path to compiler component <prog>.
  -print-multiarch         Display the target's normalized GNU triplet, used as
                           a component in the library path.
  -print-multi-directory   Display the root directory for versions of libgcc.
  -print-multi-lib         Display the mapping between command line options and
                           multiple library search directories.
  -print-multi-os-directory Display the relative path to OS libraries.
  -print-sysroot           Display the target libraries directory.
  -print-sysroot-headers-suffix Display the sysroot suffix used to find headers.
  -Wa,<options>            Pass comma-separated <options> on to the assembler.
  -Wp,<options>            Pass comma-separated <options> on to the preprocessor.
  -Wl,<options>            Pass comma-separated <options> on to the linker.
  -Xassembler <arg>        Pass <arg> on to the assembler.
  -Xpreprocessor <arg>     Pass <arg> on to the preprocessor.
  -Xlinker <arg>           Pass <arg> on to the linker.
  -save-temps              Do not delete intermediate files.
  -save-temps=<arg>        Do not delete intermediate files.
  -no-canonical-prefixes   Do not canonicalize paths when building relative
                           prefixes to other gcc components.
  -pipe                    Use pipes rather than intermediate files.
  -time                    Time the execution of each subprocess.
  -specs=<file>            Override built-in specs with the contents of <file>.
  -std=<standard>          Assume that the input sources are for <standard>.
  --sysroot=<directory>    Use <directory> as the root directory for headers
                           and libraries.
  -B <directory>           Add <directory> to the compiler's search paths.
  -v                       Display the programs invoked by the compiler.
  -###                     Like -v but options quoted and commands not executed.
  -E                       Preprocess only; do not compile, assemble or link.
  -S                       Compile only; do not assemble or link.
  -c                       Compile and assemble, but do not link.
  -o <file>                Place the output into <file>.
  -pie                     Create a dynamically linked position independent
                           executable.
  -shared                  Create a shared library.
  -x <language>            Specify the language of the following input files.
                           Permissible languages include: c c++ assembler none
                           'none' means revert to the default behavior of
                           guessing the language based on the file's extension.

Options starting with -g, -f, -m, -O, -W, or --param are automatically
 passed on to the various sub-processes invoked by g++.  In order to pass
 other options on to these processes the -W<letter> options must be used.

For bug reporting instructions, please see:
<file:///usr/share/doc/gcc-11/README.Bugs>.
Apptainer> g++ --help | grep pch
Apptainer> 
```

and  my build folder has this
```
grep -r GCH .
./include/verilated.mk:CFG_GCH_IF_CLANG = 
./include/verilated.mk:VK_PCH_I_FAST = $(CFG_CXXFLAGS_PCH_I) $(VM_PREFIX)__pch.h.fast$(CFG_GCH_IF_CLANG)
./include/verilated.mk:VK_PCH_I_SLOW = $(CFG_CXXFLAGS_PCH_I) $(VM_PREFIX)__pch.h.slow$(CFG_GCH_IF_CLANG)
Apptainer> 
```

but there's other code in the repo that assumes the .gch suffix and that's why it fails. the PR makes use of the variabble CFG_GCH_IF_CLANG instead of hardcoding the suffix

wsnyder, 2024-12-20:
The point of CFG_GCH_IF_CLANG was because clang wrote to a .gch but when reading you need to drop the .gch on the filename.

You confirmed that gcc is adding the .gch on write (the `-o`), perhaps it is now matching clang in that it does not want it on reading, can you try an experiment to confirm that?  If so, changing the configure.am to do a better experiment (create a .h file, compile, then try to use it with and without .gch) to set CFG_CGH_IF_CLANG should fix it, and no makefile change needed.

wsnyder, 2025-03-30:
I'm assuming #5724 resolved this, if not please run on git master version and reply to questions above and reopen.

closed 2025-03-30 by wsnyder