verilator/verilator #5701: The simulation efficiency of Verilator varies significantly under different timescales. I would like to know the reason.

IceMilkTea, 2024-12-21:
Dear Author,

I have two questions regarding the efficiency of using Verilator and would like to seek your guidance:

When the timescale of a simulation project is set to 1ns/100ps, Verilator performs very well. However, when the timescale is changed to 1ns/1ps, the simulation efficiency drops almost 100 times. Interestingly, this issue does not seem to occur with Modelsim. Could you explain the reason behind such a significant reduction in efficiency?

I understand that Verilator, during simulation, performs predictive evaluations based on the signals in the sensitivity list. However, instead of placing this evaluation process in the value-update phase, it is implemented within a while loop and executed repeatedly within each time slice. This approach seems to differ somewhat from the description of simulation algorithms in the Verilog standard. Could you elaborate on the rationale behind this design choice?

I look forward to your response and sincerely appreciate your guidance. Thank you very much.

Attached below is the relevant description of the simulation algorithm from the 1800-2023 standard for reference:
execute_simulation {
      T = 0;
      initialize the values of all nets and variables;
      schedule all initialization events into time zero slot;
      while (some time slot is nonempty) {
                   move to the first nonempty time slot and set T;
                  execute_time_slot (T);
        }
}
execute_time_slot {
        execute_region (Preponed);
        execute_region (Pre-Active); 
        while (any region in [Active ... Pre-Postponed] is nonempty) {
                  while (any region in [Active ... Post-Observed] is nonempty) {
                           execute_region (Active);
                          R = first nonempty region in [Active ... Post-Observed];
                          if (R is nonempty)
                          move events in R to the Active region;
                  }
                 while (any region in [Reactive ... Post-Re-NBA] is nonempty) {
                           execute_region (Reactive);
                            R = first nonempty region in [Reactive ... Post-Re-NBA];
                           if (R is nonempty)
                             move events in R to the Reactive region;
                             }
                 if (all regions in [Active ... Post-Re-NBA] are empty)
                            execute_region (Pre-Postponed);
                            }
            execute_region (Postponed);
}
**execute_region {
          while (region is nonempty) {
                     E = any event from region;
                     remove E from the region;
                     if (E is an update event) {
                                 update the modified object;
                                schedule evaluation event for any process sensitive to the object;
                      } else { /* E is an evaluation event */
                                 evaluate the process associated with the event and possibly
                                schedule further events for execution;
                                }
           }
}**

wsnyder, 2024-12-21:
You probably have some event that occurs every time precision tick, so more ticks, the slower.  Use profiling to determine why.

Verilator's choices generally perform better, but it may require model changes to see that.  Best performance is with --no-timing, or no #delays/forks etc.

wsnyder, 2025-01-02:
Also suggest running the profiling tools described in the documentation.

Closing as not sure what else to suggest, if you have more questions feel free to post.

closed 2025-01-02 by wsnyder