verilator/verilator #5702: Multiple verilator instances of SystemC models in one single simulation

robertbu, 2024-12-23:
Hi,
I have multiple RTL sources from different vendors. It's not possible to integrate the RTL designs into a single one and then verilate it.
Is that possible for them to provide their verilated models each, then instantiate the verilated SystemC models in one single SystemC simulation?

Thanks
Robert

wsnyder, 2024-12-23:
Yes, build the models and connect in  SystemC.  The Verilator --lib-create  is an all-Verilog option too though likely a rougher process, as requires Verilator for integration and is non-standard.  I'd suggest SystemC first if that works.

robertbu, 2024-12-25:
Thanks for the quick response. If I instantiate two verilated models for two different RTL module in one SystemC simulation, that works. However, If I instantiate the same verilated model twice in one SystemC simulation, the AXI bus interface hangs. I'll dig further.