verilator/verilator #5703: $urandom seeding

pis7, 2024-12-23:
I have testbenches set up with multiple test cases, where I want to be able to provide a seed to each testcase such that $urandom generates its random numbers based off of that seed. I have a methodology working in Synopsys VCS that is essentially the following:

input: seed

integer dummy_rand = $urandom(seed);

(use $urandom as many times to generate the same sequence of random numbers for a given seed).

However, when running the same testbench in Verilator, this seeding on the dummy_rand line is ignored and different sequences of random numbers are generated anyway. I was hoping someone may be able to provide insight on how to accomplish the behavior I am seeing in VCS (the intended behavior from the start) with Verilator. Thanks.

wsnyder, 2024-12-23:
Perhaps a bug, can you please post a complete example showing the issue, thanks.

pis7, 2024-12-23:
@wsnyder Hi Wilson, thanks so much for your quick response! Let me try to provide as much detail as possible on the issue - please let me know if you need more information.

Here is a task I have in my testbench representing a test case. Note that for this particular example, giving a different seed to the test case actually causes the exact same sequence of values to be produced every time, which is actually the opposite of what I said before. I think for what I was talking about above, I actually passed in the seed plus some incremented index for each call to try to mitigate this issue, but which resulted in the original issue I posted. Whereas here, I only seed $urandom once and then call $urandom without arguments for subsequent calls. I think these two issues are related, however, as I believe the way $urandom is implemented is ignoring the seed.

```
task automatic test_basic_async_reset (
  string name,
  integer clk_period = -1,
  integer rst_delay  = -1,
  integer seed       = 32'(get_system_time_seed() + $time)
);
  integer dummy_rand = $urandom(seed);

  if (clk_period == -1) clk_period = 2 + ($urandom % (p_max_clock_period - 1));
  if (rst_delay  == -1) rst_delay  = 1 + ($urandom % (p_max_rst_delay));

  $display("clk_period = %0d, rst_delay = %0d, seed = %0d", clk_period, rst_delay, seed);

  // Initialize test case and ensure async reset is propagated through synchronizer
  tb.test_case_begin (
    name,
    clk_period,
    rst_delay,
    seed
  );
  
  // steady state - all zeros
  @(posedge clk);
  @(posedge clk);
  @(negedge clk);
  tb.test_case_check(0, dut.reg1, "reg1");
  tb.test_case_check(0, dut.reg2, "reg2");
  tb.test_case_check(0, reset, "reset");

  #(1 + ($urandom % (p_max_wait_cycles)));

  // async reset - all ones
  @(negedge clk);
  tb.do_reset(1);
  #1;
  tb.test_case_check(1, dut.reg1, "reg1");
  tb.test_case_check(1, dut.reg2, "reg2");
  tb.test_case_check(1, reset, "reset");

  // wait for one clock cycle, reg 1 should have propagated a zero
  @(posedge clk);
  @(negedge clk);
  tb.test_case_check(0, dut.reg1, "reg1");
  tb.test_case_check(1, dut.reg2, "reg2");
  tb.test_case_check(1, reset, "reset");

  // wait for one clock cycle, reg 2 should have propagated a zero, reset is zero
  @(negedge clk);
  tb.test_case_check(0, dut.reg1, "reg1");
  tb.test_case_check(0, dut.reg2, "reg2");
  tb.test_case_check(0, reset, "reset");

  #100;
endtask
```

In all instances, I call the test case task with only the name argument, such that the three other integer arguments are all their default values. Note that I print out these values after randomization to illustrate the issue.

Compiling and running the test in Verilator, I get the same values for every run. Below are some example outputs (note how the seed changes but the value does not):
`clk_period = 21, rst_delay = 21, seed = 1734982119`
`clk_period = 21, rst_delay = 21, seed = 1734982521`
`clk_period = 21, rst_delay = 21, seed = 1734982529`

Running the exact same test after compiling with Synopsys VCS, the following expected outputs are obtained:
`clk_period = 15, rst_delay = 62, seed = 1734982564`
`clk_period = 31, rst_delay = 49, seed = 1734982571`
`clk_period = 26, rst_delay = 31, seed = 1734982578`

If, still in VCS, I were instead to pass in a desired seed to the test case instead of using the custom get_time() method, the same sequence of values are produced as expected:
`clk_period = 10, rst_delay = 100, seed = 1`
`clk_period = 10, rst_delay = 100, seed = 1`
`clk_period = 10, rst_delay = 100, seed = 1`

I appreciate your help with this, thanks!

wsnyder, 2024-12-23:
Can you please provide a _standalone_ minimized example that is runnable (a task is not runnable by itself, and all the posedges are likely unrelated and can be simplifed out), with the command line and what you expect as output?

Also avoid anything other than simple variable reference in the urandom call, as that may be misleading due to side effects.

Alternatively and better modify one of the existing test_regress/t/*.v files that calls urandom to show the problem.

pis7, 2025-01-11:
Hi Wilson, sorry for the delayed response - holidays were a bit crazy and we had our tapeout deadline this past Wednesday. Here is an independent example that is runnable:

```
module Top();

initial begin
  integer seed, dummy_rand;
  if (!$value$plusargs("seed=%d", seed)) $finish(0); // get seed
  dummy_rand = $urandom(seed); // seed the generator

  $display("Seed = %0d", seed);
  $display("Random value 1 = %0d", $urandom);
  $display("Random value 2 = %0d", $urandom);
  $display("Random value 3 = %0d", $urandom);

  $finish(0);
end

endmodule
```

When running with VCS, I get the following outputs where every time I run the executable with the provided seed, the same random values are produced:

Seed = 1234567890
Random value 1 = 2188305506
Random value 2 = 3717059908
Random value 3 = 3287696678

Seed = 66666666
Random value 1 = 4054588818
Random value 2 = 1872905237
Random value 3 = 1311434289

When running with Verilator, I get the following outputs where every time I run the executable with the provided seed, the same random values are produced:

Seed = 1234567890
Random value 1 = 2667435266
Random value 2 = 186539420
Random value 3 = 2037806020

Seed = 66666666
Random value 1 = 2667435266
Random value 2 = 186539420
Random value 3 = 2037806020

It seems like Verilator is ignoring the seeding of the generator...

wsnyder, 2025-01-11:
It removed the seed because of the dummy_rand being unused, calling it as a task worked.

pis7, 2025-01-11:
Thanks so much, Wilson. Looks like it is working perfectly now!