verilator/verilator #5704: Passing a string in a parameter with an implicit type

punzik, 2024-12-24:
An error occurred when hierarchically passing a string in a parameter with an implicit type in the last module.

Example:

```verilog
module string_param_0 #(parameter S = "");
    initial $display("S = '%s'", S);
endmodule

module string_param_1 #(parameter string S = "");
    string_param_0 #(.S(S)) u_sp0();
endmodule

module test;
    string_param_1 #(.S("Greetings from the old shoes")) u_sp1();

    initial #1 $finish;
endmodule
```

Expected output:
```
S = 'Greetings from the old shoes'
...
```

Actual output:
```
S = '                            '
...
```

If you specify the parameter type `string` in the `string_param_0` module, the error does not occur.

Verilator version: 72a47e16c124e3c0f93c16d5049e0db79764bc01

OS: NixOS 24.11

wsnyder, 2025-01-02:
This is uglier than it might seem, the implicit type means the data type needs to be resolved during elaboration and might be different for each instantiation.  I'll need to get back to this.

```
`define stop $stop
`define checkh(gotv,expv) do if ((gotv) != (expv)) begin $write("%%Error: %s:%0d:  got='%h' exp='%h'\n", `__FILE__,`__LINE__, (gotv), (expv)); `stop; end while(0);
`define checks(gotv,expv) do if ((gotv) != (expv)) begin $write("%%Error: %s:%0d:  got='%s' exp='%s'\n", `__FILE__,`__LINE__, (gotv), (expv)); `stop; end while(0);

module param_0 #(bit STR = 0, parameter S = "");

   initial begin
      if (STR == 0) begin
         `checkh(S, 12);
      end
      if (STR == 1) begin
         `checks(S, "Greetings");
         $write("*-* All Finished *-*\n");
      end
   end

endmodule

module string_param_1 #(parameter string S = "");
   param_0 #(.STR(1), .S(S)) u_p0();
   param_0 #(.STR(0), .S(12)) u_p12();
endmodule

module test;
   string_param_1 #(.S("Greetings")) u_sp1();
endmodule
```

wsnyder, 2025-08-01:
#5616 is another example of this.

```
// DESCRIPTION: Verilator: Verilog Test module
//
// This file ONLY is placed under the Creative Commons Public Domain, for
// any use, without warranty, 2024 by Dan Petrisko
// SPDX-License-Identifier: CC0-1.0

module test_rom #(
    parameter FILENAME_P
);

  logic [3:0] mem[0:0];
  initial begin
    $display("\$readmem of %s", FILENAME_P);
    $readmemb(FILENAME_P, mem);
    if (mem[0][0] != 1'b1) $stop;
    if (mem[0][1] != 1'b0) $stop;
    if (mem[0][2] != 1'b1) $stop;
    if (mem[0][3] != 1'b0) $stop;
  end

endmodule

module t;

  // %Warning: $readmem file not found
  localparam string STR_TRACE_FILE_LP = "t/t_sys_readmem_str.mem";
  test_rom #(.FILENAME_P(STR_TRACE_FILE_LP)) str_rom ();

  // Successfully finds file
  localparam DEF_TRACE_FILE_LP = "t/t_sys_readmem_str.mem";
  test_rom #(.FILENAME_P(DEF_TRACE_FILE_LP)) def_rom ();

  initial begin
    #1000;
    $write("*-* All Finished *-*\n");
    $finish;
  end

endmodule
// FIXME Same as issue #5704
// FIXME t_sys_readmem_str.mem contains: 1010

```