verilator/verilator #5705: Assignment pattern with too many elements

B83C, 2024-12-29:
Thanks for taking the time to report this.
I am getting this error:
`
%Error: ../src/pipeline_unit.sv:24:24: Assignment pattern with too many elements
                                     : ... note: In instance 'tb2.rv32.pu1'
   24 |   pbuffer #(7, word_t, '{
`
But it works just fine with vivado...

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
```sv
  pbuffer #(7, word_t, '{
        '{F, W},
        '{D, D},
        '{M, M},
        '{E, E},
        '{E, E},
        '{E, E},
        '{W, W}
  }) rp ( clk, rst_n, p.stalls, p.flushes, '{
        ....
```
Definition of pbuffer:
```sv
module pbuffer #(
    parameter Nitems = 2,
    parameter type T = word_t,
    parameter stage_t meta [Nitems-1:0][1:0] = '{'{F, W}, '{F, W}}
) (
    ...
```

What 'verilator' command line do we use to run your example?
verilator --binary -j 16  --build tb2.sv --trace-fst  -I../src/tb/ -I../src/ --timing --trace-max-array 700 --Mdir ./obj_tb2

What 'verilator --version' are you using?  Did you try it with the git master version?
git master version

What OS and distribution are you using?
Arch linux

May we assist you in trying to fix this in Verilator yourself?
yes

(Please avoid attaching screenshots that show text - you can convert images to text using e.g. https://ocr.space)

wsnyder, 2025-01-02:
Can you please convert this into an example that is self contained and passes when put directly into vivado?

wsnyder, 2025-01-25:
Ping: Can you please convert this into an example that is self contained and passes when put directly into vivado?

closed 2025-02-15 by wsnyder