verilator/verilator #5706: make-hello-c test fails on armel

ldoolitt, 2024-12-29:
Debian's CI for verilator/5.030-2 faults on armel doing make-hello-c
https://tracker.debian.org/pkg/verilator
https://ci.debian.net/packages/v/verilator/testing/armel/55575533/
(but passes on seven other arches!)
```
obj_dir/Vtop
Vtop: /usr/share/verilator/include/verilated_imp.h:97: VerilatedEvalMsgQueue::VerilatedEvalMsgQueue(): Assertion `atomic_is_lock_free(&m_depth)' failed.
```

Much better than the pre-issue-#3891 days, eh?  But still enough to get verilator kicked out of Debian, if it's not resolved soon.

I can try digging in a little, but I don't have access to a real armel machine.  Maybe I can get something running on qemu?
Posting now, in case an actual arm expert can help out faster than me.

wsnyder, 2024-12-29:
This is likely a compiler/libc problem... Searching on the error indeed: https://forums.raspberrypi.com/viewtopic.php?t=287547

Probably need the right -m flag?

wsnyder, 2025-02-15:
@ldoolitt did Verilator on Armel get resolved?

ldoolitt, 2025-02-15:
> [@ldoolitt](https://github.com/ldoolitt) did Verilator on Armel get resolved?

Yes and no.  Latest build log says
```
autopkgtest [17:22:22]: testing package verilator version 5.032-1
make-hello-c         SKIP Test declares architecture as not supported: armel
```
but with that workaround, at least the package is no longer kicked out of testing.
Ahmed El-Mahmoudy, the Debian package maintainer, added that and a number of other small changes to get the package ready for the upcoming Trixie release.
I don't have access to an armel machine, so I can't add more detail on the make-hello-c problem; sorry.

wsnyder, 2025-02-15:
Thanks. I'm personally good with that skip, if someone really needs that arch on Verilator, they can reopen and do the work ;)

closed 2025-02-15 by wsnyder