verilator/verilator #5707: Crash when accessing array of integers

YuriNenakhov, 2025-01-02:
Minimal example demonstrating the bug:

```
function integer fun;
integer array[0:0];
begin
    array[0] = 10;
    fun = array[0];
end
endfunction

module test ();
begin
    localparam something = fun();
end
endmodule
```

Changing to `fun = 10; //array[0];` works.

Command used:

`docker run -it -v/tmp:/tmp --rm verilator/verilator:v5.032 --cc /tmp/test.v`

OS: Ubuntu 18.04.6 LTS with Docker version 24.0.2

This is a regression as it used to work before.

wsnyder, 2025-01-02:
@gezalore this seems to appear because V3Simulate is fetching value from a AstConst, which conflicts with your changes to track const usage.  Might you take a look? Thanks.

wsnyder, 2025-01-03:
@YuriNenakhov thanks for the good test, & @gezalore thanks for the good fix.

PR #5709 (2025-01-03): Fix invalid lookup on array initializer (#5707)
Use a generational allocator for reusing AstConst across V3Simulate::clear(), instead of using user1 (which is also used to store values of nodes).

Also fix invalid lookup on array initializer

Fixes #5707