verilator/verilator #5711: simulation of RTL that contains x and z

mano009, 2025-01-03:
Hi Team,

I am trying to simulate a design, and this design contains x and z Comparisions like CEN_ === 1'bz. Can you please help me on how to simulate the designs that contains x and z?

Currently I am using below command to simulate:
verilator --binary sram_tb.sv sram_sp_hde.v -Wno-fatal --x-initial unique --trace 

My verilator version is 5.028 2024-08-21

Best regards,
Aravind

wsnyder, 2025-01-03:
That's a very general question, what specific problem are you having?

Verilator is a 2-state simulator, so comparison to X will never work.  Comparisons to z only work for "tri" nets, e.g. internal tristate buses or pad drivers.  Please see the documentation.

mano009, 2025-01-04:
I have an if condition in my design like below. 

if (CEN_ === 1'bx || CEN_ === 1'bz) begin
 // my logic
end

mylogic inside if condition is triggered when I use iVerilog. When I use verilator this code is not triggered. 
I understand that this particular code will not trigger because verilator is 2-state simulator. Is the any alternative for this particular code to trigger( like using 'ifdef )?

wsnyder, 2025-01-04:
If you use 'tri CEN_' and somewhere in the logic drive it with e.g. `CEN_ = 1'bz" you can trigger this.  Otherwise not.

closed 2025-01-04 by wsnyder

mano009, 2025-01-05:
Understood. Thanks for your response 👍