verilator/verilator #5721: Can non-standard Force / Release behavior be changed?

seldridge, 2025-01-09:
Is Verilator's non-standard implementation of force/release (discussed [here](https://verilator.org/guide/latest/languages.html#language-keyword-limitations) and summarized as Verilator implements `force` like `$deposit` as opposed to a continuous assignment) something that is intentional or something that could be changed?

The background is that the CIRCT project is running afoul of this because to emit Verilog that works with both VCS and Verilator (and other simulators) we would need to specialize the emission of `force`/`release` just for Verilator. I think specializing the emission is doable, but it would help us out if Verilator was spec-compliant in this regard. 😉 

I couldn't find the discussion about this originally, but support was added in https://github.com/verilator/verilator/commit/0c3ffa1841cc2db3901928bc63e011ce3a08436b and then documented (or changed?) to the non-standard behavior in https://github.com/verilator/verilator/commit/539c9d4c63852ed2f259d1180f6e6832414d4016.

wsnyder, 2025-01-09:
There was no change to make less-standard behavior I am aware of, it was just documenting what worked, and some other changes to get closer to standard in other respects.

The problem is that the force changes how the scheduling needs to work in the model, and this is all statically determined.  I don't see us likely changing this in the short/medium term.  If you have ideas to otherwise make a pull to make this more compatible or otherwise help your use case we can discuss.

seldridge, 2025-01-14:
I thought I understood the Verilator semantic for force. However, I'm now getting confused.

What is the Verilator semantic for force?

The documentation indicates the following:

> Verilator instead evaluates the current value of b when the force statement is executed, and forces a to that value, without updating it until a new force or release statement is encountered that applies to a.

This led me to believe that the value of `a` would hold the value of `b` until a release. However, Verilated code is reflecting changes to the value `a` after a force which would seem to contradict this. It's closer to a `$deposit` semantic. The example I'm working with is the following:

``` verilog
module Foo();

  reg [2:0] a = 3'b000;

  initial begin
    a = 3'b001;
    $display(a);

    force a = 3'b010;
    $display(a);

    a = 3'b011;
    $display(a);

    release a;
    $display(a);

    a = 3'b100;
    $display(a);

    $finish;
  end

endmodule
```

Verilator (`Verilator 5.026 2024-06-15 rev v5.026`) produces the following prints (`verilator Foo.sv --binary && ./obj_dir/VFoo`):

```
1
2
3
0
4
```

I've tested this with the five simulators available on EDA Playground and they all produce:

```
1
2
2
2
4
```

I've considered having CIRCT emit a Verilator-compatible `force`, i.e., adding some additional code guarded with `` `ifdef VERILATOR `` that would replicate the spec-compliant semantic using the Verilator semantic. However, how I think Verilator works, this would be more complicated than I anticipated. (The force needs to be refreshed whenever the value could be overwritten with another assignment or the force must be put into an always block with a sensitivity list that is both the `b` forced value and any value which could try to write to `a`.)

wsnyder, 2025-01-16:
Analysis:

```
    a = 3'b001;
    $display(a);  // Correctly 1

    force a = 3'b010;
    $display(a);  // Correctly 2

    a = 3'b011;
    $display(a);  // Verilator 3 Others 2 - this appears due to V3Life making a bad optimization where it knows from above a=3 so substitutes it in.   (The V3Force conversion runs later.)  Pull to fix this by e.g. disabling it on any potentially forced net is welcome.

    release a;
    $display(a);  // Verilator 0 Others 2 - appears a bug in the V3Force release code - again pull welcome to fix.

    a = 3'b100;
    $display(a); // Correctly 4
```

seldridge, 2025-01-16:
Thanks for the analysis and pointers on where to look to fix this, @wsnyder! I can't promise anything... but if I get so motivated to fix it, I'll take a look or have someone here take a look. Thanks again.

closed 2025-03-04 by wsnyder