verilator/verilator #5723: Encountered process that should've been resumed at an earlier simulation time. Missed a time slot?

Timmmm, 2025-01-09:
Verilator version: 4503b627d863d12e337c504919f5d5ba89215b42

OS: RHEL 8

I get this error:

```
%Error: /usr/local/include/verilated_timing.cpp:83: %Error: Encountered process that should've been resumed at an earlier simulation time. Missed a time slot?
```

I am using `--timing` and have some tasks, so it's probably something to do with that. Unfortunately it's in a very large proprietary project and we haven't had Verilator working for some time (I'm trying to fix it), so it's very difficult for me to know where the problem is and make a small reproducer.

But I thought it was worth putting this here anyway. Something is clearly broken, and maybe other people will find this from google, or have an idea for how to debug it.

gezalore, 2025-01-09:
Can't offer a solution, but this might be a good candidate for sv-bugpoint?

https://antmicro.com/blog/2024/09/sv-bugpoint-for-improved-debugging-in-sv-tools

https://github.com/antmicro/sv-bugpoint

Timmmm, 2025-01-09:
Ah cool I didn't know about that. Might have a go at some point, but I suspect the size of this project is a fair bit beyond what is practical for that tool. E.g. it takes about 10 minutes to compile the C++ code, and it isn't pure SV - we have DPI etc.

wsnyder, 2025-02-25:
Closing until we can get an isolated test case.

closed 2025-02-25 by wsnyder

joecrop, 2025-10-15:
I have been able to isolate this to happening only when I add the `--x-initial-edge`.