verilator/verilator #5726: Is $dumpvars(1,foo.bar) working in system verilog right now?

pbreuer, 2025-01-10:
I can get variable-specific dumps from the Vtestbench_main.cpp file just fine via

contextp->traceEverOn(true);
topp->trace(trace_object, 1);
trace_object->dumpvars(1, "TOP.testbench.dut.core.ifu.PCF");
trace_object->open("/tmp/sim.vcd");
...
       topp->eval();
        VL_PRINTF("time: %dns\n", contextp->time());
        trace_object->dump(contextp->time());

and the sim.vcd correctly contains just a dereference for the one wire, (128-bit) PCF.

Yes, I am compiling with --trace for verilator.

If I remove those lines and leave the verilator generated .cpp main file alone, and instead put
in the testbench.sv file init section:

  $display("[%0t] Tracing to /tmp/sim.vcd: ...\n", $time);
  $dumpfile("/tmp/sim.vcd");
  $dumpvars(1, dut.core.ifu.PCF);

then "nothing" happens, for various forms of "nothing". 

That is, depending on the hierarchy expansion value, I either get every single variable in the whole of existence tracked ("0") or none (anything else), modulated by whatever I write instead of dut.core.ifu.PCF.

Moreover, if instead of dut.core.ifu.PCF I name something there that does not exist I don't get a compile error, I get behavior like that for "0" just described, no matter what hierarchy value.

To me it seems those references are translated as strings before linting, and those are the wrong strings, somehow, or
else the translation is to something that just takes no notice anyway. Certainly there is a dumpvars in c++ to translate to,
as the version where it works from the main() routine shows.

Is there some touch of magic I am missing?

Regards

PTB

(whatever verilator is in debian ...  Verilator 5.027 devel rev v5.026-130-g1b15509a9)

wsnyder, 2025-01-10:
The argument is ignored, see the documentation https://verilator.org/guide/latest/languages.html

If you'd like to contribute a pull to fix this it would be welcome, otherwise we are unlikely to support this.

pbreuer, 2025-01-10:
So you're saying it doesn't work? That's a bug. The documentation online actually says this:

_Pass the [--trace](https://veripool.org/guide/latest/exe_verilator.html#cmdoption-trace) option to Verilator. Then you may use $dumpfile and $dumpvars to enable traces, **the same as with any Verilog simulator**, although Verilator ignores the arguments to $dumpvars. See examples/make_tracing_c in the distribution._

(my bold). Presumably somebody has added recently "although Verilator ignores the arguments  to $dumpvars"!

Unfortunately, that is not how to alter a document. It now contradicts itself, since that is NOT "the same as with any Verilog simulator" (see https://www.referencedesigner.com/tutorials/verilog/verilog_62.php on that) . In a contradictory document, one is at liberty to choose what one likes or wishes, it's undefined.

Moreover, in the very next paragraph, it says: "Pass the [--trace](https://veripool.org/guide/latest/exe_verilator.html#cmdoption-trace) option to Verilator, and in your top-level sc_main(), call Verilated::traceEverOn(true). **Then you may use $dumpfile and code:$dumpvars to enable traces, as with any Verilog simulator**; see the non-SystemC example in ..."

(my bold) and there it specifically says that $dumvars works just as in (system)Verilog under simulation anywhere (including Verilator! Mind, the claim was rather too sweeping .. I can write a simulator that does not do that and will not feel bound by that sentence, a lawyer would laugh).

Let's not get to the examples scattered around the world wide web thang which show it working. There was a report from 2008 I saw which said it wasn't working, but what's the situation? Was it working and it broke? Or has it never worked? 

As to me doing a patch, I am not a C++ programmer, I hate object oriented code as written by people who cannot write classes that correspond to things I already know _perfectly_ and therefore do not obscure nasty surprises or unknowns, and I have one thousand million other things to do, like debug a processor written IN system verilog instead of some higher order language like DEITY intended.  I'm not even a programmer, come to that.

However, I am an expert at everything supposedly,  including code transformation, being the author of a higher order compiler compiler used  worldwide for decades, as well as a lazy functional programming language with duration and numerous other nasty things both at high and low level, so if this silly thing annoys me enough, I MAY do it, even though I know nothing whatsoever about your mountain of specialised code, not even what it is supposed to be doing (putting up a schedule of planned events and modifying it, then moving time along to the next planned event, probably).

So if you feel like giving me some actual CLUE as to where to look that beats what I would do with grep, I might get around to it.

Do you want a bug filed as a documentation bug or a code bug, meanwhile?
(it's clearly a code bug, since a fix with a patch is desired!)

Many thanks.

PTB

wsnyder, 2025-01-10:
If you're not familiar with C programming this will probably be a frustrating project for you, so understand not doing it!

It's something that is unsupported, if you feel better calling it an unsupported bug, that's fine, we have many issues here asking for unsupported language features to be supported.  BTW this has never been supported.

If the wording of the documentation bothers you please submit a pull request to reword how you think is better, thanks.

pbreuer, 2025-01-10:
I'm very happy with C, not C++.
So it's never been implemented! Curious. It seems to me to be easy, so am I thinking too little about it? Anyway, you should want to point me at the bit of your codebase that does translations of leaf constructs in the syntax tree, probably via the mountain of opaque embedded integrated mutually dependent friend classes that I pre-squawked about above.
A theoretically worse problem is that I don't know system verilog either, this time in the "at all" sense, but I don't need to know it. Your parser already does. I can very probably hack it in some awful way completely incompatible with project guidelines.
Cut down my work by pointuing me somewhere _small_ in the code, and I'll see if I get annoyed enough to do something. Insulting my manhood and manners might help.

So what do you want me to do about modifying Vtestbench__main.cpp? Every time I rewrite it to include tracing, verilator overwrites it again to remove my changes. I was going to add patching it into the make sequence, but if you have some off the shelf solution ready, I'm all ears.

(I'll leave the document alone for now, to remind me via annoying me again if I see it).

Regards

PTB

wsnyder, 2025-01-10:
In most use cases people trace everything rather than specialize, so you are the first person to want this.  There are lots of "easy" features that add up to years of work, so we do the popularly needed ones first.

As to code, the parser (verilog.y) would need to recognize the scope, and pass it into V3Link where it gets resolved to a AstScopeName, then the code that outputs the tracing needs to make that into a string.  This isn't trivial, there's no current place where we handle scopes like this - the VPI feature this relies on is not supported.

>So what do you want me to do about modifying my Vtestbench__main.cpp?

You are not supposed to edit any generated files.  If you want your own main, don't use the --main flag and put your .cpp file on the command line.

pbreuer, 2025-01-10:
The current scope must be known somewhere during translation, else how are relative names with dot references in resolved? I've seen them! But never mind, ... hang that, one can always insist on absolute dot references starting at TOP just like in the C++ call, and you can document that :-).

pbreuer, 2025-01-11:
It looks like most of the dump stuff is deliberately broken/incomplete. The relevant code is in V3EmitCFunc.h, where the parse of the dump statement is translated to some kind of C++. One should have been able to do better than this by way of translation! (I have extended the parse of dumpvars to include the "vars", previously the parser just ignored them, so info about them never even reached this point). I'll see what I can do about it ..

```
    void visit(AstDumpCtl* nodep) override {
        switch (nodep->ctlType()) {
        case VDumpCtlType::FILE:
            putns(nodep, "vlSymsp->_vm_contextp__->dumpfile(");
            emitCvtPackStr(nodep->exprp());
            puts(");\n");  
            break;      
        case VDumpCtlType::VARS:
            // We ignore number of levels to dump in exprp()
            ...
                putns(nodep, "vlSymsp->_traceDumpOpen();\n");
            ...
            break; 
        case VDumpCtlType::NVARS:
             .........        <-------------------------------- to insert new stuff here for $dumpvars(expr,cStrList)
            break; 
        case VDumpCtlType::ALL:
            // $dumpall currently ignored
            break;
        case VDumpCtlType::FLUSH:
            // $dumpall currently ignored; would need rework of VCD single thread,
            // or flag we pass-through to next eval() iteration
            break;
        case VDumpCtlType::LIMIT:
            // $dumplimit currently ignored
            break;
        case VDumpCtlType::OFF:
            // Currently ignored as both Vcd and Fst do not support them, as would need "X" dump
            break;

```

Your clues are

1) vlSymsp->_vm_contextp__->dumpfile( file );    currently translates  $dumpfile(file);
2) vlSymsp->_traceDumpOpen(); currently translates $dumpvars(hier);

and that was it! As the comment says, "hier" was ignored, but the form $dumpvars(hier,exprList) also got one that, with both (or however many) arguments ignored!

I believe I've now got a separate parse for $dumpvars(hier,exprList) but have restricted it to $dumpvars(hier,cStrList) for now for more control. Shrug. That's what works in C++. 

So now I need to see if I can produce some output from that, just in the form of a comment for now ... then figure what output it really should be.

WHY am I elected to do this? I don't know any C++, am not a programmer, and this looks as though it should have been well within the abilities of practically anyone working on verilator, as they would have (a) known C++, (b) been a programmer, (c) been familiar with the host of classes in verilator that I have never seen or heard of and known how to make use of them and what code one could generate that makes sense, and how to do that.

The only good points are that at least the names of things are sufficiently abbreviated that I don't get cramp writing them out and can remember what they are, as well as guess what they are. It would have been more fun if all methods were named just m1,m2,m3, etc. but heigh ho. As to the opaque hierarchy of classes, I have yet to be able to locate even a single ultimate parent for anything, thus I am unaware of what is actually in anything at all and what it can do.

pbreuer, 2025-01-12:
I see what you mean about context. main() starts off by inventing a clean context with `const std::unique_ptr<VerilatedContext> contextp{new VerilatedContext};` so there is no easy object-oriented way to communicate to main what the parser has learned.

pbreuer, 2025-01-13:
OK, I have something working (or appears to). I'll put up a patch tomorrow.
This note is just to say while I remember that the final lap was very hard. It turns out that although main() has no access to any "context", starting a fresh one, miraculously the first execution of eval() in main's for loop magically enters data from the parse context into the new context. I do not know if it is intended and I do not know how precisely, but there is a load of one-time ad-hoc initializations done in several methods that one/I would not expect , and it must be one of those. So one can call the C++ dumpvars (which works fine) after the first call to eval() with the data from the parse that has been magically picked up, then carry on with all the remaining calls of eval as normal.
One also has to close and reopen the vcd file between first and second calls of  eval, in order - apparently - to flush the dumpvars(0,top) data that it invented on its ownsome.
Yes, I have no clue. More details when I figure it out tomorrow.

The main loop now looks like this. Best compromise I could make.

```
  const std::unique_ptr<Vtestbench> topp{new Vtestbench{contextp.get()}};
  topp->trace(trace_object, 1);

 trace_object->open(VCD_FILE);

    bool loaded_dumpvars = false;

    // Simulate until $finish
    while (!contextp->gotFinish()) {
        // Evaluate model
        topp->eval();

        // eval miraculously has loaded contextp with dumpvars from source
        // (better late than never, better wrong place that no place)

        if (!loaded_dumpvars) {
            // close to clean trace object of default dumpvars notions
            trace_object->close();
            // reset the trace object with the correct list in contextp
            trace_object->dumpvars(contextp->dumpvars());

            loaded_dumpvars = true;
            // reopen the trace object with the correct list
            trace_object->open(VCD_FILE);
        }

        if (contextp->time() % 10 == 0)
          VL_PRINTF("time: %d cpu cycles\n", contextp->time() / 10);

        trace_object->dump(contextp->time());

        // Advance time
        if (!topp->eventsPending()) break;
        if (contextp->time() > TIME_LIMIT) break;
        contextp->time(topp->nextTimeSlot());
    }

    trace_object->close();
```

I didn't want to hide it in the innards of eval or dump. Best to let people know what they're doing. Conscious assent and all that. I've had enough of opaque c++ classes anyway! No clue where anything happens ...

wsnyder, 2025-01-13:
I don't understand why dumpvars can't be loaded before topp->trace is called at the top.

pbreuer, 2025-01-13:
I edited to make my 3am comment clearer. Just waking up ... 
It's the first execution of eval that transfers the CORRECT dumpvars information across into the running context (don't exactly know how, but take what you can get).
At any point before that one has to insert the information via extra  dumpvars() calls in the C++ main() itself, and it looks like one also still has to content with the dumpvars(0) info that may be in there under some conditions, don't exactly know what those. 
My opinion is that would almost be less hacky to store the dumpvars info into an extra file during parse, and insert a dumpvars_from_file() call in the main loop to pick it up. But I suspect it still would be fighting the default dumpvars info.

Another problem with that default  dumpvars(0) info is that it seems to be in a different trace object than the ones one can set up from the C++ code, so one can't get at it, at least not easily. It seems to be running in a separate thread, and it will dump its second load of everything some time after/during the main() loop finishes, as far as I can tell. The thing must have been hacked to death at some point with people adding in if(whatever) clauses to get some kind of sensible behavior out under ordinary circumstances. I'm not going to even try to figure it. I'm just saying what I observed. I even used a debugger! Me!

pbreuer, 2025-01-13:
Following your hint, I checked and the main loop can be simplified to "as usual plus one extra call of eval before it starts". Like this:

```
    const std::unique_ptr<Vtestbench> topp{new Vtestbench{contextp.get()}};
    topp->trace(trace_object, 1);

    // eval  first time loads contextp with dumpvars from source
    topp->eval();

    // load now correct dumpvars info into trace object
    trace_object->dumpvars(contextp->dumpvars());

    trace_object->open(VCD_FILE);

    // Simulate until $finish
    while (!contextp->gotFinish()) {
        // Evaluate model
        topp->eval();

        trace_object->dump(contextp->time());

        // Advance time
        if (!topp->eventsPending()) break;
        contextp->time(topp->nextTimeSlot());
    }

    trace_object->close();
```

Better? While I really appreciate the hint, I would appreciate even more the _thinking_ behind the hint, or suspicion, or whatever it is. Then I wouldn't have to try one thousand things to see what they do, I would know at least fuzzily what they would do and not have to try them!

What, for example, does the first call  topp->trace(trace_object, 1); do? Why is it there? The format says it should be registering the trace_object just created with "topp" (which is a sort of context, what sort? A "model"? What is that?), but WHY does topp need to know, if that is happening, and what else is happening? Why can't I call eval before that? Why doesn't it matter that the trace file is not yet open? Why does it matter (if it does) later on? Etc. Please add explanations.

pbreuer, 2025-01-13:
Here's a tiny first ingredient of the patch. It does nothing on its own. It's evidently harmless. It adds a m_dumpvars field to VerilatedContext objects containing a list (vector) of the desired dumpvars(level,name) argument pairs. There is a method to add one more to the list, and a method to show the whole vector.

I presume this is the context that magically gets filled with that information by the first call of eval() in main()? I suppose we'll find out as I go through more of the patch.

```
diff --git a/include/verilated.h b/include/verilated.h
--- a/include/verilated.h
+++ b/include/verilated.h
@@ -413,6 +413,12 @@ protected: 
     std::string m_timeFormatSuffix VL_GUARDED_BY(m_timeDumpMutex);  // $timeformat printf format
     std::string m_dumpfile VL_GUARDED_BY(m_timeDumpMutex);  // $dumpfile setting
     
+    // PTB dumpvars list attributes inserted in VerilatedContext object
+    // -- FIXME, use value not ptr, so copy of list passed out, not real one
+    std::vector<std::pair<int, std::string>* > *m_dumpvars
+        = new std::vector<std::pair<int, std::string>* >(); // dumpvar() entries
+    // PTB end extra attribute of VerilatecContext objects
+
     struct NonSerialized final {  // Non-serialized information
         // These are reloaded from on command-line settings, so do not need to persist
         // Fast path
@@ -649,6 +655,19 @@ public:
     void dumpfile(const std::string& flag) VL_MT_SAFE_EXCLUDES(m_timeDumpMutex);
     std::string dumpfileCheck() const VL_MT_SAFE_EXCLUDES(m_timeDumpMutex);
 
+    // PTB new methods added to VerilatedContext objects
+    // Internal: $dumpvars
+    // add new entry at end of dumpvars list
+    void dumpvar(int level, const std::string& name) {
+      m_dumpvars->push_back(new std::pair(level,name));
+    } VL_MT_SAFE_EXCLUDES(m_timeDumpMutex);
+    
+    // return dumpvars list - FIXME return copy: change attr from ptr to value
+    std::vector<std::pair<int, std::string>* > *dumpvars() {
+      return m_dumpvars;
+    }
+    // PTB end new methods added to VerilatedContext objects
+
     // Internal: --prof-exec related settings
     uint64_t profExecStart() const VL_MT_SAFE { return m_ns.m_profExecStart; }
     void profExecStart(uint64_t flag) VL_MT_SAFE;

```

pbreuer, 2025-01-13:
Here's a second bit of the patch, also obviously harmless. It adds a method to the VerilatedVcdC objects to force load their existing dumpvars list from an external.  This appears to be an extension class of VerilatedTrace, so I imagine the change should have been made there instead, because this VerilatedVcdC class appears to be trying to represent VCD files, and there seem to be more logging file-types than that. Here it is:

```
diff --git a/include/verilated_vcd_c.h b/include/verilated_vcd_c.h
--- a/include/verilated_vcd_c.h
+++ b/include/verilated_vcd_c.h
@@ -320,6 +320,21 @@ public:
         m_sptrace.dumpvars(level, hier);
     }

+    // PTB add method to force load VerilatedVcdC objects's dumpvars list
+    // -- FIXME this only ADDs entries, should it have zeroed first?
+    void dumpvars(std::vector<std::pair<int,std::string>*>* v) {
+    // There's other initialization around so this is not so out of place.
+        if (VL_LIKELY(v->size() <= 0))
+           return;
+
+       for ( int i; i < v->size(); i++) {
+           // does existing dumpvars know if something is added twice, and skip?
+          std::pair<int,std::string>* entry = v->at(i);
+           dumpvars( entry->first, entry->second);
+       }
+    }
+    // PTB end of extra method to load VerilatedVcdC object's dumpvars list
+    
     // Internal class access
     VerilatedVcd* spTrace() { return &m_sptrace; }
 };

So that's changes to two class headers so far. Both harmless additions not interfering with anything. Feel free to tell me about better ways. I have no clue.

```

pbreuer, 2025-01-13:
Getting into parser changes now, continuing the obviously harmless theme. This tiny change adds an extra possibility to the classification kinds for $dumpvars statements encountered by the parser in a source system verilog text, because I will leave as much of the existing parse behavior as-is as I can, only dealing with the particular case where the parser explicitly ignored the exprList argument of a $dumpvars(expr,exprList) call before.
```

diff --git a/src/V3Ast.h b/src/V3Ast.h
--- a/src/V3Ast.h
+++ b/src/V3Ast.h
@@ -1319,7 +1319,7 @@ constexpr bool operator==(VDisplayType::
 
 class VDumpCtlType final {
 public:
-    enum en : uint8_t { FILE, VARS, ALL, FLUSH, LIMIT, OFF, ON };
+    enum en : uint8_t { FILE, VARS, ALL, FLUSH, LIMIT, OFF, ON, NVARS };
     enum en m_e;
     VDumpCtlType()
         : m_e{ON} {}

```

pbreuer, 2025-01-13:
More parser changes simply to support the improved parse that will be done. This adds an extra field to the AstDumpCtl object that will represent a $dumpvars(expr,exprList) to contain the second argument that has been up till now not captured at all. Nobody will know about it, the extension to the object is harmless. The field is zero by default.

```
diff --git a/src/V3AstNodeOther.h b/src/V3AstNodeOther.h
--- a/src/V3AstNodeOther.h
+++ b/src/V3AstNodeOther.h
@@ -3034,12 +3034,22 @@ class AstDumpCtl final : public AstNodeS
     // Parents: expr
     // @astgen op1 := exprp : Optional[AstNodeExpr] // Expression based on type of statement
     const VDumpCtlType m_ctlType;  // Type of operation
+    AstNode *cStrList;
 public:
-    AstDumpCtl(FileLine* fl, VDumpCtlType ctlType, AstNodeExpr* exprp = nullptr)
-        : ASTGEN_SUPER_DumpCtl(fl)
-        , m_ctlType{ctlType} {
+    AstDumpCtl( FileLine* fl
+              , VDumpCtlType ctlType
+              , AstNodeExpr* exprp = nullptr
+              // PTB add extra field for strList in $dumpvars(expr,strList)
+              , AstNode*     strsp = nullptr
+              // PTB end of adding extra field
+              ) : ASTGEN_SUPER_DumpCtl(fl), m_ctlType{ctlType} {
         this->exprp(exprp);
+        cStrList = strsp;
     }
+    // PTB add methods for reporting and changing extra field
+    AstNode* strsp() { return this->cStrList; }
+    void strsp(AstNode* strsp) { this->cStrList = strsp; }
+    // PTB end of methods added for extra "strsp" (pointer to strings) field
     ASTGEN_MEMBERS_AstDumpCtl;
     string verilogKwd() const override { return ctlType().ascii(); }
     bool isGateOptimizable() const override { return false; }
```

wsnyder, 2025-01-13:
What the model uses to setup context information before the first eval is the __Syms constructor, e.g. this calls '__vm_contextp->timeunit'.  That would be the place to pass the model's scopevars up.

However I think it's legal to have multiple dumpvar calls controlled by e.g. `$value$plusargs` so the scope is done at runtime, so that suggests this should be done much later when the file is opened.

pbreuer, 2025-01-13:
More still harmless changes to the parser, because I haven't actually changed the parse yet in this sequence of patches. This patch adds something that will create the syntax tree specific to a $dumpvars(expr,strs) entry when I modify the parser to recognize that specific form as such, whereas before it would have been ignoring the second argument completely. Nothing triggers this code yet. It's triggered by the NVARS enum value that I introduced, and nothing in the present parse (so far) creates that.

 NB, this part of the patch isn't quite correct yet. This piece of code actually  emits noise as the syntax tree is built, and that noise ends up as C++ code in a  Vtestbench___024root__DepSet_hfc24d085__0__Slow.cpp: file that gets built. The trouble is that it emits an extraneous vlSymsp->_traceDumpOpen() line in that file, one per each $dumpvars(expr,strs) entry in the system verilog.

What's wrong with that? I'm not quite sure what's up there right now. But one per each entry was definitely too much before, I think it erased the previous entries each time, and one at the end after them all was OK somehow (so the second ones must have done the harm whatever that was), and right now I am emitting NONE at all and everything is fine. So I'll have to figure it out.  I've marked that below.

```
diff --git a/src/V3EmitCFunc.h b/src/V3EmitCFunc.h
--- a/src/V3EmitCFunc.h
+++ b/src/V3EmitCFunc.h
@@ -675,6 +675,28 @@ public:
                 puts("\\n\");\n");
             }
             break;
+        // PTB New case added to deal with 2nd arg in $dumpvars(expr,strs)
+        case VDumpCtlType::NVARS:
+            if (v3Global.opt.trace()) {
+                // (NVARS guarantees nonzero strsp)
+                puts("vlSymsp->_vm_contextp__->dumpvar(");
+                iterateAndNextConstNull(nodep->exprp());
+                puts(", \""); // FIXME so won't need to add quotes
+                iterateAndNextConstNull(nodep->strsp());
+                puts("\"");   // FIXME so won't need to add quotes
+                puts(");\n");
+                // FIXME, only want ONE open at most, zero OK, not one for each dumpvars
+                // putns(nodep, "vlSymsp->_traceDumpOpen();\n");
+            } else {
+                putns(nodep, "VL_PRINTF_MT(\"-Info: ");
+                puts(V3OutFormatter::quoteNameControls(protect(nodep->fileline()->filename())));
+                puts(":");
+                puts(cvtToStr(nodep->fileline()->lineno()));
+                puts(": $dumpvar ignored, as Verilated without --trace");
+                puts("\\n\");\n");
+            }
+            break;
+        // PTB end of case added to deal with  $dumpvars(expr,strs)
         case VDumpCtlType::ALL:
             // $dumpall currently ignored
             break;
```

I can see no "vlSymsp->_traceDumpOpen()" at all in the current contents of the .cpp and .h files that are generated, so apparently nothing at all is needed there. I'll confirm later. The  section of the Vtestbench___024root__DepSet_hfc24d085__0__Slow.cpp generated by those lines looks like this at present:
```

        VL_FINISH_MT("/usr/local/src/cvw-main/testbench/testbench.sv", 362, "");
    }
    __Vtemp_7[0U] = 0x2e766364U;
    __Vtemp_7[1U] = 0x2f73696dU;
    __Vtemp_7[2U] = 0x2f746d70U;
    vlSymsp->_vm_contextp__->dumpfile(VL_CVT_PACK_STR_NW(3, __Vtemp_7));
    vlSymsp->_vm_contextp__->dumpvar(1U, "TOP.testbench.dut.core.ifu.PCF");
    vlSymsp->_vm_contextp__->dumpvar(1U, "TOP.testbench.dut.core.ifu.PCPF");
    vlSelfRef.testbench__DOT__dut__DOT__core__DOT__PMPCFG_ARRAY_REGW[0U] = 0U;
    vlSelfRef.testbench__DOT__dut__DOT__core__DOT__PMPADDR_ARRAY_REGW[0U] = 0U;
}
```
Those Vtemp7 lines seem to be outputting ascii code as numbers. Could be the dumpfile name. Yes it is, see the line following them. Ours not to reason why.

Anyway, definitely no ...Open() line, and it works.

wsnyder, 2025-01-13:
Thanks for these, BTW unless I see something that is in the wrong direction, I'll wait to review the pull as a whole when it comes.

pbreuer, 2025-01-13:
Re your comment about this sequence:

```
    topp->trace(trace_object, 1);
    // eval  first time loads contextp with dumpvars from source
    topp->eval();
    // load now correct dumpvars info into trace object
    trace_object->dumpvars(contextp->dumpvars());
    trace_object->open(VCD_FILE);
```
That was mostly swedish to me ("What the model uses to setup context information before the first eval is the __Syms constructor, e.g. this calls '__vm_contextp->timeunit'. That would be the place to pass the model's scopevars up. However I think it's legal to have multiple dumpvar calls controlled by e.g. $value$plusargs so the scope is done at runtime, so that suggests this should be done much later when the file is opened.") but I think you're saying to push whatever it is that happens there into the trace_object->open(VCD_FILE).

My personal feeling is that if anyone else tries to hide anything ELSE, I will come over there and explain with a crowbar or other heavy implement in hand how much trouble that causes coders.

Yerrs, like something called "open" should have the side-effect of loading dumpvars info! Riiiight! The info shouldn't have been lost in the first place but the right place to pick it up is in something that at least begins with the word "dump".

My impression was that the first call of dump() does do a load of other initializations, so presumably this kind of thing has been encountered before.

What one ACTUALLY should do is figure out which bit of the eval() code does this mysterious movement of information, and take it out and isolate it in some holding prison cell while one figure where it should go for real, and meanwhile stick it in some method called init() and explain to endusers that it's part of the magic incantation to call init() and leave it at that. Ditto for initialization stuff presently found in open() or dump().

Then at least one wouldn't have to go crazy looking for it.

pbreuer, 2025-01-13:
For completeness I'll put up the only missing part of the patch. I haven't actually worked out what it does yet, but it's the only part that actually DOES anything. I'll comment on it later. It's to V3EmitCSyms.cpp . I'll have to look at the code it generates to see what it is that I intended ... presumably this is the missing link patching the information flow of $dumpvars data from the source code, is all I can say for now. Apparently this is in the EmitCSyms "class", surprise surprise, but it's really going to be a subroutine, or a layer, or something like that, conceptually, not an object. It's purely procedural.

```
diff --git a/src/V3EmitCSyms.cpp b/src/V3EmitCSyms.cpp
--- a/src/V3EmitCSyms.cpp
+++ b/src/V3EmitCSyms.cpp
@@ -736,13 +736,37 @@ void EmitCSyms::emitSymImp() {

         puts("\nvoid " + symClassName() + "::_traceDumpOpen() {\n");
         puts("const VerilatedLockGuard lock(__Vm_dumperMutex);\n");
+        puts("VL_PRINTF(\"Enter " + symClassName() + "::_traceDumpOpen().\\n\");\n");
+        puts("bool need_dumpfile_open = false;\n");
+
+        // one shot guard for create trace object is __Vm_dumperp
         puts("if (VL_UNLIKELY(!__Vm_dumperp)) {\n");
-        puts("__Vm_dumperp = new " + v3Global.opt.traceClassLang() + "();\n");
-        puts("__Vm_modelp->trace(__Vm_dumperp, 0, 0);\n");
+        puts("__Vm_dumperp = new "+ v3Global.opt.traceClassLang()+"();\n");
+        puts("__Vm_modelp->trace(__Vm_dumperp, 1);\n");
+
+        // queue open for dumpfile one time only after trace object created
+        puts("need_dumpfile_open = true;\n");
+        puts("} // endif\n");
+
+        // traverse dumpvars list one time only
+        puts("for(_vm_contextp__->dumpvarsInit();\n");
+        puts("_vm_contextp__->dumpvarsCheck();\n");
+        puts("_vm_contextp__->dumpvarsNext()\n");
+        puts(") {\n");
+        puts("__Vm_dumperp->dumpvars( _vm_contextp__->dumpvarsLevel()\n");
+        puts(", _vm_contextp__->dumpvarsName()->c_str()\n");
+        puts(");\n");
+        puts("VL_PRINTF(\"Tracing level %d %s.\\n\", _vm_contextp__->dumpvarsLevel(), _vm_contextp__->dumpvarsName()->c_str());\n");
+        puts("} // endfor\n");
+
+        // carry out queued possible open for dumpfile
+        puts("if (need_dumpfile_open) {\n");
         puts("std::string dumpfile = _vm_contextp__->dumpfileCheck();\n");
-        puts("__Vm_dumperp->open(dumpfile.c_str());\n");
+        puts("VL_PRINTF(\"Tracing to %s.\\n\", dumpfile.c_str());\n");
         puts("__Vm_dumping = true;\n");
-        puts("}\n");
+        puts("__Vm_dumperp->open(dumpfile.c_str());\n");
+        puts("} // endif\n");
+
         puts("}\n");

         puts("\nvoid " + symClassName() + "::_traceDumpClose() {\n");

```

wsnyder, 2025-01-14:
Great, note that dumpvars doesn't take a string but takes multiple module and/or variable scopes.

    $dumpvars(3, top.foo.instname, top.foo.signame, top.foo.signame2);

pbreuer, 2025-01-14:
On Mon, 13 Jan 2025 17:30:53 -0800
Wilson Snyder ***@***.***> wrote:

> Great, note that dumpvars doesn't take a string but takes multiple module and/or variable scopes.

I know. You can fix that! I don't know which parser production does
ids.  I settled for strings, which the C++code dumpvars works on, and
which I could see in the BNF.

Note also that the strings are absolute id reference paths, starting
"TOP.".

>     $dumpvars(3, top.foo.instname, top.foo.signame, top.foo.signame2);

I've noted other improvements with FIXMEs.

PTB

pbreuer, 2025-01-14:
A thing I noticed today is that the patch retains the " putns(nodep, "vlSymsp->_traceDumpOpen();\n");" in the new NVARS case in src/V3EmitCFunc.h that was held over from the old VARS case. That line was giving me a lot of trouble and I thought I had decided to get rid of it as it seemed to do nothing that I could detect,  while leaving it in put yet another `vlSymsp->_traceDumpOpen(); `after every single `vlSymsp->_vm_contextp__->dumpvar(1U, "TOP.foo");` line in the generated code, and that had bad effects - rather, it used to. It seems to be fine now, which is why probably I left it in! But I don't like it. I don't know why it had bad effects and I don't know why it is OK now, which adds up to "don't like".
I would feel easier with it gone. At least if then that causes some problem that I do not detect someone would find out why the line was there and say so ...
(The bad effect was seemingly to erase memory of every dumpvars item except the last).