verilator/verilator #5728: Support for `` `resetall ``

sifferman, 2025-01-11:
Hello!

I ran into timescale issues with simulating Alex Forencich's IPs. and realized the issue is that `` `resetall `` seems to not be fully supported.

alexforencich/verilog-axi rtl/axi_dma.v: <https://github.com/alexforencich/verilog-axi/blob/38915fb5330cb8270b454afc0140a94489dc56db/rtl/axi_dma.v#L27-L29>

I believe the Verilator implementation only resets `` `default_nettype ``:
https://github.com/verilator/verilator/blob/bb871728c92553c3813f556782807b5879135ea1/src/verilog.l#L1065-L1066

However, these are the requirements from the LRM:

> ### 22.3 `` `resetall ``
> When the `` `resetall `` compiler directive is encountered during compilation, all compiler directives are set to the default values. This is useful for ensuring that only directives that are desired in compiling a particular source file are active.

The affected directives seem to be the following:

* `` `default_nettype `` [22.8]: `` `resetall `` resets to `wire`
* `` `celldefine `` [22.10]: `` `resetall `` acts as `` `endcelldefine ``
* `` `pragma `` [22.11.1]: `` `resetall `` reverts all pragmas to defaults
* `` `timescale ``[22.7]: `` `resetall `` reverts timescale back to tool default
* `` `unconnected_drive `` [22.9]: `` `resetall `` acts as `` `nounconnected_drive ``

Here is the Icarus implementation: <https://github.com/steveicarus/iverilog/blob/30123f89451939fc946a2395c91e39911fcd354a/lexor.lex#L1616-L1622>

I can add a fix this weekend. Would this be best fixed by adding more to verilog.l? Or is there a good .cpp file to put a resetall function?

Thanks!

wsnyder, 2025-01-11:
Yes, all of those seem to be best in verilog.l, thanks.

sifferman, 2025-01-12:
Actually, my issue had nothing to do with `` `resetall ``; I was just misunderstanding how timescale scopes work. All is resolved

But I can still submit a fix to finish `` `resetall ``

wsnyder, 2025-01-12:
If you're willing, otherwise we can close this until someone else hits it.

wsnyder, 2025-01-12:
Thanks, I can live without strict tests as we do check `resetall at least in some respects.

PR #5730 (2025-01-12): Added support for `resetall
(Probably) Fixes #5728

I will not be providing a test case because I couldn't figure out a situation that works on commercial tools, is consistent with the LRM, and was fixed by this PR.

You are welcome to close this without merging if you feel it is not worth pursuing further. Or you are welcome to merge without testing if you are feeling *adventurous*. :monkey:

------

This was my failed attempt at a test case, and I don't understand the output.
```systemverilog
localparam real default_timeunit = 100s;

`timescale 10s/1s
`default_nettype none
`celldefine

local [... cut]
closed 2025-01-12 by wsnyder