verilator/verilator #5733: Segfault in WidthVisitor

lefp, 2025-01-15:
[Backtrace](https://github.com/user-attachments/files/18429475/bt.log)

Can't share the code that broke it. Will try to create an MVA if I figure out what triggers the bug.

Command used:
```
verilator -j 0 --assert --timing --trace-fst -Wno-fatal --binary -I<redacted> test/<redacted>.v --top <redacted> src/*.v src/spinalhdl_gen/* --debug --gdbbt
```

Version: latest master at time of writing this.
```
Verilator 5.033 devel rev v5.032-37-g20faa9946
```

OS: Fedora 40.

lefp, 2025-01-15:
Here's an MVA that triggers the same bug:
```verilog
module TestModule;

    wire clk;

    let f(x) = (x << 1);

    always @(posedge clk) begin
        case (0)
            0: f(0); // this is clearly a programmer error
        endcase
    end

endmodule
```
Run with `verilator --binary test.v`.

wsnyder, 2025-01-16:
Thanks for the test, I see the issue and will look into a fix.

closed 2025-02-26 by wsnyder