verilator/verilator #5736: Internal Error: multiple write refs on LHS of NBA

7FM, 2025-01-17:
Error Message:
```
%Warning-WIDTHTRUNC: CommonLogicModule.sv:2471:26: Operator VAR 'MAX_PTR_VAL' expects 2 bits on the Initial value, but Initial value's SUB generates 32 or 3 bits.
                                                                : ... note: In instance 'SCAL.INPUTFIFO_WrCommit_spawn_sqrt_decoupled_valid_INPUTs_inst'
 2471 | localparam FIFOPointer_t MAX_PTR_VAL = NR_ELEMENTS-1;
      |                          ^~~~~~~~~~~
                     ... For warning description see https://verilator.org/warn/WIDTHTRUNC?v=5.030
                     ... Use "/* verilator lint_off WIDTHTRUNC */" and lint_on around source to disable this message.
%Error: Internal Error: CommonLogicModule.sv:2483:56: ../V3Delayed.cpp:931: Multiple Write refs on LHS of NBA
 2483 | function FIFOPointer_t prevPointer(input FIFOPointer_t val);
      |                                                        ^~~
```

[CommonLogicModule.sv.txt](https://github.com/user-attachments/files/18456032/CommonLogicModule.sv.txt)
`verilator --lint-only CommonLogicModule.sv`

What 'verilator' command line do we use to run your example?

I tried version `5.032` and `5.030`, both can reproduce this issue. Version `5.022` does not throw an internal error.

wsnyder, 2025-01-18:
@gezalore looks related to your recent rework in V3Delayed.  The AstAssignDly with AstExprStmt with AstVarRef-lvalue is confusing it.  Might you take a look? Thanks.

wsnyder, 2025-01-18:
Simpler test case

[bug5736.v.txt](https://github.com/user-attachments/files/18462009/bug5736.v.txt)

PR #5740 (2025-01-20): Fix AstExprStmt with assignment under LHS of NBA
The problem is the following form, which arises from function inlining:

```
foo[AstExprStmt(x = 2; x)] <= ...
```

We turn it into:

```
x = 2;
foo[x] <= ...
```

Fixes #5736

We really should get rid of AstExprStmt