verilator/verilator #5737: static objects equality

fonarick, 2025-01-18:
hi, i simulating this simple module and $display says "1", which means static variables (2 different allocated objects) are equal. Is it a right behaviour? As i understand, they point to the same memory address
> module top();
> 
>class A;
>endclass
>
>class B;
>  static A obj = new();
>endclass
>
>class C;
>  static A obj = new();
>endclass
>
>initial begin
>  $display(B::obj == C::obj);
>end
>
>endmodule

wsnyder, 2025-01-18:
Initial order is not defined. Probably both are null at start, try a #1 before your display test and use --timing.

fonarick, 2025-01-18:
@wsnyder Thanks for answer
i tried #1 with --timing, but still getting same result. They are not null and 2 added $display show it
Code:
>module top();
>
>class A;
>	int num;
>	function new(int num);
>		this.num = num;
>	endfunction
>endclass
>
>class B;
>static A obj = new(2);
>endclass
>
>class C;
>static A obj = new(5);
>endclass
>
>initial begin
>#1;
>$display(B::obj);
>$display(C::obj);
>$display(B::obj == C::obj);
>end
>
>endmodule

Output:
'{num:'h2}
'{num:'h5}
1

[log.txt](https://github.com/user-attachments/files/18465779/log.txt)

closed 2025-01-18 by wsnyder